MichaIng/DietPi #3746: The access to smbus and i2c is missing on NanoPi NEO2

ImBearChild, 2020-08-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `10.5`
- Kernel version | `Linux DietPi 5.7.15-sunxi64 #20.08 SMP Mon Aug 17 10:32:57 CEST 2020 aarch64 GNU/Linux`
- SBC model | `NanoPi NEO2 (aarch64)` 
- Power supply used | 5V 2A USB Power
- SDcard used | Kingston

#### Additional Information (if applicable)
- Software title | Linux Kernel 

#### Steps to reproduce
The access to smbus and i2c is missing. There're no  `/dev/i2c-0` or `/dev/i2c/0' on file system, and Python module `smbus
` dosen't work .

```python
import smbus
bus = smbus.SMBus(0)
IOError: [Errno 2] No such file or directory
```
i2cdetect can't work too:
```bash
# i2cdetect -y 0
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
```

#### Expected behaviour
Just make i2c work.

#### Actual behaviour
The access to smbus and i2c is missing.

MichaIng, 2020-08-28:
Many thanks for your report. You should be able to enable those additional features by:
0. Check `/boot/dtb/allwinner/overlay/`, the `sun50i-h2-*` prefixed files for available device trees overlays
1. Open `/boot/armbianEnv.txt`
2. Append overlays to the `overlays=` line that you want to enable, leaving away the prefix, like:
    ```
    overlays=... i2c0
    ```
    to enable the first I2C device. It is even possible to adjust the CPU clock, I just see. We really need to make those overlays/features accessible through `dietpi-config`!

ImBearChild, 2020-08-30:
> 
> 
> Many thanks for your report. You should be able to enable those additional features by:
> 
>     1. Check `/boot/dtb/allwinner/overlay/`, the `sun50i-h5-*` prefixed files for available device trees overlays
> 
>     2. Open `/boot/armbianEnv.txt`
> 
>     3. Append overlays to the `overlays=` line that you want to enable, leaving away the prefix, like:
>        ```
>        overlays=... i2c0
>        ```
>        
>        
>        to enable the first I2C device. It is even possible to adjust the CPU clock, I just see. We really need to make those overlays/features accessible through `dietpi-config`!
> 
> 
> Yes indeed we didn't add WiringNP, only WiringPi for RPi and Odroids so far. Would be great if someone could contribute an implementation for NanoPi indeed. Since our images are based on Armbian, not on any official FriendlyARM image (they don't publish Debian images anymore), I'm not sure if the instructions from their wiki work, at least probably some additional steps are required, like enabled the hardware features via device tree overlays, lets see if that is enough.

Thank you! It works. But WiringNP still not work, no matter what overlays I choose. I guess there may be some difference between official FriendlyARM Kernel and Armbian Kernel. 

Considering that i2c can work now, differences in `/proc/cpuinfo` may be the source of the problem. It is just ` Unable to determine board revision from /proc/cpuinfo`.

MichaIng, 2020-08-30:
Jep this makes sense, /proc/cpuinfo content depends on the kernel and Armbian simply ships an own one. Now I remember there was a discussion about exactly that issue already, ah even a software request: #3514
Probably you even find a solution for your particular device there.

MichaIng, 2021-06-02:
I'm gonna mark this as closed. Adding a device tree overlay toggles in dietpi-config is a separate topic and WiringNP seems to be a difficult once (much manual work to get it working with Armbian kernel), tracked here: #3514

closed 2021-06-02 by MichaIng