MichaIng/DietPi #3749: Error when installing Docker

MonsieurKoZo, 2020-08-31:
#### Details:
- Date           | Mon Aug 31 16:00:04 CEST 2020
- Bug report     | 3e770cb3-574a-496a-a8b5-8d13d8b8a3b5
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | ./DockerInstall.sh
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install docker on RPi 3B under dietpi with dietpi-software

#### Expected behaviour:

Install Docker

#### Actual behaviour:
- Exit with error

#### Extra details:

- ...
#### Additional logs:
```
# Executing docker install script, commit: 26ff363bcf3b3f5a00498ac43694bf1c7d9ce16c
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
+ sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
+ sh -c apt-get update -qq >/dev/null
+ [ -n  ]
+ sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MonsieurKoZo, 2020-08-31:
Additional information.

After the failing install I run : 

systemctl start docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
root@DietPi:~# systemctl status docker.service
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2020-08-31 16:20:47 CEST; 1s ago
     Docs: https://docs.docker.com
  Process: 4483 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
 Main PID: 4483 (code=exited, status=1/FAILURE)

août 31 16:20:47 DietPi systemd[1]: docker.service: Failed with result 'exit-code'.
août 31 16:20:47 DietPi systemd[1]: Failed to start Docker Application Container Engine.

Then : 

dockerd
INFO[2020-08-31T16:23:01.397402188+02:00] Starting up                                  
INFO[2020-08-31T16:23:01.406165075+02:00] parsed scheme: "unix"                         module=grpc
INFO[2020-08-31T16:23:01.406352366+02:00] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-08-31T16:23:01.406502105+02:00] ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock 0  <nil>}] <nil>}  module=grpc
INFO[2020-08-31T16:23:01.406598407+02:00] ClientConn switching balancer to "pick_first"  module=grpc
INFO[2020-08-31T16:23:01.411106360+02:00] parsed scheme: "unix"                         module=grpc
INFO[2020-08-31T16:23:01.411268234+02:00] scheme "unix" not registered, fallback to default scheme  module=grpc
INFO[2020-08-31T16:23:01.411364484+02:00] ccResolverWrapper: sending update to cc: {[{unix:///run/containerd/containerd.sock 0  <nil>}] <nil>}  module=grpc
INFO[2020-08-31T16:23:01.411454692+02:00] ClientConn switching balancer to "pick_first"  module=grpc
ERRO[2020-08-31T16:23:01.424748397+02:00] failed to mount overlay: no such device       storage-driver=overlay2
ERRO[2020-08-31T16:23:01.441135268+02:00] AUFS was not found in /proc/filesystems       storage-driver=aufs
ERRO[2020-08-31T16:23:01.451646274+02:00] failed to mount overlay: no such device       storage-driver=overlay
WARN[2020-08-31T16:23:01.471400738+02:00] Your kernel does not support cgroup memory limit 
WARN[2020-08-31T16:23:01.471495634+02:00] Your kernel does not support cgroup cfs period 
WARN[2020-08-31T16:23:01.471538915+02:00] Your kernel does not support cgroup cfs quotas 
WARN[2020-08-31T16:23:01.471579800+02:00] Your kernel does not support cgroup rt period 
WARN[2020-08-31T16:23:01.471620061+02:00] Your kernel does not support cgroup rt runtime 
WARN[2020-08-31T16:23:01.471660581+02:00] Unable to find blkio cgroup in mounts        
INFO[2020-08-31T16:23:01.472524068+02:00] Loading containers: start.                   
WARN[2020-08-31T16:23:01.493282435+02:00] Running modprobe bridge br_netfilter failed with message: modprobe: ERROR: could not insert 'bridge': Exec format error
modprobe: ERROR: could not insert 'br_netfilter': Exec format error
insmod /lib/modules/5.4.51-v7+/kernel/net/llc/llc.ko 
insmod /lib/modules/5.4.51-v7+/kernel/net/llc/llc.ko 
, error: exit status 1 
WARN[2020-08-31T16:23:01.514204187+02:00] Running iptables --wait -t nat -L -n failed with message: `iptables/1.8.2 Failed to initialize nft: Protocol not supported`, error: exit status 1 
WARN[2020-08-31T16:23:01.559628824+02:00] Failed to read iptables version: exit status 1 
INFO[2020-08-31T16:23:01.924548313+02:00] stopping event stream following graceful shutdown  error="<nil>" module=libcontainerd namespace=moby
failed to start daemon: Error initializing network controller: error obtaining controller instance: failed to create NAT chain DOCKER: iptables failed: iptables -t nat -N DOCKER: iptables/1.8.2 Failed to initialize nft: Protocol not supported
 (exit status 1)

After that, I uninstalled completely with : 
sudo apt-get purge docker-ce docker-ce-cli containerd.io
sudo rm -rf /var/lib/docker

And run the install without dietpi-software : 
curl -sSL https://get.docker.com | sh

Same result.

Joulinar, 2020-08-31:
Hi,

many thanks for your message. I did a test installation on my RPi3B+ without issues.

```
DockerInstall.sh                                            100%[========================================================================================================================================>]  13.53K  --.-KB/s    in 0s

2020-08-31 17:49:29 (125 MB/s) - ‘DockerInstall.sh’ saved [13857/13857]

[ INFO ] DietPi-Software | Running Docker installer, please wait...
# Executing docker install script, commit: 26ff363bcf3b3f5a00498ac43694bf1c7d9ce16c
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
+ sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
+ sh -c apt-get update -qq >/dev/null
+ [ -n  ]
+ sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
+ sh -c docker version
Client: Docker Engine - Community
 Version:           19.03.12
 API version:       1.40
 Go version:        go1.13.10
 Git commit:        48a6621
 Built:             Mon Jun 22 15:53:41 2020
 OS/Arch:           linux/arm
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.12
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.10
  Git commit:       48a6621
  Built:            Mon Jun 22 15:47:34 2020
  OS/Arch:          linux/arm
  Experimental:     false
```
Pls can you run `apt install docker-ce` manually

MichaIng, 2020-08-31:
> modprobe: ERROR: could not insert 'br_netfilter': Exec format error

Since the loaded kernel version and module path match what is expected, this looks like a corrupted file, so I guess the second install attempt will fail the same wait. If that is true, could you go on with the following steps and print the result here:
```
ls -al /lib/modules/5.4.59-v7+/kernel/net/bridge/
modprobe br_netfilter.ko
dmesg | tail -10
```

MonsieurKoZo, 2020-08-31:
Hi,
Thanks for you answers.
Indeed, manual install as advised by Joulinar also failed.

@MichaIng I do not have the same module number. Here is what I have in the folder : 
ls -al /lib/modules/
total 24
drwxr-xr-x  6 root root 4096 juil. 24 23:56 .
drwxr-xr-x 17 root root 4096 juin   1 18:54 ..
drwxr-xr-x  3 root root 4096 août  31 13:56 5.4.51+
drwxr-xr-x  3 root root 4096 août  31 13:57 5.4.51-v7+
drwxr-xr-x  3 root root 4096 août  31 13:56 5.4.51-v7l+
drwxr-xr-x  3 root root 4096 août  31 13:56 5.4.51-v8+

Do you think this is the cause of that issue?
Thanks

Joulinar, 2020-08-31:
Pls can you post the entire error message from manual installation as well as the 2 commands @MichaIng requested.

MonsieurKoZo, 2020-08-31:
@Joulinar  Sorry I forgot to post it. Here it is (sorry the OS is in French) : 

apt install docker-ce
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Les paquets supplémentaires suivants seront installés : 
  containerd.io docker-ce-cli
Paquets recommandés :
  cgroupfs-mount | cgroup-lite pigz xz-utils
Les NOUVEAUX paquets suivants seront installés :
  containerd.io docker-ce docker-ce-cli
0 mis à jour, 3 nouvellement installés, 0 à enlever et 0 non mis à jour.
Il est nécessaire de prendre 15,1 Mo/57,7 Mo dans les archives.
Après cette opération, 255 Mo d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] 
Réception de :1 https://download.docker.com/linux/raspbian buster/stable armhf containerd.io armhf 1.2.13-2 [15,1 MB]
15,1 Mo réceptionnés en 3s (4 767 ko/s)     
Sélection du paquet containerd.io précédemment désélectionné.
(Lecture de la base de données... 30041 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../containerd.io_1.2.13-2_armhf.deb ...
Dépaquetage de containerd.io (1.2.13-2) ...
Sélection du paquet docker-ce-cli précédemment désélectionné.
Préparation du dépaquetage de .../docker-ce-cli_5%3a19.03.12~3-0~raspbian-buster_armhf.deb ...
Dépaquetage de docker-ce-cli (5:19.03.12~3-0~raspbian-buster) ...
Sélection du paquet docker-ce précédemment désélectionné.
Préparation du dépaquetage de .../docker-ce_5%3a19.03.12~3-0~raspbian-buster_armhf.deb ...
Dépaquetage de docker-ce (5:19.03.12~3-0~raspbian-buster) ...
Paramétrage de containerd.io (1.2.13-2) ...
Created symlink /etc/systemd/system/multi-user.target.wants/containerd.service → /lib/systemd/system/containerd.service.
Paramétrage de docker-ce-cli (5:19.03.12~3-0~raspbian-buster) ...
Paramétrage de docker-ce (5:19.03.12~3-0~raspbian-buster) ...
Created symlink /etc/systemd/system/multi-user.target.wants/docker.service → /lib/systemd/system/docker.service.
Created symlink /etc/systemd/system/sockets.target.wants/docker.socket → /lib/systemd/system/docker.socket.
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
invoke-rc.d: initscript docker, action "start" failed.
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2020-08-31 18:01:03 CEST; 35ms ago
     Docs: https://docs.docker.com
  Process: 5443 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
 Main PID: 5443 (code=exited, status=1/FAILURE)
dpkg: erreur de traitement du paquet docker-ce (--configure) :
 installed docker-ce package post-installation script subprocess returned error exit status 1
Traitement des actions différées (« triggers ») pour systemd (241-7~deb10u4+rpi1) ...
Des erreurs ont été rencontrées pendant l'exécution :
 docker-ce
E: Sub-process /usr/bin/dpkg returned an error code (1)

MonsieurKoZo, 2020-08-31:
Hi again,
I don't know if it is related but I see error messages regarding iptables.
When running iptables --list, I get the message : 
iptables/1.8.2 Failed to initialize nft: Protocol not supported

MonsieurKoZo, 2020-08-31:
Well guys I am so sorry because it seems I wasted your time :(
After the purge and reboot, I installed via dietpi-software without any issue...

Sorry again and thanks a lot for your help!

Joulinar, 2020-08-31:
no worries, we are here to help. Good that you found the root. Maybe something hanging on `iptabes` after some update? At least it's solved now. I'm going to close the ticket.

closed 2020-08-31 by Joulinar

MichaIng, 2020-08-31:
Strange, then probably there was a kernel package upgrade without any version string change, I remember there was one a few days ago. I didn't know that this requires a reboot then as well to have loaded kernel and modules matching again. Great that it works.

scorgn, 2021-08-07:
Same thing happened to me (same error messages, everything), purge and reboot and everything installed!

So, @MonsieurKoZo you definitely did not waste anyone's time. In fact, you saved me a bunch of time because I would have had no idea to try that!