MichaIng/DietPi #3752: Boot from USB Raspberry pi Menu Option

davidjmorin, 2020-09-02:
Would love to see this option in the config menu.  I updated today and no joy.

Does dietpi have a step by step on how to do this with current Version? 

I have looked all around and only found ones for raspberry os.

Joulinar, 2020-09-02:
Hi,

many thanks for your question. There isn't such an option available via DietPi options. There are different ways described by Raspberry Foundation depending on your devices.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

MichaIng, 2020-09-02:
The easiest way is to flash the image to the USB drive directly, on RPi3+ and RPi4 this works OOTB. All ways to move the boot partition afterwards are not that save, best is dd cloning the SD card from an external system.

On RPi3 non-plus as described on the RPi docs, this OTP bit needs to be set to allow USB boot. This can be done via dietpi-config as well, but since its an RPi3-nonplus-only option, its also only shown there 😉. All models prior to RPi3 do not support USB boot, hence moving the root partition only (not the boot partition) is the best you can do, e.g. via `dietpi-drive_manager`.

davidjmorin, 2020-09-02:
Has dietpi been updated so that stable will update when running the
update?

davidjmorin, 2020-09-02:
`-- A start job for unit rpi-eeprom-update.service has begun execution.
--
-- The job identifier is 749.
Sep 02 18:02:37 DietPi rpi-eeprom-update[6234]: /usr/bin/rpi-eeprom-update: 271: /usr/bin/rpi-eeprom-update: hexdump: not found
Sep 02 18:02:37 DietPi systemd[1]: rpi-eeprom-update.service: Main process exited, code=exited, status=127/n/a
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit rpi-eeprom-update.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 127.
Sep 02 18:02:37 DietPi systemd[1]: rpi-eeprom-update.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit rpi-eeprom-update.service has entered the 'failed' state with result 'exit-code'.
Sep 02 18:02:37 DietPi systemd[1]: Failed to start Check for Raspberry Pi EEPROM updates.
-- Subject: A start job for unit rpi-eeprom-update.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit rpi-eeprom-update.service has finished with a failure.
--
-- The job identifier is 749 and the job result is failed.
`

davidjmorin, 2020-09-02:
`[  OK  ] APT update
root@DietPi:/mnt/mydrive# G_AGI rpi-eeprom rpi-eeprom-images flashrom libraspberrypi-bin raspberrypi-bootloader
[ INFO ] APT install for: rpi-eeprom rpi-eeprom-images flashrom libraspberrypi-bin raspberrypi-bootloader, please wait...
[  OK  ] APT install for: rpi-eeprom rpi-eeprom-images flashrom libraspberrypi-bin raspberrypi-bootloader
root@DietPi:/mnt/mydrive# G_AG_CHECK_INSTALL_PREREQ binutils
[ INFO ] Checking for required APT packages: binutils
[  OK  ] All required APT packages are already installed.
root@DietPi:/mnt/mydrive# rpi-eeprom-update -a
/usr/bin/rpi-eeprom-update: 271: /usr/bin/rpi-eeprom-update: hexdump: not found
root@DietPi:/mnt/mydrive#
`

Joulinar, 2020-09-02:
@davidjmorin 
can I ask a question? On what Raspberry Pi model you are? because `rpi-eeprom-update` is working on RPi4 only.

davidjmorin, 2020-09-02:
raspberry pi4 4gb

bbsixzz, 2020-09-02:
`rpi-eeprom-update` is not working on two of my RPi4 4GB after the latest update:

`/usr/bin/rpi-eeprom-update: 271: /usr/bin/rpi-eeprom-update: hexdump: not found`

Joulinar, 2020-09-02:
hmm on my RPi4B 4GB it'S working fine

```
root@DietPi4:~# rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 LATEST: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
root@DietPi4:~#
```

bbsixzz, 2020-09-02:
`apt update` and `apt upgrade` produced `Unpacking rpi-eeprom (7.12-1) over (7.11-1) ... ` but same result after reboot.

davidjmorin, 2020-09-02:
> hmm on my RPi4B 4GB it's working fine
> 
> ```
> root@DietPi4:~# rpi-eeprom-update
> BCM2711 detected
> Dedicated VL805 EEPROM detected
> BOOTLOADER: up-to-date
> CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
>  LATEST: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
>  FW DIR: /lib/firmware/raspberrypi/bootloader/critical
> VL805: up-to-date
> CURRENT: 000137ad
>  LATEST: 000137ad
> root@DietPi4:~#
> ```
> 
> Question: Are you using 32bit or 64bit image?

`root@DietPi:~# rpi-eeprom-update
/usr/bin/rpi-eeprom-update: 271: /usr/bin/rpi-eeprom-update: hexdump: not found
root@DietPi:~#
`
This is 32bit

Joulinar, 2020-09-02:
I have installed version `7.10-1` still.

```
root@DietPi4:~# dpkg -l rpi-eeprom
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-==================================
ii  rpi-eeprom     7.10-1       all          Raspberry Pi 4 boot EEPROM updater
root@DietPi4:~#
```
But it's offered as update. 
```
root@DietPi4:~# apt list --upgradable
Listing... Done
rpi-eeprom-images/testing 7.12-1 all [upgradable from: 7.11-1]
rpi-eeprom/testing 7.12-1 all [upgradable from: 7.10-1]
root@DietPi4:~#
```
Maybe some bug on latest version of `rpi-eeprom`

davidjmorin, 2020-09-02:
> I have installed version `7.10-1` still.
>
> ```
> root@DietPi4:~# dpkg -l rpi-eeprom
> Desired=Unknown/Install/Remove/Purge/Hold
> | [...]

`root@DietPi:~# dpkg -l rpi-eeprom
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-==================================
ii  rpi-eeprom     7.12-1       all          Raspberry Pi 4 boot EEPROM updater
`

Odd mine is different

Joulinar, 2020-09-02:
Looks like 7.12 is buggy one?

davidjmorin, 2020-09-02:
> Looks like 7.12 is buggy one?

Very possible.  Gonna squash this attempt for now.  Hopefully this gets fixed soon.

bbsixzz, 2020-09-02:
I should prolly not update the eeprom so often, this is what you get :[

Joulinar, 2020-09-02:
Unfortunately something we can't influence from DietPi side. 😢

Joulinar, 2020-09-02:
found this issue. But not sure if this exactly same or similar https://github.com/raspberrypi/rpi-eeprom/issues/211

MichaIng, 2020-09-02:
Do you guys actually have the hexdump command? `which hexdump`

Joulinar, 2020-09-02:
@MichaIng 
would it make sense to downgrade `eeprom` to 7.10 ?

```
cd /tmp
wget http://archive.raspberrypi.org/debian/pool/main/r/rpi-eeprom/rpi-eeprom_7.10-1_all.deb
dpkg -i ./rpi-eeprom_7.10-1_all.deb
```

🤔 

At least on my test it was working. But you are the expert 😄 

```
root@DietPi4:~# rpi-eeprom-update
/usr/bin/rpi-eeprom-update: 271: /usr/bin/rpi-eeprom-update: hexdump: not found
root@DietPi4:~# cd /tmp/
root@DietPi4:/tmp# wget http://archive.raspberrypi.org/debian/pool/main/r/rpi-eeprom/rpi-eeprom_7.10-1_all.deb
--2020-09-02 22:17:08--  http://archive.raspberrypi.org/debian/pool/main/r/rpi-eeprom/rpi-eeprom_7.10-1_all.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 176.126.240.84, 176.126.240.86, 176.126.240.167, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|176.126.240.84|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 351188 (343K) [application/x-debian-package]
Saving to: ‘rpi-eeprom_7.10-1_all.deb’

rpi-eeprom_7.10-1_all.deb             100%[=========================================================================>] 342.96K   508KB/s    in 0.7s

2020-09-02 22:17:14 (508 KB/s) - ‘rpi-eeprom_7.10-1_all.deb’ saved [351188/351188]

root@DietPi4:/tmp# dpkg -i ./rpi-eeprom_7.10-1_all.deb
dpkg: warning: downgrading rpi-eeprom from 7.12-1 to 7.10-1
(Reading database ... 83296 files and directories currently installed.)
Preparing to unpack ./rpi-eeprom_7.10-1_all.deb ...
Unpacking rpi-eeprom (7.10-1) over (7.12-1) ...
Setting up rpi-eeprom (7.10-1) ...
root@DietPi4:/tmp# rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 LATEST: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
root@DietPi4:/tmp#
```

MichaIng, 2020-09-02:
Bug reported: https://github.com/raspberrypi/rpi-eeprom/issues/215

Joulinar, 2020-09-02:
@MichaIng we are on 32bit not 64bit. As far as I can see you linked following. Does it makes a difference 

> This is part of the bsdmainutils package: https://packages.debian.org/buster/amd64/bsdmainutils/filelist

MichaIng, 2020-09-02:
It's the same for armhf and all other architectures and the same for Raspbian as well, but Raspbian doesn't have such a nice online portal to view package details 😉.

Joulinar, 2020-09-02:
as I don't want to play that much on my RPi4 (prod), I was testing it on my RPi3B+. And yes it seems to fix it

```
root@DietPi3:/tmp# dpkg -l rpi-eeprom
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-==================================
ii  rpi-eeprom     7.12-1       all          Raspberry Pi 4 boot EEPROM updater
root@DietPi3:/tmp#
```

```
root@DietPi3:/tmp# systemctl status rpi-eeprom-update.service
● rpi-eeprom-update.service - Check for Raspberry Pi EEPROM updates
   Loaded: loaded (/lib/systemd/system/rpi-eeprom-update.service; enabled; vendor preset: enabled)
   Active: active (exited) since Wed 2020-09-02 22:36:23 CEST; 2s ago
  Process: 1309 ExecStart=/usr/bin/rpi-eeprom-update -a (code=exited, status=0/SUCCESS)
 Main PID: 1309 (code=exited, status=0/SUCCESS)

Sep 02 22:36:23 DietPi3 systemd[1]: Starting Check for Raspberry Pi EEPROM updates...
Sep 02 22:36:23 DietPi3 systemd[1]: Started Check for Raspberry Pi EEPROM updates.
root@DietPi3:/tmp# dpkg -l rpi-eeprom
rpi-eeprom         rpi-eeprom-images

```

MichaIng, 2020-09-02:
Although on RPi3 there is no EEPROM, this RPi4 only 😉.

Joulinar, 2020-09-02:
I know 😉 but I could have it installed anyway. Just to play around with this.

MichaIng, 2020-09-02:
Yes, also I cannot see an explicit exit path for non-RPi4 models, I guess some of the EEPROM firmware version checks will give an empty output to make the script exit gracefully without applying something.

Joulinar, 2020-09-02:
correct, it's running but don't do anything on the RPi3B+

MichaIng, 2020-09-02:
I just opened a PR to add the dependency: https://github.com/raspberrypi/rpi-eeprom/pull/216

Joulinar, 2020-09-03:
ok `rpi-eeprom` has been fixed on version `7.13-1`. Thx @MichaIng for your time on this.

```
root@DietPi4:~# dpkg -l rpi-eeprom
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-==================================
ii  rpi-eeprom     7.13-1       all          Raspberry Pi 4 boot EEPROM updater
root@DietPi4:~#
```

```
root@DietPi4:~# rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 LATEST: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
root@DietPi4:~#
```

MichaIng, 2020-09-03:
Great, thanks to the RPi devs for the quick release. Also the final solution was IMO the best possible, moving to a command that is part of essential `coreutils` so that no additional dependency was required anymore 👍.

closed 2020-09-03 by MichaIng