MichaIng/DietPi #3753: pi zero missing network link

tcurdt, 2020-09-02:
### Creating a bug report/issue

#### Required Information

- DietPi version | 6, 30, 0, master, MichaIng
- Distro version | 10.4
- Kernel version | 4.19.118+
- SBC model | RPi Zero W (armv6l)
- Power supply used | 5V 2A
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
#### Steps to reproduce
1. Flash Diepi
2. Change `diepi-wifi.txt' to have two wifi entries
```
aWIFI_SSID[0]='x'
aWIFI_KEY[0]='x'
aWIFI_KEYMGR[0]='WPA-PSK'
aWIFI_PROTO[0]=''
aWIFI_PAIRWISE[0]=''
aWIFI_AUTH_ALG[0]=''
aWIFI_EAP[0]=''
aWIFI_IDENTITY[0]=''
aWIFI_PASSWORD[0]=''
aWIFI_PHASE1[0]=''
aWIFI_PHASE2[0]=''
aWIFI_CERT[0]=''

aWIFI_SSID[1]='x'
aWIFI_KEY[1]='x'
aWIFI_KEYMGR[1]='WPA-PSK'
aWIFI_PROTO[1]=''
aWIFI_PAIRWISE[1]=''
aWIFI_AUTH_ALG[1]=''
aWIFI_EAP[1]=''
aWIFI_IDENTITY[1]=''
aWIFI_PASSWORD[1]=''
aWIFI_PHASE1[1]=''
aWIFI_PHASE2[1]=''
aWIFI_CERT[1]=''
```
2. Change `diepi.txt`
```
AUTO_SETUP_LOCALE=C.UTF-8
AUTO_SETUP_KEYBOARD_LAYOUT=us
AUTO_SETUP_TIMEZONE=Europe/Berlin
AUTO_SETUP_NET_ETHERNET_ENABLED=0
AUTO_SETUP_NET_WIFI_ENABLED=1
AUTO_SETUP_NET_ETH_FORCE_SPEED=0
AUTO_SETUP_NET_WIFI_COUNTRY_CODE=DE
AUTO_SETUP_NET_USESTATIC=0
AUTO_SETUP_NET_STATIC_IP=192.168.0.100
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
AUTO_SETUP_NET_STATIC_DNS=9.9.9.9
AUTO_SETUP_NET_HOSTNAME=PiZero
AUTO_SETUP_SWAPFILE_SIZE=1569
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
AUTO_SETUP_HEADLESS=0
AUTO_UNMASK_LOGIND=0
AUTO_SETUP_CUSTOM_SCRIPT_EXEC=0
AUTO_SETUP_SSH_SERVER_INDEX=-1
AUTO_SETUP_FILE_SERVER_INDEX=0
AUTO_SETUP_LOGGING_INDEX=-1
AUTO_SETUP_RAMLOG_MAXSIZE=50
AUTO_SETUP_WEB_SERVER_INDEX=-2
AUTO_SETUP_AUTOSTART_TARGET_INDEX=0
AUTO_SETUP_AUTOSTART_LOGIN_USER=root
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_GLOBAL_PASSWORD=dietpi
SURVEY_OPTED_IN=0
CONFIG_CPU_GOVERNOR=ondemand
CONFIG_CPU_ONDEMAND_SAMPLE_RATE=25000
CONFIG_CPU_ONDEMAND_SAMPLE_DOWNFACTOR=40
CONFIG_CPU_USAGE_THROTTLE_UP=50
CONFIG_CPU_MAX_FREQ=Disabled
CONFIG_CPU_MIN_FREQ=Disabled
CONFIG_CPU_DISABLE_TURBO=0
CONFIG_PROXY_ADDRESS=MyProxyServer.com
CONFIG_PROXY_PORT=8080
CONFIG_PROXY_USERNAME=
CONFIG_PROXY_PASSWORD=
CONFIG_BOOT_WAIT_FOR_NETWORK=2
CONFIG_G_CHECK_URL_TIMEOUT=10
CONFIG_G_CHECK_URL_ATTEMPTS=2
CONFIG_CHECK_CONNECTION_IP=9.9.9.9
CONFIG_CHECK_DNS_DOMAIN=dns9.quad9.net
CONFIG_CHECK_DIETPI_UPDATES=1
CONFIG_NTP_MODE=2
CONFIG_SERIAL_CONSOLE_ENABLE=1
CONFIG_SOUNDCARD=none
CONFIG_LCDPANEL=none
CONFIG_ENABLE_IPV6=1
CONFIG_PREFER_IPV4=1
CONFIG_APT_RASPBIAN_MIRROR=http://raspbian.raspberrypi.org/raspbian/
CONFIG_APT_DEBIAN_MIRROR=https://deb.debian.org/debian/
CONFIG_NTP_MIRROR=debian.pool.ntp.org
SOFTWARE_EMONHUB_APIKEY=
SOFTWARE_VNCSERVER_WIDTH=1280
SOFTWARE_VNCSERVER_HEIGHT=720
SOFTWARE_VNCSERVER_DEPTH=16
SOFTWARE_VNCSERVER_DISPLAY_INDEX=1
SOFTWARE_VNCSERVER_SHARE_DESKTOP=0
SOFTWARE_OWNCLOUD_NEXTCLOUD_USERNAME=admin
SOFTWARE_OWNCLOUD_DATADIR=/mnt/dietpi_userdata/owncloud_data
SOFTWARE_NEXTCLOUD_DATADIR=/mnt/dietpi_userdata/nextcloud_data
SOFTWARE_WIFI_HOTSPOT_SSID=DietPi-HotSpot
SOFTWARE_WIFI_HOTSPOT_KEY=dietpihotspot
SOFTWARE_WIFI_HOTSPOT_CHANNEL=3
SOFTWARE_XORG_DPI=96
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com
SOFTWARE_HOMEASSISTANT_APT_DEPS=
SOFTWARE_HOMEASSISTANT_PIP_DEPS=
DEV_GITBRANCH=master
DEV_GITOWNER=MichaIng
```
3. Boot

#### Expected behaviour
It should boot, do all the first run things and at least join the wifi to be accessible via ssh

#### Actual behaviour
It starts up and gets stuck like this:

[image]

When I then log in I get this:

[image]

There is no network link but `lo`.

MichaIng, 2020-09-02:
Many thanks for your report. I ran some tests and indeed the network check can succeed even if not route exists. Fixed this for next release: https://github.com/MichaIng/DietPi/commit/729dd7bb750876c5334f3e1f5990f63fc462a936

However, this does not explain why you have no network device at all.

What I am indeed missing is the output of the `dietpi-firstboot.service`, which should run between `DietPi-Set_CPU` and `DietPi-Boot` can you please check if this:
```
cat /var/tmp/dietpi/logs/dietpi-firstboot.log
systemctl status dietpi-firstboot
cat /etc/network/interfaces
```

tcurdt, 2020-09-02:
Many thanks for the quick feedback!
The log shows that wireless is disabled `wifimodules (disabled)` - but I don't get why 

[image]

[image]

Joulinar, 2020-09-02:
according your screen shot, it's adding `dtoverlay=disable-wifi` to `/boot/config.txt`. Removing this will enable WiFi. Somehow strange. As I can see you have set following, which should enable WiFi 🤔 

```
AUTO_SETUP_NET_ETHERNET_ENABLED=0
AUTO_SETUP_NET_WIFI_ENABLED=1
```

MichaIng, 2020-09-02:
Yeah strange, this is the check we do:
```
grep -cm1 '^[[:blank:]]*AUTO_SETUP_NET_WIFI_ENABLED=1' /boot/dietpi.txt
```
This should report `1` and WiFi should hence be enabled.

However not a too big problem, for a single case at least, please do the following:
```
/boot/dietpi/func/dietpi-set_hardware wifimodules enable
sed -i "/allow-hotplug wlan/c\allow-hotplug wlan0" /etc/network/interfaces
sed -i "/allow-hotplug eth/c\#allow-hotplug eth0" /etc/network/interfaces
/boot/dietpi/func/dietpi-wifidb 1
/boot/dietpi/func/dietpi-set_hardware wificountrycode DE
reboot
```
On next login, setup should complete.

tcurdt, 2020-09-02:
> according your screen shot, it's adding `dtoverlay=disable-wifi` to `/boot/config.txt`. Removing this will enable WiFi. Somehow strange. As I can see you have set following, which should enable WiFi 🤔
> 
> ```
> AUTO_SETUP_NET_ETHERNET_ENABLED=0
> AUTO_SETUP_NET_WIFI_ENABLED=1
> ```

Hence my confusion.

>    grep -cm1 '^[[:blank:]]*AUTO_SETUP_NET_WIFI_ENABLED=1' /boot/dietpi.txt
>
> This should report 1 and WiFi should hence stay enabled and be configured.

I just tried that. It does report 1.

> However not a too big problem, for a single case at least, please do the following:

Thanks for the workaround - but I am here to help fixing it :)
The goal is to have a config I can keep using.

MichaIng, 2020-09-02:
I'll try to replicate tomorrow after I updated our image. The check didn't change but just to rule out any other fixed issues and concentrate testing on current code 😉.

tcurdt, 2020-09-02:
Where is script located that does the check?

MichaIng, 2020-09-02:
`/var/lib/dietpi/services/dietpi-firstboot.bash`

tcurdt, 2020-09-02:
I added some debugging but it seems a reboot did not re-run the first-run script.
When I ran it manually I was able to see it doing the right thing - but it cannot find the `nl80211` module.

A manual `modprobe nl80211` did not work either.
Neither did `find /lib/modules -name nl8*` give any results.

MichaIng, 2020-09-02:
Probably it is builtin instead a module. Yes firstboot script should not be re-triggered, but the commands above should have been all that is required. After login, firstrun update and installs run instead.

tcurdt, 2020-09-03:
@MichaIng did you arrive at any further suggestions yet?

MichaIng, 2020-09-03:
I should be able to create the new image tomorrow to start the replication attempt with.

tcurdt, 2020-09-09:
@MichaIng I don't want to come across pushy - just checking in. Anything else I can help with in this matter?

MichaIng, 2020-09-10:
New image uploaded, after 1-2 positive feedback I'll move it to stable downloads: https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv6-Buster.7z

tcurdt, 2020-09-10:
Flashed, copied over the two files, booted up - and all works.
Still wondering what was wrong with the other setup, but all seems fine now. Thanks!!

closed 2020-09-10 by tcurdt

Joulinar, 2020-09-10:
probably it has something to do with latency. In normal case all goes well, but in critical environments it failed.

MichaIng, 2020-09-10:
Yeah not sure, or something changed the way the DHCP lease is applied, although cannot see any recent related package update.