MichaIng/DietPi #3754: Move FS to USB

davidjmorin, 2020-09-03:
Move to file system does not work.  Pi does not boot at all.  

[image]
[image]

davidjmorin, 2020-09-03:
This is a fresh install of dietpi on rp4.  No installed programs.  I then went to dietpi-drive_manager to move FS to SSD.  Formatted correctly and upon reboot this is all I see.

Joulinar, 2020-09-03:
Hi,

Many thanks for your message. The 2nd screen message is normal and can be ignored. Pls can you let us know which file system you moved to SSD?

MichaIng, 2020-09-03:
For better debugging please enable more boot logs by removing the `quite` flag from `cmdline.txt`, if you have access to the SD card boot partition from external system.

And although we want to investigate and in case fix the issue, generally on RPi3+ and up I recommend to flash the image completely and directly to the USB drive, so you have full USB boot and do not need to copy data on the running system.

davidjmorin, 2020-09-03:
Ok so I have it booted now. Process I used was to first boot rpi4 with official raspberry image. Do all the updates and install new eeprom. Removed SD card and the imaged ssd usb with diet pi. Plugged into USB3 and it did not boot at all for a while. Then booted in emergency mode. I moved usb to usb2 and it booted up fine. Seems USB3 does not work. This is a 4gb version.

mrbluecoat, 2020-09-08:
So is booting dietpi from a USB 3.1 drive not supported? I was thinking of doing something like this myself.

davidjmorin, 2020-09-08:
> So is booting dietpi from a USB 3.1 drive not supported? I was thinking of doing something like this myself.

No it is supported.  Just follow those steps unless you have already setup rasbian to do it.  After I did these steps it works flawlessly.  Except soft reboots do not work.

MichaIng, 2020-09-08:
Again a Seagate drive that requires a USB quirk, I think this can be added statically to the RPi kernel, else er can apply this via modprobe config.

MichaIng, 2020-09-17:
Issue reported at RPi kernel repo: https://github.com/raspberrypi/linux/issues/3855
I could not reliably find out how/where to add this to the source code directly, let's see if there is a more efficient way of sending a PR directly in the future.

MichaIng, 2020-09-28:
Okay RPi devs do not maintain or care UAS blacklists. Also a good argument is that firmware updates (respectively controllers with same ID but shipped with newer firmware) might provide UAS support for drives, then the blacklist is rendered obsolete and the quirk decreases performance, hence it is practically not possible to have a blacklist that is always right.

What we might want to do instead:
- Add option to disable UAS, if it is a kernel module (on RPi it is!).
- Add an option to add a UAS blacklist quirk for an attached drive if we have a reliable way to alter the cmdline (RPi, x86, Odroids at least).