MichaIng/DietPi #3756: Cannot create swap file on ZeroPi

auanasgheps, 2020-09-03:
### Creating a bug report/issue

I've done a fresh installation of DietPi on a ZeroPi board, which is running off a 64 GB microSD formatted with BalenaEtcher.

When installing PiHole, this error message appeared

[image]

I've also tried dietpi-drive_manager and set a swap there, getting the exact same message.

#### Required Information
- DietPi version | 9.32
- Distro version | buster
- Kernel version | Linux DietPi 5.7.15-sunxi #20.08
- SBC model | ZeroPi (armv7l)
- Power supply used | Asus Router USB3 port 
- SDcard used | Netac 64G U3

#### Additional Information (if applicable)
- Software title | PiHole
- Was the software title installed freshly or updated/migrated? Entirely new install
- Can this issue be replicated on a fresh installation of DietPi? That's an actually new install

- Bug report ID | fc964d38-04ab-414c-94a3-54be66010cb1

#### Steps to reproduce

1. Install DietPi on ZeroPi
2. Install PiHole
3. Go trough the wizard
4. Error about swap occurs

#### Expected behaviour

Swap file should be created

#### Actual behaviour

Swap file cleis not created

#### Extra details

No swap!
[image]

Joulinar, 2020-09-03:
Hi,

many thanks for your report. Pls can you try following

`/boot/dietpi/func/dietpi-set_swapfile 1`

auanasgheps, 2020-09-03:
Nope, it produced the exact same behaviour.

[image]

Joulinar, 2020-09-03:
can you scroll up on the error message. Does it show anything?

For me this is working fine 🤔 
```
root@DietPi3:~# /boot/dietpi/func/dietpi-set_swapfile 1

 DietPi-Set_swapfile
─────────────────────────────────────────────────────
 Mode: Applying 1 /var/swap

[ INFO ] DietPi-Set_swapfile | Disabling and deleting all existing swap files
[  OK  ] DietPi-Set_swapfile | swapoff -a
removed '/var/swap'
[  OK  ] DietPi-Set_swapfile | Free space check: path=/var/ | available=13220 MiB | required=1073 MiB
[  OK  ] DietPi-Set_swapfile | Generating new swap space
[ INFO ] DietPi-Set_swapfile | Size = 1073 MiB
[ INFO ] DietPi-Set_swapfile | Path = /var/swap
[  OK  ] DietPi-Set_swapfile | fallocate -l 1073M /var/swap
[  OK  ] DietPi-Set_swapfile | mkswap /var/swap
[  OK  ] DietPi-Set_swapfile | chmod 600 /var/swap
[  OK  ] DietPi-Set_swapfile | swapon /var/swap
[  OK  ] DietPi-Set_swapfile | eval echo '/var/swap none swap sw' >> /etc/fstab
[  OK  ] DietPi-Set_swapfile | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_SIZE=1073
[  OK  ] DietPi-Set_swapfile | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
[ INFO ] DietPi-Set_swapfile | Setting /tmp tmpfs size: 1024 MiB
[  OK  ] DietPi-Set_swapfile | mount -o remount /tmp
root@DietPi3:~#
```

auanasgheps, 2020-09-03:
[image]

Joulinar, 2020-09-03:
ok the swap file get's deleted and a new one created. However activation fails.

pls can you do `swapon /var/swap`

auanasgheps, 2020-09-03:
Well, I cant.
```
root@DietPi:~# swapon /var/swap
swapon: /var/swap: swapon failed: Invalid argument
```

Joulinar, 2020-09-03:
what gives `swapon -a`

auanasgheps, 2020-09-03:
Nothing:

```
root@DietPi:~# swapon -a
root@DietPi:~# swapon -s
root@DietPi:~#

```

Joulinar, 2020-09-03:
ok let's check if the swap file already exist

auanasgheps, 2020-09-03:
Yes, something is showing up!

```
root@DietPi:~# ls -la /var |grep swap
-rw-------  1 root root  1632632832 Sep  4 00:15 swap
root@DietPi:~#
```

Joulinar, 2020-09-03:
ok let's try it this way. Add the swap location to `/etc/fstab` and activate it

```
eval echo '/var/swap none swap sw' >> /etc/fstab
swapon -a
```

and let's check
`swapon -s`

auanasgheps, 2020-09-03:
No luck this time 
```
root@DietPi:~# eval echo '/var/swap none swap sw' >> /etc/fstab
root@DietPi:~# swapon -a
swapon: /var/swap: swapon failed: Invalid argument
swapon: /var/swap: swapon failed: Invalid argument
swapon: /var/swap: swapon failed: Invalid argument
swapon: /var/swap: swapon failed: Invalid argument
swapon: /var/swap: swapon failed: Invalid argument
swapon: /var/swap: swapon failed: Invalid argument
root@DietPi:~#
```

Joulinar, 2020-09-04:
strange that you still get `Invalid argument`. 🤔 

Maybe @MichaIng has an idea

MichaIng, 2020-09-04:
Although on a fresh install its `ext4` but just to be sure, which file system is it?
```
df -T /var
```

`ext4` definitely support `fallocate`, but to be sure lets see if pre-creating the file via `dd` works instead in your case:
```
rm /var/swap
dd if=/dev/zero of=/var/swap bs=1M count=1557
chmod 600 /var/swap
mkswap /var/swap
swapon /var/swap
```

If that still fails, try to reinstall the `mount` package which ships the binary. Probably it is broken somehow:
```
apt install  --reinstall mount
```

auanasgheps, 2020-09-04:
It's ext4 indeed.

```
root@DietPi:~# df -T /var
Filesystem     Type 1K-blocks    Used Available Use% Mounted on
/dev/mmcblk0p1 ext4  59144097 2230247  56909754   4% /
root@DietPi:~#
```
I've executed your commands and the procedure has worked! 

```
root@DietPi:~# free -m
              total        used        free      shared  buff/cache   available
Mem:            491          50         114           3         326         425
Swap:          1556           0        1556
root@DietPi:~#
```

So, there's something wrong with the script that creates the swap?

MichaIng, 2020-09-05:
Did it work already before reinstalling the `mount` package?

> So, there's something wrong with the script that creates the swap?

If really `fallocate` is the issue, then there is something not right with your file system as `ext4` is perfectly known to support `fallocate` with swap. Could you please check for possibly related kernel errors:
```
dmesg -l emerg,alert,crit,err
```

auanasgheps, 2020-09-06:
Yesactly I did not reinstall the package. 
I ran the following and worked:

```
rm /var/swap
dd if=/dev/zero of=/var/swap bs=1M count=1557
chmod 600 /var/swap
mkswap /var/swap
swapon /var/swap
```

This is what I see if I run dsmeg

```
[    7.746494] systemd-fstab-generator[214]: Failed to create unit file /run/systemd/generator/var-swap.swap, as it already exists. Duplicate entry in /etc/fstab?
[    7.761109] systemd-fstab-generator[214]: Failed to create unit file /run/systemd/generator/var-swap.swap, as it already exists. Duplicate entry in /etc/fstab?
[    7.775693] systemd-fstab-generator[214]: Failed to create unit file /run/systemd/generator/var-swap.swap, as it already exists. Duplicate entry in /etc/fstab?
[    7.790340] systemd-fstab-generator[214]: Failed to create unit file /run/systemd/generator/var-swap.swap, as it already exists. Duplicate entry in /etc/fstab?
[    7.804952] systemd-fstab-generator[214]: Failed to create unit file /run/systemd/generator/var-swap.swap, as it already exists. Duplicate entry in /etc/fstab?
[    7.824624] systemd[210]: /lib/systemd/system-generators/systemd-fstab-generator failed with exit status 1.
```

Additonally, since I've done nothign with it, I can wipe the card, start all over and see if the problem occurs again.

MichaIng, 2020-09-07:
Hmm, although this should not be related, but are there really multiple swap entries in fstab? `cat /etc/fstab`

Joulinar, 2020-09-07:
maybe they are entered by trying to fix the error 🤔 

https://github.com/MichaIng/DietPi/issues/3756#issuecomment-686809084

MichaIng, 2020-09-07:
Ah yes that should be it. Means to resolve that:
```
sed -i '\|/var/swap|d' /etc/fstab
echo '/var/swap none swap sw' >> /etc/fstab
```

auanasgheps, 2020-09-07:
This is what I get.

[image]

MichaIng, 2020-09-07:
Okay, run the two commands above or remove the last 5 lines manually 😉.

auanasgheps, 2020-09-10:
Redone the microSD so the install from scratch.
The issue appears again during DietPi install.

Here are some commands:

```
root@DietPi:~# df -T /var
Filesystem     Type 1K-blocks    Used Available Use% Mounted on
/dev/mmcblk0p1 ext4  59144097 1160225  57979776   2% /
root@DietPi:~# free -m
              total        used        free      shared  buff/cache   available
Mem:            491          49         378           3          63         425
Swap:             0           0           0
root@DietPi:~# cat /etc/fstab
# Please use "dietpi-drive_manager" to setup mounts
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1024M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf (VirtualBox shared folder), gluster, bind mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAPFILE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
UUID=2046b9b3-c939-46d4-8fa6-3be42b9609c7 / auto noatime,lazytime,rw 0 1
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777
root@DietPi:~#
```

MichaIng, 2020-09-10:
Very strange, probably SD card related? Since really this is the first report ever that this did not work. But its probably okay to add a fallback, even if it's the only case. If all previous steps succeeded but `swapon` fails, then trying `dd` would be the first attempt to fix it anyway. Since it's quite a significant large write and slow + creates wear on SD card, it should should still be avoided if not really necessary.

MichaIng, 2020-09-15:
How we're getting closer, a bug or regression in Linux 5.7? https://bugs.archlinux.org/task/66979

MichaIng, 2020-09-16:
Verified:
```
2020-09-16 15:20:50 root@VM-Bullseye:~# swapon /mnt/external/.swapfile
swapon: /mnt/external/.swapfile: swapon failed: Invalid argument
2020-09-16 15:20:58 root@VM-Bullseye:~# uname -a
Linux VM-Bullseye 5.7.0-3-amd64 #1 SMP Debian 5.7.17-1 (2020-08-23) x86_64 GNU/Linux
```
So it's a Linux 5.7 issue.

auanasgheps, 2020-09-16:
Nice find :)

MichaIng, 2020-09-16:
Especially nasty is that it even effects swap files that worked fine before. So you upgrade the kernel (on Armbian this is done by simple `apt upgrade`) and on reboot the swap file is broken.

Patch for the `ext4` module (yes `ext4` only) is there: https://lore.kernel.org/linux-fsdevel/CAGqt0zw92aOxA2edGcugqzzQJFbjuAOoDeFirKkx1EiRR9rUdQ@mail.gmail.com/
But it was to be send end of August while until now no news on that mail list.

I'll keep an eye on this. If there is no patch merged (+backported to Linux 5.7) until DietPi v6.33 release, I'll make our script use `dd` for `ext4` temporarily.

MichaIng, 2020-09-24:
I added the most reliable workaround to original post:
#### Workaround
```
sed -i 's/f2fs/ext4/' /boot/dietpi/func/dietpi-set_swapfile
/boot/dietpi/func/dietpi-set_swapfile
```
This forces `dd` file allocation on ext4 file system while allowing (faster) `fallocate` on F2FS, which is possible on newer kernels, so both can be applied in the same turn. It then re-creates the swap file based on already chosen settings.

MichaIng, 2020-09-28:
Issues seems to have been fixed: https://lkml.org/lkml/2020/9/24/810
But I cannot see it in upstream repository yet: https://github.com/torvalds/linux/blob/master/fs/ext4/inode.c
But I'm not sure where patch versions (Linux 5.7.x and 5.8.y) are actually hosted and where/how the procedure of merging patches works 😄.

auanasgheps, 2020-09-28:
What an honor to see the main Linux repository from Linus. I didn't know was on Github 😁

MichaIng, 2020-09-28:
To be true, the GitHub repo is a read-only mirror only, development is done via mailing lists, which is the reason I lack the ability to track where what has been send to and merged: https://github.com/torvalds/linux/pull/805#issuecomment-593375542

Not sure if this is just because it was always like that or if there are still real reasons to develop via mailing lists. For me it makes it in-transparent, hard to search through etc, but likely it is a question what one is used to (me used to use GitHub, GitLab or similar development platform 😄).

MichaIng, 2020-10-02:
Solution implemented: https://github.com/MichaIng/DietPi/commit/752f6986c6cd381ae112cb353bfa17ef78bf2304
This is not only a workaround but a good generic method as it will now always try to use `fallocate` first, which e.g. works on F2FS and XFS on newer kernels but we skipped it as it failed on older ones. If any step until including swapon fails, it will try again via `dd`.
Changelog: https://github.com/MichaIng/DietPi/commit/530cf7137122df7bf7af5da3a2d0c8297770b6c8

closed 2020-10-17 by MichaIng