MichaIng/DietPi #3757: 2nd ethernet

symbios24, 2020-09-04:
Hello i have a Raspberry pi model and i get internet from an ethernet cable i putted a usb ethernet to give internet to a second raspberry pi but it does not sems to give internet im using latest dietpi as a hotspot  but thats irrelevant to the issue
The usb ethernet seems to work the leds are on
But i dont have internet on the second Rpi
Any help?

Joulinar, 2020-09-04:
Hi,

many thanks for your question. I guess you would need to setup your device to bridge Ethernet connect between eth0 and your USB Ethernet.

https://wiki.debian.org/BridgeNetworkConnections

There are a couple of guides on the web how to archive this. Basically it's about to install and setup `bridge-utils`.

symbios24, 2020-09-04:
Strange i had another pi with desktop connected and a second usb ethernet connected to give in another pi and was working great without any setting now with dietpi does not work automatic

MichaIng, 2020-09-04:
There are a few possibilities and if this was done automatically on another OS, then I guess `network-manager` or a similar high level network setup utility was installed. You can install that on DietPi as well and see if it does the job, but our tools indeed can't do that yet, planned as feature as part of this project: https://github.com/MichaIng/DietPi/projects/1

symbios24, 2020-09-05:
Nevermind i connected a switch and everything worked thanks for the help

Joulinar, 2020-09-05:
I'm going to close the topic if thinks are sorted. Feel free to reopen if needed.

closed 2020-09-05 by Joulinar

symbios24, 2020-09-05:
Any change that you can add in the wifi adapter settings an option to set the max dbm depending from the country code ??

MichaIng, 2020-09-07:
That looks like a doable solution indeed: https://hackingvision.com/2017/02/18/increasing-wifi-tx-power-signal-strength-in-linux/

So default seems to be 20 dBm. Wile this is as well the max allowed in most countries and iw/iwconfig seem to deny raising it based on country code, when using a different country code (legally or illegally, out of our control of course) that allows higher values, it can be raised.

Since we cannot maintain a database, we either need a way to get the max dBm value via iw/iwconfig or some web API, otherwise it would be trial and error?