MichaIng/DietPi #3758: No space left on /boot

oscarmarsh, 2020-09-04:
- Date           | Fri Sep  4 13:25:25 BST 2020
- Bug report     | 1d834674-d3c5-4cf4-9e09-51dcbd10a413
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | Linux seetwit 4.14.176+ #1 SMP PREEMPT Tue Apr 14 20:02:49 UTC 2020 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | cp -Rf DietPi-master/dietpi /boot/
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. login as root
2. dietpi-update
#### Expected behaviour:

System updates

#### Actual behaviour:

See below

#### Extra details:

#### Additional logs:
```
cp: error writing '/boot/dietpi/patch_file': No space left on device
```

Joulinar, 2020-09-04:
Hi,

many thanks for your request. As far as I can see, your created a post on our forum already https://dietpi.com/phpbb/viewtopic.php?f=11&t=8057

There is no need to create it twice. 😉 

pls post output of `df -h`

MichaIng, 2020-09-04:
I just checked the bug report, indeed `/boot` is full:
```
/dev/mmcblk1p1 vfat       71M   71M  2.0K 100% /boot
```
I recognised, especially on the older Meveric images, that old kernel/initramfs files were left in place by times. He switched to a different build system (from Hardkernel to Debian) which as well removes obsolete files more reliably, but probably those from the old build system can be left in place.

@oscarmarsh 
Please paste:
```
ls -Al /boot
```

Joulinar, 2020-09-04:
what about extending BootFS via GParted? It's 71MB only?

oscarmarsh, 2020-09-04:
> I just checked the bug report, indeed `/boot` is full:
> 
> ```
> /dev/mmcblk1p1 vfat       71M   71M  2.0K 100% /boot
> ```
> 
> I recognised, especially on the older Meveric images, that old kernel/initramfs files were left in place by times. He switched to a different build system (from Hardkernel to Debian) which as well removes obsolete files more reliably, but probably those from the old build system can be left in place.
> 
> @oscarmarsh
> Please paste:
> 
> ```
> ls -Al /boot
> ```
> 
> **EDIT: Ah an old kernel package is installed as well:**
> 
> ```
> ii  linux-image-4.14.66+                4.14.66-20180823-xu4              armhf        Linux kernel, version 4.14.66+
> ```
> 
> While there might be other obsolete files, running the following command should solve the issue already:
> 
> ```
> apt purge linux-image-4.14.66+
> ```
> 
> Then rerun the update:
> 
> ```
> dietpi-update 1
> ```

First the output of ls -Al  /boot

```
total 50534
-rwxr-xr-x 1 root root    5447 May 13 13:50 boot.ini
-rwxr-xr-x 1 root root  151274 Apr 14 21:05 config-4.14.176+
-rwxr-xr-x 1 root root  151102 Aug 23  2018 config-4.14.66+
drwxr-xr-x 4 root root    4096 Sep  4 13:20 dietpi
-rwxr-xr-x 1 root root    6285 Sep 10  2018 dietpi-README.md
-rwxr-xr-x 1 root root   11346 Jul  6 14:23 dietpi.txt
-rwxr-xr-x 1 root root   56387 Apr 14 21:09 exynos5422-odroidhc1.dtb
-rwxr-xr-x 1 root root   63517 Apr 14 21:09 exynos5422-odroidxu3.dtb
-rwxr-xr-x 1 root root   62462 Apr 14 21:09 exynos5422-odroidxu3-lite.dtb
-rwxr-xr-x 1 root root   63484 Apr 14 21:09 exynos5422-odroidxu4.dtb
-rwxr-xr-x 1 root root   63628 Apr 14 21:09 exynos5422-odroidxu4-kvm.dtb
drwxr-xr-x 2 root root    2048 Aug 11 14:09 .fseventsd
-rwxr-xr-x 1 root root 5723810 Aug 11 12:37 initrd.img-4.14.176+
-rwxr-xr-x 1 root root 5716943 May 13 13:48 initrd.img-4.14.66+
drwxr-xr-x 3 root root    2048 Aug 11 14:09 .Spotlight-V100
-rwxr-xr-x 1 root root 2704214 Apr 14 21:05 System.map-4.14.176+
-rwxr-xr-x 1 root root 2695982 Aug 23  2018 System.map-4.14.66+
-rwxr-xr-x 1 root root 5723645 Aug 11 12:36 uInitrd
-rwxr-xr-x 1 root root 5723645 Aug 11 12:36 uInitrd-4.14.176+
-rwxr-xr-x 1 root root 5717007 May 13 15:48 uInitrd-4.14.66+
-rwxr-xr-x 1 root root 5701416 Apr 14 21:05 vmlinuz-4.14.176+
-rwxr-xr-x 1 root root 5677560 Aug 23  2018 vmlinuz-4.14.66+
-rwxr-xr-x 1 root root 5701416 May 13 15:48 zImage
```

Next I purged the old kernel package...

and ran the update again... 

and it completed. Thanks all. Will look into making the partition a wee bit bigger too. :)

closed 2020-09-04 by Joulinar

MichaIng, 2020-09-04:
The `/boot` partition cannot be easily expended since the root partition starts directly afterwards, similar issue than with older RPi systems when RPi4 was released with its additional kernel files which filled `/boot` until the installer started to skip RPi4 files when space is too low 😄. Yeah moving root is possible via gparted indeed, if one has a desktop/Xserver anyway.

The 70 MiB are indeed a bid low here if two concurrent installed kernel packages are too much already, especially since the newer kernel does not overwrite the older one automatically (like on RPi). I just checked our new Debian Buster images and luckily Meveric increased the `/boot` partition size to 256 MiB, again just like it was done on RPi. Kinda many parallels here 😄:
[image]

oscarmarsh, 2020-09-04:
Thanks for the clarification. As I'm running my DietPi headless, what would you suggest I do about enlarging /boot, whether by moving root or migrating my system to one of the new images? I want something straightforward without having to run any of the installation processes for my dietpi- and self-installed-software.

Joulinar, 2020-09-04:
AHH I missed it. It's still a Stretch system. Indeed, going to Buster would be a good option. More future proof than tweaking around an old Stretch image.