MichaIng/DietPi #3759: Can't install or update.

josh7574, 2020-09-04:
#### Details:
- Date           | Fri Sep  4 23:52:48 BST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-atheros_20190114-1+rpt8_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20190114-1+rpt8_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-iwlwifi_20190114-1+rpt8_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-misc-nonfree_20190114-1+rpt8_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-realtek_20190114-1+rpt8_all.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/d/dbus/libdbus-1-3_1.12.20-0+deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/bind9/libisc-export1100_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/bind9/libdns-export1104_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/j/json-c/libjson-c3_0.12.1+ds-2+deb10u1_armhf.deb  Could not resolve 'raspbian.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200819-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20200819-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20200819-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20200819-1_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-sys-mods/raspberrypi-sys-mods_20200812_armhf.deb  Could not resolve 'archive.raspberrypi.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2020-09-04:
Hi,

many thanks for your message. Basically you have issues to resolve hoste name for  `archive.raspberrypi.org`

`Could not resolve 'archive.raspberrypi.org'`

Do you use an own DNS Server like PiHole? Or a global DNS? Maybe it's just a temp issue.

josh7574, 2020-09-04:
thanks, I was using cloudflare, 1.1.1.3 and 1.0.0.3 on my router dns.

josh7574, 2020-09-04:
i've now changed them to the standard cloudflare 1.1.1.1 and 1.0.0.1

Joulinar, 2020-09-04:
are you able to resolve the host name `archive.raspberrypi.org` from another device on your network?

josh7574, 2020-09-05:
I can ping archive.raspberrypi.org from Pi just fine.

josh7574, 2020-09-05:
sorry, I'm new to this not quite sure what you mean by resolve hostname from another device? thanks

Joulinar, 2020-09-05:
If you are able to ping, you should be able to complete the update

josh7574, 2020-09-05:
strange, still not working though. :(

MichaIng, 2020-09-05:
Can you try the following:
```
ping -c 1 raspbian.raspberrypi.org
ping -c 1 archive.raspberrypi.org
wget --spider http://raspbian.raspberrypi.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb
wget --spider https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200819-1_armhf.deb
```
What is strange is that `apt update` obviously succeeds while only on actual package download it fails, and not only for raspbian.raspberrypi.org but archive.raspberrypi.org as well.

josh7574, 2020-09-05:
thanks, I've just pasted those in and this is what I got.

root@DietPi:~# ping -c 1 raspbian.raspberrypi.org
PING mirrordirector.raspbian.org (93.93.128.193) 56(84) bytes of data.
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=1 ttl=56 time=12.4 ms

--- mirrordirector.raspbian.org ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 12.351/12.351/12.351/0.000 ms
root@DietPi:~# ping -c 1 archive.raspberrypi.org
PING lb.raspberrypi.org (46.235.230.122) 56(84) bytes of data.
64 bytes from 46.235.230.122 (46.235.230.122): icmp_seq=1 ttl=56 time=12.3 ms

--- lb.raspberrypi.org ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 12.320/12.320/12.320/0.000 ms
root@DietPi:~# wget --spider http://raspbian.raspberrypi.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb
Spider mode enabled. Check if remote file exists.
--2020-09-05 14:01:37--  http://raspbian.raspberrypi.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb [following]
Spider mode enabled. Check if remote file exists.
--2020-09-05 14:01:37--  http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb
Resolving www.mirrorservice.org (www.mirrorservice.org)... 212.219.56.184, 2001:630:341:12::184
Connecting to www.mirrorservice.org (www.mirrorservice.org)|212.219.56.184|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1047252 (1023K) [application/x-debian-package]
Remote file exists.

root@DietPi:~# wget --spider https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200819-1_armhf.deb
Spider mode enabled. Check if remote file exists.
--2020-09-05 14:02:01--  https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200819-1_armhf.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 93.93.130.212, 46.235.230.122, 93.93.135.118, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|93.93.130.212|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 73181200 (70M) [application/x-debian-package]
Remote file exists.

i tried to update again and still not working

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Mich                                                                             aIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/                                                                             MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.30.0
[ INFO ] DietPi-Update | Latest version  : v6.32.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=25505 MiB | requir                                                                             ed=100 MiB
[ SUB1 ] DietPi-Services > stop
[ INFO ] DietPi-Services | skip : cron (due to mask)

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patch file
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
[... 17 lines cut ...]
 DietPi v6.30.0 : Update available
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.1.5 (eth0)
[ INFO ] DietPi-Login | Checking network connectivity, please wait...  (1/2)
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=56 time=12.9 ms

--- 9.9.9.9 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 12.928/12.928/12.928/0.000 ms
[  OK  ] DietPi-Login | Checking network connectivity
[  OK  ] DietPi-Login | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ INFO ] DietPi-Update | Concurrent execution of DietPi-Update detected: Retry (                                                                             0/5), please wait...
[ INFO ] DietPi-Update | Concurrent execution of DietPi-Update detected: Retry (                                                                             1/5), please wait...
[ INFO ] DietPi-Update | Concurrent execution of DietPi-Update detected: Retry (                                                                             2/5), please wait...
[ INFO ] DietPi-Update | Concurrent execution of DietPi-Update detected: Retry (                                                                             3/5), please wait...
[... cut ...]                   4/5), please wait...
[FAILED] DietPi-Update | Cancelled DietPi-Update due to concurrent execution
 ─────────────────────────────────────────────────────
 DietPi v6.30.0 : Update available
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.1.5 (eth0)
[  OK  ] DietPi-Login | Desired setting in /boot/dietpi.txt was already set: AUT                                                                             O_SETUP_AUTOMATED=0
root@DietPi:/tmp/DietPi-Update#

josh7574, 2020-09-05:
OK so it's finally updating!! The only thing I have done differently was log into my unifi controller and under clients section, I unselect "Use fixed IP address" for the pi. thanks for all your help @MichaIng and @Joulinar it's much appreciated!!

Joulinar, 2020-09-05:
Indeed strange. Maybe changing to Google DNS did the trick? But good that it is solved now. I'm going to close the topic. If needed, feel free to reopen.

closed 2020-09-05 by Joulinar

cdlenfert, 2020-09-14:
I'm having similar issues when running `apt-update`.
```
root@DietPi:~# apt update
Err:1 https://archive.raspberrypi.org/debian buster InRelease                                                        
  Could not resolve 'archive.raspberrypi.org'
Err:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                      
  Could not resolve 'raspbian.raspberrypi.org'
Err:3 https://downloads.plex.tv/repo/deb public InRelease                    
  Could not resolve 'downloads.plex.tv'
Reading package lists... Done       
Building dependency tree       
Reading state information... Done
All packages are up to date.
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Could not resolve 'raspbian.raspberrypi.org'
W: Failed to fetch https://downloads.plex.tv/repo/deb/dists/public/InRelease  Could not resolve 'downloads.plex.tv'
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/buster/InRelease  Could not resolve 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
```
Testing @MichaIng suggestions above:
```
root@DietPi:~# ping -c 1 raspbian.raspberrypi.org
PING mirrordirector.raspbian.org (93.93.128.193) 56(84) bytes of data.
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=1 ttl=56 time=126 ms

--- mirrordirector.raspbian.org ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 125.977/125.977/125.977/0.000 ms
root@DietPi:~# ping -c 1 archive.raspberrypi.org
PING lb.raspberrypi.org (176.126.240.167) 56(84) bytes of data.
64 bytes from 176.126.240.167 (176.126.240.167): icmp_seq=1 ttl=54 time=126 ms

--- lb.raspberrypi.org ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 126.323/126.323/126.323/0.000 ms
root@DietPi:~# wget --spider http://raspbian.raspberrypi.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb
Spider mode enabled. Check if remote file exists.
--2020-09-14 09:05:18--  http://raspbian.raspberrypi.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb [following]
Spider mode enabled. Check if remote file exists.
--2020-09-14 09:05:18--  http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u5_armhf.deb
Resolving mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)... 72.5.72.15, 2606:c680:0:b:3830:34ff:fe66:6663
Connecting to mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)|72.5.72.15|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1047252 (1023K) [application/octet-stream]
Remote file exists.

root@DietPi:~# wget --spider https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200819-1_armhf.deb
Spider mode enabled. Check if remote file exists.
--2020-09-14 09:05:32--  https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200819-1_armhf.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 176.126.240.167, 176.126.240.84, 176.126.240.86, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|176.126.240.167|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 73181200 (70M) [application/x-debian-package]
Remote file exists.
```
`Current version : v6.32.2`
I am running pi-hole with DNS set to my modem's IP.
I thought my modem was using Google servers for DNS, but apparently it was not.
```
root@DietPi:~# cat /etc/resolv.conf
# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#     DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
nameserver 192.168.0.1
nameserver 206.67.220.220
search domain
```
After switching my modem DNS to 8.8.8.8 and 8.8.4.4 I renewed my dhcp lease on the Pi.
```
root@DietPi:~# dhclient -r
root@DietPi:~# dhclient
RTNETLINK answers: File exists
root@DietPi:~# cat /etc/resolv.conf
# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#     DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
nameserver 192.168.0.1
nameserver 8.8.4.4
search domain
```
Now `apt update` runs without any issue.
```
root@DietPi:~# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]                           
Hit:3 https://downloads.plex.tv/repo/deb public InRelease                                                     
Hit:4 https://archive.raspberrypi.org/debian buster InRelease                                                 
Fetched 13.0 MB in 20s (651 kB/s)                                                                                                             
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
```
My internet service provider is CenturyLink.

Joulinar, 2020-09-14:
hmm what IP address your PiHole is running on? because you had configured these 2

```
nameserver 192.168.0.1
nameserver 206.67.220.220
```

Looks like 192.168.0.1 is your router? and the 2nd one from your provider?

Now, you changed to this
```
nameserver 192.168.0.1
nameserver 8.8.4.4
```

Probably DNS resolution on your provider was taking longer than expected and is faster with google now?

cdlenfert, 2020-09-15:
My PiHole runs on the Pi that has an IP of `192.168.0.44`. It uses the router `192.168.0.1` for DNS servers and the router was was defaulting to CenturyLinnk's own DNS servers (giving the `206.67.220.220` address). Changing my router to use Google's DNS was the only change I made. That got everything working for me.

Joulinar, 2020-09-15:
Probably something was not working on your ISP side. Now, using Google DNS fixed it.