MichaIng/DietPi #3760: Update Remote IT to proper

davidjmorin, 2020-09-05:
Remoteit has switched from connectd to remoteit.  Can we get an update for this in the installer?  

```
sudo apt update && sudo apt install remoteit
```

Now with the new instead of using connectd you would use the remoteit command line command.

Joulinar, 2020-09-05:
Hi,

many thanks for your request. As far as I can see, this is applicable for Raspberry OS only. On all other Debian systems, it's still `connectd`.

davidjmorin, 2020-09-05:
> Hi,
> 
> many thanks for your request. As far as I can see, this is applicable for Raspberry OS only. On all other Debian systems, it's still `connectd`.

I have it setup currently on my dietpi.  I manually installed it.  Works great.

Joulinar, 2020-09-05:
Yes because you are on Raspberry Pi, isn't it? On other SBC's it's not available. DietPi is supporting quite a large amount of SBC's. And Raspberry OS is used on Raspberry Pi devices only.

davidjmorin, 2020-09-05:
> Yes because you are on Raspberry Pi, isn't it? On other SBC's it's not available. DietPi is supporting quite a large amount of SBC's. And Raspberry OS is used on Raspberry Pi devices only. Anyway, at the end it's up to @MichaIng to decide if he like implement different scenarios for different devices.

Oh got ya.  Yes that makes more sense.  I noticed it installed the older one so I updated it.

MichaIng, 2020-09-07:
I just checked that all. `connectd` is definitely not the "old" package, their docs is a bid misleading there, it still the official name of the Remot3.it client daemon and the method we use to install it leads to an up-to-date upstream Remot3.it version.

I don't know what is included in the `remoteit` package from Raspberry Pi repo, but it's much more then the `connectd` client itself, as the package size is [13 MiB](http://archive.raspberrypi.org/debian/pool/main/r/remoteit/) compared to [113 KiB](https://github.com/remoteit/installer/releases), so more than **100x larger**. One thing is a command line utility, ~2 MiB in size: https://docs.remote.it/adding-remote.it-to-your-device/using-the-cli
Additionally what the `connectd_installer` can do, it can install some additional tools. and as well uninstall the service, not sure if `connectd_installer` can do that as well as I have no account to proceed from sign in menu, but reviewing the library shell scripts reveals that all sign-in/out, monitoring and connection control and status features are included as well.

Then there is a 11 MiB archive with 35 MiB of Node.js source code called "remoteid_headless". Not sure if it is a web interface hosted on the RPi? Isn't this done through https://remote.it/?

So generally we tend to ship lightweight implementations with DietPi and the RPi repo package ships a bunch of other stuff that is installed where I don't know the benefit of in case of the source code don't even know what it is, not is anything about this documented. The `remoteit` package is trivial to install and automatically replaces the `connectd` package, so there is not much we can do to enhance this install. If we add it, then indeed as additional option for RPi only. @davidjmorin would be great if you could report a bid what those additional tools can do and if there is a real benefit over plain `connectd`.

closed 2020-09-28 by davidjmorin