MichaIng/DietPi #3761: Error with Cuberite

jeffthejeffthejeff, 2020-09-06:
To be honest I don't know if it's anything you can fix, may just be on cuberite's side to update the package. Here's the report:

#### Details:
- Date           | Sun  6 Sep 22:51:39 BST 2020
- Bug report     | 34bbb264-2e65-4422-9d6f-287f90cd00c7
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | Connection test: https://builds.cuberite.org/job/Cuberite%20Linux%20raspi-armhf%20Master/lastSuccessfulBuild/artifact/Cuberite.tar.gz
- Exit code      | 8
- Software title | DietPi-Software

#### Steps to reproduce:

1. You install Cuberite with dietpi-software and wait for the error.
2. You see the error
3. You have reproduced the error!

#### Expected behaviour:

- It should install Cuberite without issues

#### Actual behaviour:

- Cuberite have changed their website to download the builds so the install has thrown an error as it is using an old link.

#### Extra details:

- I am using a slightly outdated version, v6.28.0 and the current one is v6.32.2 but that shouldn't change anything.
#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2020-09-06 22:49:58--  https://builds.cuberite.org/job/Cuberite%20Linux%20raspi-armhf%20Master/lastSuccessfulBuild/artifact/Cuberite.tar.gz
Resolving builds.cuberite.org (builds.cuberite.org)... 91.121.65.133, 2001:41d0:1:7485::1
Connecting to builds.cuberite.org (builds.cuberite.org)|91.121.65.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

Joulinar, 2020-09-06:
Hi,

many thanks for your report. Pls update to latest version DietPi v6.32.2. The download link hast been changed recently and was updated on latest version.

https://github.com/MichaIng/DietPi/blob/7bf5e1041f3b2972d7827c48215069d1c90eee07/CHANGELOG.txt#L41

jeffthejeffthejeff, 2020-09-06:
Thank you, I didn't think that an update would change anything, but apparently so! 

**Closed Issue.**

closed 2020-09-06 by jeffthejeffthejeff