MichaIng/DietPi #3763: cant install wireguard on rpi4 64bit dietpi

frezeen, 2020-09-07:
#### Details:
- Date           | Mon Sep  7 16:34:02 CEST 2020
- Bug report     | 97ca8c09-2d2d-40ed-8308-471a0c4e040a
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.4.61-v8+ #1339 SMP PREEMPT Tue Sep 1 18:58:26 BST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Setting up wireguard-dkms (1.0.20200712-1~bpo10+1) ...
Removing old wireguard-1.0.20200712 DKMS files...

------------------------------
Deleting module version: 1.0.20200712
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-1.0.20200712 DKMS files...
It is likely that 5.4.61-v8+ belongs to a chroot's host
Building for 5.4.51+, 5.4.51-v7+, 5.4.51-v7l+, 5.4.51-v8+, 5.4.61+, 5.4.61-v7+, 5.4.61-v7l+ and 5.4.61-v8+
Building initial module for 5.4.51+
Error! Bad return status for module build on kernel: 5.4.51+ (aarch64)
Consult /var/lib/dkms/wireguard/1.0.20200712/build/make.log for more information.
dpkg: error processing package wireguard-dkms (--configure):
 installed wireguard-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 wireguard-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2020-09-07:
Hi,

many thanks for your message. Yes, this is a known downside of the 64bit Raspberry OS Kernel.

MichaIng, 2020-09-07:
Btw while we discussed the missing headers in #3570, a tool I totally forgot is [rpi-source](https://github.com/RPi-Distro/rpi-source). Since also on the RPi repo issues this has not been mentioned as far as I remember, not sure if it works on the 64-bit image, so all testing. So the schedule would be:
```
# Update kernel to lastest master/development state
apt update
apt install rpi-update
rpi-update
# Reboot to load it
reboot
# Install Python 3
apt update
apt install python3
# Use rpi-source to pull latest headers
wget https://raw.githubusercontent.com/RPi-Distro/rpi-source/master/rpi-source -O /usr/local/bin/rpi-source
chmod +x /usr/local/bin/rpi-source
rpi-source -q --tag-update
rpi-source
# Install WireGuard and hope for the best
apt install wireguard
```

Joulinar, 2020-09-07:
ok will do test with `rpi-source` later the evening. Let me see if this is working.

Joulinar, 2020-09-07:
@MichaIng 

hmm dosn't seems to work that way. It still give errors while building the package

`cat /var/lib/dkms/wireguard/1.0.20200712/build/make.log`
<details>
  <summary>Log file, click to expand!</summary>

```
root@DietPi3:~# cat /var/lib/dkms/wireguard/1.0.20200712/build/make.log
DKMS make.log for wireguard-1.0.20200712 for kernel 5.4.61-v8+ (aarch64)
Mon Sep  7 22:47:55 CEST 2020
make: Entering directory '/root/linux-a3e07c5dfef2664fc1d900ebb4a3f91c04ab700b'
  AR      /var/lib/dkms/wireguard/1.0.20200712/build/built-in.a
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/main.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/device.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/noise.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/peer.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/timers.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/queueing.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/send.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/receive.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/socket.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/peerlookup.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/allowedips.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/ratelimiter.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/cookie.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/netlink.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/chacha20/chacha20.o
  PERLASM /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/chacha20/chacha20-arm64.S
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/poly1305/poly1305.o
  PERLASM /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/poly1305/poly1305-arm64.S
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/chacha20poly1305.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/blake2s/blake2s.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/curve25519/curve25519.o
  AS [M]  /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/chacha20/chacha20-arm64.o
  AS [M]  /var/lib/dkms/wireguard/1.0.20200712/build/crypto/zinc/poly1305/poly1305-arm64.o
  LD [M]  /var/lib/dkms/wireguard/1.0.20200712/build/wireguard.o
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "_mcount" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "_raw_spin_unlock" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "fpsimd_context_busy" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_cmp4" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_const_cmp2" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_cmp8" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "find_next_bit" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__warn_printk" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_const_cmp1" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_cmp2" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_pc" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "preempt_count_sub" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "debug_smp_processor_id" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__rcu_read_unlock" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_const_cmp8" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_const_cmp4" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "preempt_schedule" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "preempt_schedule_notrace" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "cpu_hwcap_keys" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "cpu_have_feature" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_cmp1" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__sanitizer_cov_trace_switch" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "__rcu_read_lock" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "preempt_count_add" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
ERROR: "arm64_const_caps_ready" [/var/lib/dkms/wireguard/1.0.20200712/build/wireguard.ko] undefined!
make[1]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make: *** [Makefile:1638: modules] Error 2
make: Leaving directory '/root/linux-a3e07c5dfef2664fc1d900ebb4a3f91c04ab700b'
root@DietPi3:~#
```
</details>

MichaIng, 2020-09-07:
Lots of undefined functions, strange. Okay it was worth to give it a try but it seems we need to wait for the kernel headers package...

Joulinar, 2020-09-07:
still, building own kernel (headers) seems to be best option atm

frezeen, 2020-09-13:
joulinar sorry for delay, so, can u explain how "building own kernel" for to solve this?

Joulinar, 2020-09-13:
@frezeen 

That's basically what I did

https://github.com/MichaIng/DietPi/issues/3570#issuecomment-669347814

MichaIng, 2020-09-16:
I mark this as closed, feel free to reopen if required.

closed 2020-09-16 by MichaIng