MichaIng/DietPi #3764: problem python object and openSSL

maury77, 2020-09-07:
What to investigate

I have this problme with OpenSSL 

Traceback (most recent call last):
File “/mnt/dietpi_userdata/homeassistant/deps/lib/python3.8/site-packages/pyatv/support/init.py”, line 29, in error_handler
return await func(*args, **kwargs)
File “/mnt/dietpi_userdata/homeassistant/deps/lib/python3.8/site-packages/pyatv/mrp/auth.py”, line 32, in start_pairing
self.srp.initialize()
File “/mnt/dietpi_userdata/homeassistant/deps/lib/python3.8/site-packages/pyatv/mrp/srp.py”, line 98, in initialize
self._signing_key = Ed25519PrivateKey.from_private_bytes(os.urandom(32))
File “/home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/site-packages/cryptography/hazmat/primitives/asymmetric/ed25519.py”, line 61, in from_private_bytes
raise UnsupportedAlgorithm(
cryptography.exceptions.UnsupportedAlgorithm: ed25519 is not supported by this version of OpenSSL.

the my version of openssl is

/usr/bin# openssl version
OpenSSL 1.1.1d 10 Sep 2019

 DietPi v6.32.2 : 21:36 - Mon 07/09/20
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B+ (armv7l)

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"

Expected outcome

MichaIng, 2020-09-08:
Many thanks for your report. Hmm, actually OpenSSL 1.1.1 supports Ed25519: https://www.openssl.org/blog/blog/2018/09/11/release111/

Probably it has to do with the Python version, we should increase it anyway. But to be able to test, since Ed25519 is no officially accepted algorithm for HTTPS, for which Home Assistant module and task was this certificate required?

maury77, 2020-09-08:
the modules are

- pyatv (appletv integration)
- Homekit controller (for integration via homekit)

the problem is for autentication the device during paring 

thanks

MichaIng, 2020-09-08:
Okay, and that authentication can only be done via Ed25519 key? Reasonable to use it if possible, modern, small size etc, but probably it is simply not yet supported in that combination.

I have no actual devices I can test pairing with, but probably I can trigger the same function with an Ed25519 key.

That the Buster OpenSSL version supports Ed25519 btw I do know for sure since I use it myself for SSH authentication on two Buster systems via OpenSSH server, which uses the same OpenSSL library. So this must be a Python environment internal issue.

maury77, 2020-09-09:
t seems that for apple devices it is the only method for paring

If you want I am available to test

maury77, 2020-09-15:
any idea for a workaround o manual acation?

thanks

MichaIng, 2020-09-15:
Checking where/how this check is performed:
```
    def ed25519_supported(self):
        return not self._lib.CRYPTOGRAPHY_OPENSSL_LESS_THAN_111B
```
1.1.1d > 1.1.1b so that should return true.

Adding some debug code to `site-packages/cryptography/hazmat/backends/openssl/backend.py` reveals that this check returns false:
```
import logging
if backend._lib.CRYPTOGRAPHY_OPENSSL_LESS_THAN_111B:
    logging.error('true')
else:
    logging.error('false')
```
```
2020-09-15 16:19:57 homeassistant@VM-Buster:~/.pyenv/versions/3.8.0/lib/python3.8/site-packages/cryptography/hazmat/backends/openssl$ python3 backend.py
/home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/site-packages/cryptography/hazmat/primitives/constant_time.py:22: CryptographyDeprecationWarning: Support for your Python version is deprecated. The next version of cryptography will remove support. Please upgrade to a release (2.7.7+) that supports hmac.compare_digest as soon as possible.
  warnings.warn(
ERROR:root:false
ERROR:root:false
```
Includes another hint that Python needs to be upgraded.
__________
Done: https://github.com/MichaIng/DietPi/commit/06a6bb8866736901857e84fe693ce27201e6efa0
Let's see if this makes a difference.

MichaIng, 2020-09-15:
Okay, reading a bid further: https://github.com/home-assistant/core/issues/26949
Known issue that it is broken currently. And I was wondering why I couldn't select the Apple TV component from HA GUI anymore 😄. It's being rewritten from ground up.
@maury77 you may test the beta: https://github.com/postlund/hass-atv-beta
So while it's good to have Python up-to-date, this issue is outside of our control.

maury77, 2020-09-15:
i write so:

import base64
import collections
**import logging**
import contextlib
import itertools
from contextlib import contextmanager

   def ed25519_supported(self):
        **if backend._lib.CRYPTOGRAPHY_OPENSSL_LESS_THAN_111B:
            logging.error('true')
        else:
            logging.error('false')**
        return not self._lib.CRYPTOGRAPHY_OPENSSL_LESS_THAN_111B

but not the output si different

 /home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/site-packages/cryptography/hazmat/backends/openssl#  python3.8 backend.py
/home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/site-packages/cryptography/hazmat/primitives/constant_time.py:22: CryptographyDeprecationWarning: Support for your Python version is deprecated. The next version of cryptography will remove support. Please upgrade to a release (2.7.7+) that supports hmac.compare_digest as soon as possible.
  warnings.warn(
root@sdo:/home/homeassistant/.pyenv/versions/3.8.0/lib/python3.8/site-packages/cryptography/hazmat/backends/openssl#

MichaIng, 2020-09-15:
That function is not executed, you need to add it to a code part that is executed 😉. However no need to debug since as above it is well known. But to check back, what is the tvOS version on your Apple TV?

maury77, 2020-09-15:
ok sorry the apple tv is ATV4 with ios 13

the problem is not only with appletv but also with homekit componet for paring gateway with home assisntat.

maury77, 2020-09-15:
> Okay, reading a bid further: [home-assistant/core#26949](https://github.com/home-assistant/core/issues/26949)
> Known issue that it is broken currently. And I was wondering why I couldn't select the Apple TV component from HA GUI anymore 😄. It's being rewritten from ground up.
> @maury77 you may test the beta: https://github.com/postlund/hass-atv-beta
> So while it's good to have Python up-to-date, this issue is outside of our control.

I use the last version of the pyatv 0.7.2 and the system discover atv and I select the device form gui but I have this error during the paring  , other distribution non have this problem after upgrade.

MichaIng, 2020-09-15:
Hmm, okay, then please try the following:
```
sed -i 's/3\.8\.0/3.8.5/' /boot/dietpi/dietpi-software
dietpi-software reinstall 157
```
Let's see if the new Python version solves it.

maury77, 2020-09-21:
during the installation have this errror

[ INFO ] DietPi-Software | APT install for: gcc libc6-dev make zlib1g-dev libbz2-dev libreadline-dev libssl-dev libsqlite3-dev libffi-dev, please wait...
E: Unable to correct problems, you have held broken packages.

I'm blocked

Joulinar, 2020-09-21:
pls can you run following

```
apt update
apt upgrade
```

maury77, 2020-09-21:
I solve it with reinstall manaule package sudo apt-get install gcc-8-base

But after upgrade del problem of certificate is present

MichaIng, 2020-09-21:
Btw we found libjpeg-dev being required now on ARMs: https://github.com/MichaIng/DietPi/pull/3781
```
apt install libjpeg-dev
```
No idea about the certificate issue since from what I get https://github.com/postlund/hass-atv-beta is exactly what should solve that issue. But it is still in beta and "Issues" are disabled on that repository so we have not much chance to evaluate how reliable it works for other users. But that is basically out of our control as we provide a compatible OS environment, an independent portable Python environment and plain HA core install inside, so nothing that would break any modern certificate support.

`gcc-8-base` was not installed on your system?? That is a very essential package, you cannot even remove it without interactively ignoring a clear warning:
```
You are about to do something potentially harmful.
To continue type in the phrase 'Yes, do as I say!'
 ?]
```
So I wonder how this could have happened 🤔.

maury77, 2020-09-21:
I don't know how it happened but after restoring that package it worked fine

Joulinar, 2020-09-21:
Probably an update was hanging/not implemented correctly before?? 🤔

maury77, 2020-09-21:
I recently upgraded to buster

MichaIng, 2020-09-21:
> I recently upgraded to buster

Ah that will be the reason:
- On Stretch `gcc-6-base` is installed which satisfies package dependencies in most cases, since often there is no direct version dependency given on e.g. libc6 > libgcc1 > gcc-X-base.
- But of course on Buster the natively available `gcc-8-base` is required to assure all works fine.
- Although when running `apt full-upgrade`/`apt-get dist-upgrade` that "should" actually be included, from what I see 🤔.

However, basically such things are the reason why we always recommend to flash a fresh image when switching to a new Debian version.

maury77, 2020-09-22:
Ok non Distribution si ok 

But i ha e the dame issue , 

cryptography.exceptions.UnsupportedAlgorithm: ed25519 is not supported by this version of OpenSSL.

the author Says that si a problemi of OpenSSL

https://github.com/postlund/pyatv/issues/831#issuecomment-696948177

Have you any idea?

Thanks

maury77, 2020-09-23:
GREAT!!! the problem was solved

thank you!!!

closed 2020-09-23 by maury77

MichaIng, 2020-09-23:
Great. Note to others: The trick was updating the cryptography module which has a hardcoded dependency version at HA. So we need to forward the request to update that.

enwi, 2020-10-19:
I am having a similar issue where `ed25519 is not supported`, but while trying to add a eufy home base as a HomeKit Device. I can't find anywhere else where this issue is mentioned for eufy decvices, so i am just adding to this issue.

I am running `DietPi v6.33.3` on `Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)` and my installed Openssl version (which apt-get states is the latest) is `1.1.0l-1~deb9u1`, but running `openssl genpkey -algorithm ed25519 -outform PEM -out test25519.pem` gives `Algorithm ed25519 not found`

Joulinar, 2020-10-19:
Hi

maybe that's an issue of your Debian version?? I guess you are on Stretch?? I'm not 100% sure but probably support for `ed25519` was added on newer version only. I guess 1.1.1 would be needed.

enwi, 2020-10-20:
@Joulinar You are right I am on stretch:
```
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.13 (stretch)
Release:	9.13
Codename:	stretch
```
So how do I upgrade? Can I even upgrade?

enwi, 2020-10-20:
Alright I found instructions [here](https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988) I will try that and make a backup

Joulinar, 2020-10-20:
yes pls do a `dietpi-backup` before. Or better to clone the SD card 😄 
This will pump you to Buster.

enwi, 2020-10-20:
Yeah I got to do it a little different, since the rootfs is on the HC1 harddrive and the SD card OS loads that

MichaIng, 2020-12-05:
cryptography module dependency update as well as Apple TV integration are both done now btw. The latter will be part of upcoming HA 1.0 release:
- https://github.com/home-assistant/core/pull/40479#issuecomment-718693489
- https://github.com/postlund/pyatv/issues/209#issuecomment-739024311

nick2525, 2021-01-17:
still have same issue on OpenSSL 1.1.1, cryptography-3.3.1 ed25519 is not supported by this version of OpenSSL