MichaIng/DietPi #3765: Update failed

hai3009, 2020-09-08:
[###](url) Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

[image]

- ...

Joulinar, 2020-09-08:
Hi,

Many thanks for your report. Do you use PiHole? Because you are resolving Github with 0.0.0.0. Looks like you DNS request is blocked somewhere.

hai3009, 2020-09-08:
thanks!Don't know how to use it PiHole

Joulinar, 2020-09-08:
If you use PiHole, login to web interface, search on the logs if something from GitHub is blocked. If yes, put it to Whitelist.

MichaIng, 2020-09-08:
Jep indeed this is not the first time that Pi-hole (i.e. one of the available non-standard block lists) blocks `githubusercontent.com`, a shame.

But just to be sure it it really Pi-hole, as faulty ISP or local DNS servers or hosts file can do the same thing of course:
```
cat /etc/resolv.conf
grep '0\.0\.0\.0' /etc/hosts
curl -IL --dns-servers 9.9.9.9 raw.githubusercontent.com
```

hai3009, 2020-09-10:
When will the latest image download be released？

Joulinar, 2020-09-10:
@hai3009 
what you mean? Or what do you looking for?

MichaIng, 2020-09-10:
I'll now create a new image for RPi, but that will not solve your issue as this is not related to DietPi but a blocking list or firewall rule on your side 😉.

Joulinar, 2020-09-10:
yep, last RPI image was released less than 2 month ago and it working fine. As already said, you issue seems to be some DNS blocking, aka PiHole 😉

hai3009, 2020-09-10:
Thank you for your reply. I can only use dietpi first_ V6.18 is not upgraded

Joulinar, 2020-09-10:
a new image will bring you to latest release DietPi v6.32.2, always.

But still, you issue on your very old/outdated V6.18 has nothing to do with DietPi. You are not able to complete your update, because there is something on your side who is blocking DNS resolution for `raw.githubusercontent.com`

MichaIng, 2020-09-10:
New image uploaded, after 1-2 positive feedback I'll move it to stable downloads: https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv6-Buster.7z

hai3009, 2020-09-11:
Thank you. The new image can be used normally.

MichaIng, 2020-09-11:
Many thanks for testing and feedback! I mark this issue as closed then. Feel free to reopen if required.

closed 2020-09-11 by MichaIng