MichaIng/DietPi #3766: rc.local default file

4o4forbidden, 2020-09-08:
Hello, i edited the rc.local file by wrong can some one send le the original one please ? thanks a lot.

MichaIng, 2020-09-08:
The default basically does nothing, simply erase it and add the code you need or remove it completely. But if you insist in the original as shipped with Debian:
```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.
 
exit 0
```
Only informational comments and exist 😉.

4o4forbidden, 2020-09-09:
Thanks a lot sir !

closed 2020-09-09 by 4o4forbidden