MichaIng/DietPi #3767: Update failed

68gt500, 2020-09-08:
#### Details:
- Date           | Tue  8 Sep 19:04:33 CEST 2020
- Bug report     | N/A
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | Linux rpi3b 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
- Distro         | buster (index=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. run dietpi-update 
2. ...

#### Expected behaviour:

- update DietPi from v6.26.3 to v6.32.2.

#### Actual behaviour:

- failing with above messages

#### Extra details:

#### Additional logs:
```
 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Applying DietPi update

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : webmin
[  OK  ] DietPi-Services | stop : node-red
[  OK  ] DietPi-Services | stop : mosquitto
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Services | stop : proftpd

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Applying pre-patches

[  OK  ] DietPi-Update | wget https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -O pre-patch_file
[ INFO ] Pre-patch 20 | Mark ifupdown as manually installed to prevent its autoremoval
ifupdown was already set to manually installed.
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://ppa.launchpad.net/webupd8team/java/ubuntu xenial InRelease
Get:2 http://repo.mosquitto.org/debian jessie InRelease [11.0 kB]
Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Err:2 http://repo.mosquitto.org/debian jessie InRelease
  The following signatures were invalid: EXPKEYSIG 61611AE430993623 Mosquitto Apt Repository <repo@mosquitto.org>
Fetched 11.0 kB in 1s (7,327 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://repo.mosquitto.org/debian jessie InRelease: The following signatures were invalid: EXPKEYSIG 61611AE430993623 Mosquitto Apt Repository <repo@mosquitto.org>
W: Failed to fetch http://repo.mosquitto.org/debian/dists/jessie/InRelease  The following signatures were invalid: EXPKEYSIG 61611AE430993623 Mosquitto Apt Repository <repo@mosquitto.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
[  OK  ] DietPi-Update | G_AGUP
[ INFO ] DietPi-Update | APT upgrade, please wait...
Setting up samba-common-bin (2:4.9.5+dfsg-5+deb10u1+rpi1) ...
Checking smb.conf with testparm
handle_name_resolve_order: WARNING: Ignoring invalid list value 'hosts' for parameter 'name resolve order'
Load smb config files from /etc/samba/smb.conf
handle_name_resolve_order: WARNING: Ignoring invalid list value 'hosts' for parameter 'name resolve order'
Error loading services.
dpkg: error processing package samba-common-bin (--configure):
 installed samba-common-bin package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of samba:
 samba depends on samba-common-bin (= 2:4.9.5+dfsg-5+deb10u1+rpi1); however:
  Package samba-common-bin is not configured yet.

dpkg: error processing package samba (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of winbind:
 winbind depends on samba-common-bin (= 2:4.9.5+dfsg-5+deb10u1+rpi1); however:
  Package samba-common-bin is not configured yet.

dpkg: error processing package winbind (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 samba-common-bin
 samba
 winbind
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | G_AGUG
[ INFO ] DietPi-Update | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.

dsnyder0pc, 2020-09-08:
Looks like a read after write consistency problem with repo sync. Perhaps wait an hour and then try again. I'd manually run an `apt update` first (just to make myself feel better, likely redundant).

Joulinar, 2020-09-08:
Hi,

many thanks for your report. Basically I can see 3 issues on your system

1. Mosquitto seems to have Debian Jessie repository while you are on Debian Buster. Therefore it would make sense to switch Mosquitto to buster repository as well. Or is/was there a specific reason to have Jessie repository attached?
```
Err:2 http://repo.mosquitto.org/debian jessie InRelease
  The following signatures were invalid: EXPKEYSIG 61611AE430993623 Mosquitto Apt Repository <repo@mosquitto.org>
```
2. Samba configuration file seems to contain incorrect or wrong settings. This would need to be corrected.
```
Load smb config files from /etc/samba/smb.conf
handle_name_resolve_order: WARNING: Ignoring invalid list value 'hosts' for parameter 'name resolve order'
Error loading services.
```
3. Samba package samba-common-bin seems to have issues to get installed. Maybe you can try `apt --reinstall install samba-common-bin`
```
dpkg: error processing package samba-common-bin (--configure):
 installed samba-common-bin package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of samba:
 samba depends on samba-common-bin (= 2:4.9.5+dfsg-5+deb10u1+rpi1); however:
  Package samba-common-bin is not configured yet.
```

68gt500, 2020-09-08:
Joulinar, thanks for your input.
How would I switch mosquitto to Buster? 
apt --reinstall install samba-common-bin yields the following messages:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libev4 libhavege1 libwebsockets8
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 5 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
E: Internal Error, No file name for samba-common-bin:armhf

MichaIng, 2020-09-08:
> How would I switch mosquitto to Buster?

```
sed -i '/mosquitto/s/jessie/buster/' /etc/apt/sources.list{,.d/*.list}
```

> E: Internal Error, No file name for samba-common-bin:armhf

Hmm I saw something similar already a few weeks ago. Please try a general attempt first:
```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt -f install
```
If that fails, please paste the output of:
```
tail -5 /var/lib/dpkg/info/samba-common-bin:armhf.list
```
If I remember right, this file was either empty or missing in the other case.

68gt500, 2020-09-09:
Thankx for the quick response:
Here are the results:

 apt-cache (clean)
[  OK  ] apt-cache clean | Completed
-------------------------------------------------------------------------
apt update
Get:1 http://ppa.launchpad.net/webupd8team/java/ubuntu xenial InRelease [17.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://repo.mosquitto.org/debian jessie InRelease [11.0 kB]
Get:4 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/rpi armhf Packages [1,360 B]
Err:3 http://repo.mosquitto.org/debian jessie InRelease
  The following signatures were invalid: EXPKEYSIG 61611AE430993623 Mosquitto Apt Repository <repo@mosquitto.org>
Get:7 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:8 https://archive.raspberrypi.org/debian buster/main armhf Packages [330 kB]
Get:9 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Reading package lists... Done
W: GPG error: http://repo.mosquitto.org/debian jessie InRelease: The following signatures were invalid: EXPKEYSIG 61611AE430993623 Mosquitto Apt Repository <repo@mosquitto.org>
E: The repository 'http://repo.mosquitto.org/debian jessie InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
-------------------------------------------------------------------------------
apt -f install
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libev4 libhavege1 libwebsockets8
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up samba-common-bin (2:4.9.5+dfsg-5+deb10u1+rpi1) ...
Checking smb.conf with testparm
handle_name_resolve_order: WARNING: Ignoring invalid list value 'hosts' for parameter 'name resolve order'
Load smb config files from /etc/samba/smb.conf
handle_name_resolve_order: WARNING: Ignoring invalid list value 'hosts' for parameter 'name resolve order'
Error loading services.
dpkg: error processing package samba-common-bin (--configure):
 installed samba-common-bin package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of samba:
 samba depends on samba-common-bin (= 2:4.9.5+dfsg-5+deb10u1+rpi1); however:
  Package samba-common-bin is not configured yet.

dpkg: error processing package samba (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of winbind:
 winbind depends on samba-common-bin (= 2:4.9.5+dfsg-5+deb10u1+rpi1); however:
  Package samba-common-bin is not configured yet.

dpkg: error processing package winbind (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 samba-common-bin
 samba
 winbind
E: Sub-process /usr/bin/dpkg returned an error code (1)
----------------------------------------------------------
tail -5 /var/lib/dpkg/info/samba-common-bin:armhf.list
tail: cannot open '/var/lib/dpkg/info/samba-common-bin:armhf.list' for reading: No such file or directory

Seems that I am missing that file. How can it be recreated?

68gt500, 2020-09-09:
sed -i '/mosquitto/s/jessie/buster/' /etc/apt/sources.list{,.d/*.list}

rerun of DietPi-Update:
─────────────────────────┤ DietPi Error Handler: ├──────────────────────────────
tPi-Update: G_AGUP
Exit code: 100
DietPi version: v6.26.3 (MichaIng/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:5
Image creator: DietPi Core Team
Pre-image: Raspbian Lite

 file content:
:1 http://ppa.launchpad.net/webupd8team/java/ubuntu xenial InRelease
:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
:3 http://repo.mosquitto.org/debian buster InRelease [11.0 kB]
:4 https://archive.raspberrypi.org/debian buster InRelease
:3 http://repo.mosquitto.org/debian buster InRelease
he following signatures were invalid: EXPKEYSIG 61611AE430993623 Mosquitto Apt
ository <repo@mosquitto.org>

Joulinar, 2020-09-09:
I guess the file should be `samba-common-bin.list` and not `samba-common-bin:armhf.list`. At least that what it is on my test system.
pls check `tail -5 /var/lib/dpkg/info/samba-common-bin.list`

@MichaIng 
I guess Mosquitto source list can be removed because package should be provided by Raspbian repository.

```
root@DietPi3:# apt show mosquitto
Package: mosquitto
Version: 1.5.7-1+deb10u1
Priority: optional
Section: net
Maintainer: Roger A. Light <roger@atchoo.org>
Installed-Size: 318 kB
Depends: adduser (>= 3.10), libuuid1 (>= 2.16), lsb-base (>= 4.1+Debian3), libc6 (>= 2.28), libssl1.1 (>= 1.1.0), libsystemd0, libwebsockets8 (>= 1.6.0), libwrap0 (>= 7.6-4~)
Suggests: apparmor
Homepage: https://mosquitto.org/
Download-Size: 143 kB
APT-Sources: http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
Description: MQTT version 3.1/3.1.1 compatible message broker
 This is a message broker that supports version 3.1 and 3.1.1 of the MQTT
 protocol.
 .
 MQTT provides a method of carrying out messaging using a publish/subscribe
 model. It is lightweight, both in terms of bandwidth usage and ease of
 implementation. This makes it particularly useful at the edge of the network
 where a sensor or other simple device may be implemented using an arduino for
 example.

root@DietPi3:#
```

68gt500, 2020-09-09:
This is what I get: 
tail -5 /var/lib/dpkg/info/samba-common-bin.list
/usr/share/samba/addshare.py
/usr/share/samba/setoption.py
/var
/var/cache
/var/cache/samba

How can that be fixed?
Since I don´t really need Samba  I tried to remove it via DietPie-Sofware - but it is not listed as installed.

Joulinar, 2020-09-09:
Try `apt remove samba`

MichaIng, 2020-09-09:
> I guess Mosquitto source list can be removed because package should be provided by Raspbian repository.

Indeed we switched from Mosquitto repo to Debian a long time ago. This is a Jessie => Buster upgraded system, obviously, hence the issues. But actually the official Mosquitto repo of course ships newer versions, so if its already there, why not using it. To re-add or update the key:
```
curl -sSfL http://repo.mosquitto.org/debian/mosquitto-repo.gpg.key | apt-key add -
```

Indeed if Samba is not required anyway, simply purge it (purge to remove all related config files as well, which might or might not be related to the problem, e.g. old ones from Jessie). I guess on reinstall the issues would be gone then as well:
```
apt purge samba samba-common-bin
apt autopurge
```

Joulinar, 2020-09-09:
regarding Java8. I guess it's available via Raspbian repository, isn't it? It still should be shipped this way if I'm not mistaken. Question is if Java8 us needed of if it can be switched to Java11 😃

68gt500, 2020-09-09:
> 
> 
> Try `apt remove samba`

It did exit with the same error.

68gt500, 2020-09-09:
YESS! its is up to date : 
 DietPi v6.32.2 : 13:44 - Wed 09/09/20

I did:
curl -sSfL http://repo.mosquitto.org/debian/mosquitto-repo.gpg.key | apt-key add -
apt purge samba samba-common-bin
apt autopurge

....
[  OK  ] DietPi-Update | Incremental patching to v6.32.2 completed
[ SUB2 ] DietPi-Update > Completed
[ INFO ] DietPi-Update | Current version : v6.32.2
[ INFO ] DietPi-Update | Latest version  : v6.32.2
[  OK  ] DietPi-Survey | Sending survey data
[ SUB3 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : proftpd
[  OK  ] DietPi-Services | restart : mosquitto
[  OK  ] DietPi-Services | restart : node-red
[  OK  ] DietPi-Services | restart : webmin
[  OK  ] DietPi-Services | restart : cron
[  OK  ] DietPi-Patch | Everything done! Terminating the obsolete DietPi-Update parent instance...

Terminated

Even after a reboot all looks GREAT!

Thank you !

68gt500, 2020-09-09:
Great product, great support - Thank you!

closed 2020-09-09 by 68gt500

MichaIng, 2020-09-09:
@Joulinar 
> regarding Java8. I guess it's available via Raspbian repository, isn't it? It still should be shipped this way if I'm not mistaken. Question is if Java8 us needed of if it can be switched to Java11 😃

The repo ships Oracles Java and from what I read due to license changes their Java 11 cannot be provided that way anymore. We ship the default OpenJDK version from Debian, on Stretch its still Java 8 (last LTS) and since Buster its Java 11 (current LTS). Only on ARMv6 we are forced to install Java 8 since it is not compatible with Java 11. Damn thing as this means doors to drop ARMv6 support in general are opened. Most Java software will likely not support Java 8 anymore, so we need to disabled those options on ARMv6... I fear that a lot of RPi Zero users will be very sad... regardless other topic.

@68gt500 
Great that it finally worked. Distro-upgraded systems indeed require some manual fixes of that kind usually, due to incompatible config files or repositories. That is why we recommend to flash images freshly instead, knowing that for some admins that means quite some time of work.