MichaIng/DietPi #3769: dietpi-software install

adntsierradegata, 2020-09-09:
#### Details:
- Date           | Wed Sep  9 14:11:30 CEST 2020
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Packages were downgraded and -y was used without --allow-downgrades.
```

MichaIng, 2020-09-09:
Many thanks for your report. Please paste the output of the following command:
```
apt upgrade
```

adntsierradegata, 2020-09-09:
Los siguientes paquetes se han retenido:
  pcmanfm
Se DESACTUALIZARÁN los siguientes paquetes:
  libxcb-util0
0 actualizados, 0 nuevos se instalarán, 1 desactualizados, 0 para eliminar y 1 no actualizados.
Se necesita descargar 21,9 kB de archivos.
Se utilizarán 0 B de espacio de disco adicional después de esta operación.
¿Desea continuar? [S/n] s

MichaIng, 2020-09-09:
I guess accepting this should solve the issue already. However, I'm not sure about the issue with `libxcb-util0` (due to lack of Spanish language knowledge), is it being downgraded somehow? Which version do you have installed and which version would be installed?
```
apt policy libxcb-util0
```

For pcmanfm we actually changed things, please do:
```
apt unhold pcmanfm
apt install --reinstall pcmanfm
```

When opening the file manager on desktop, you'll see some missing icons. Those are new options provided by the RPi version of pcmanfm but the default Debian icon sets lack icons for them. You can either change the icon theme to one of the RPi desktop ones, like pix-icons (`apt install pix-icons`, then change via `lxappearance`/"Look and feel" settings) or you can disable those additional buttons in PCManFM: Edit->Preferences->Layout->disable "Display simplified user interface and menus"
Here the history of that issue any why we installed the Debian version on Raspbian before: https://github.com/MichaIng/DietPi/issues/1558

closed 2020-12-17 by MichaIng