MichaIng/DietPi #3770: Non-interactive firstrun setup fails when AUTO_UNMASK_LOGIND is set to 1 in /boot/dietpi.txt

razerbann, 2020-09-10:
### Creating a bug report/issue

#### Required Information
- DietPi version |
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=32
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux raspberrypi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Official
- SDcard used | SanDisk ultra

#### Steps to reproduce
1- Set AUTO_SETUP_AUTOMATED to 1 in /boot/dietpi.txt
2- Set AUTO_UNMASK_LOGIND to 1 in /boot/dietpi.txt

#### Expected behaviour
`systemd-logind` service unmasked and started.

#### Actual behaviour
Non-interactive firstrun setup fails and RPi doesn't reboot.

#### Extra details
`/var/tmp/dietpi/logs/dietpi-firstrun-setup.log` file indicates the following:
`Failed to start systemd-logind.service: Unit dbus.socket not found.`

razerbann, 2020-09-10:
This issue might be related to #3265

MichaIng, 2020-09-10:
Many thanks for your report. Indeed, `systemd-logind` requires `dbus`, it's install is simply missing.
I'll add that.

MichaIng, 2020-09-10:
Fixed: https://github.com/MichaIng/DietPi/commit/83931886a3b6729e9ee3934a8d4de82d95cf5877
Changelog: https://github.com/MichaIng/DietPi/commit/5bd0ae344db94ee9bdfa14b181d63e7d6ac13115

razerbann, 2020-09-11:
> Fixed: [8393188](https://github.com/MichaIng/DietPi/commit/83931886a3b6729e9ee3934a8d4de82d95cf5877)
> Changelog: [5bd0ae3](https://github.com/MichaIng/DietPi/commit/5bd0ae344db94ee9bdfa14b181d63e7d6ac13115)

Thank you for your reactivity and efficiency! 👍

closed 2020-09-13 by razerbann