MichaIng/DietPi #3772: K3S

blmhemu, 2020-09-12:
### Creating a software request

#### Formal software information
- Software title | K3S
- Short description | Rancher Labs - Lightweight Kubernetes for Edge, IoT, CI, ARM
- Official URL (if available) | https://k3s.io/
- GitHub URL (if available) | https://github.com/rancher/k3s
- (Official) install documentation | https://rancher.com/docs/k3s/latest/en/

#### Are there similar/alternative software titles available with DietPi-Software?
- Was not able to find any.

#### What makes your requested software better than the above solutions, if available?
- It is fully open source, without any (feature) limitation to free users. 
- Can be used to cluster multiple RPis
- Better than Docker (in clustering) and widespread support from community.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
- K3S needs container features to be enabled. Currently RPi supports all cgroups but disables memory cgroup by default. Adding a few lines to /boot/cmdline.txt and rebooting will solve this.
- Please look at https://blog.alexellis.io/test-drive-k3s-on-raspberry-pi/

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. Enable cgroups (esp memory for rpi. May be something else for other boards)
2. Run the scripts provided in the docs. Also add support for enabling / disabling serviceLB and Traefik when installing.
3. Please have a look at https://kauri.io/install-and-configure-a-kubernetes-cluster-with-k3s-to-self-host-applications/418b3bc1e0544fbc955a4bbba6fff8a9/a

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
- I am not exactly familiar with this code base. But have done fair amount of programming in C / Python. Familiar with bash scripting but not a pro.

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/
https://feathub.com/MichaIng/DietPi/+186

MichaIng, 2020-09-12:
Many thanks for your request. Since we aim to natively integrate multiple software into the system (without a container layer), such has not a high priority from my end, but I'll follow the votes in case 😉.

There is a reason why memory cgroup is by default disabled on most distributions, but as you link it can be easily enabled. Armbian-based images (where direct cmdline access is not that easy) have the `docker_optimization` setting for that.

theAkito, 2020-09-12:
I am sitting a long time on k3s with Raspberry Pi, however I always put it back, because other issues had a much higher priority. Perhaps the time has come that I could handle this, as I wanted to anyway, at some point.

blmhemu, 2020-09-13:
Thanks for the fast reply @MichaIng 
Just for your I was able to bring the cluster Up and Running using the links in the description.
But I have become mildly curious as to why "memory cgroup is by default disabled on most distributions".

MichaIng, 2020-09-13:
Here the one reason why it's disabled on RPi: https://github.com/raspberrypi/linux/commit/ba742b52e5099b3ed964e78f227dc96460b5cdc0
Thia also explains why the guide you posted has two entries to enable it, which is based on a mistake. Of course only `cgroup_enable=memory` is required, same like everywhere else ;).

eljojo, 2021-01-28:
K3s is a perfect fit for diet-pi, since they're so pragmatic and lightweight 👍

sintaxx, 2021-02-12:
this is a great idea, i've been wanting to experiment and learn kubernetes and i have 3 pi's sitting around doing nothing.. i think it would be easy to get up and running since dietpi is so quick and simple to install

mortenlj, 2021-06-06:
I've recently stumbled onto DietPi, looking for a lightweight OS to use as a base for running K3s. I might be interested in working on this, but since I don't know DietPi much, I need some feedback before I get too far.

I've looked over the docs, and had a good look at the (IMHO overly long) `dietpi-software` script. I have a good idea about how to install k3s, simply using the official k3s installer. I have a few questions though:

The k3s installer takes quite a few options. It can take them as env-variables, or as a yaml-file. I think "importing" all of them into the `/boot/dietpi.txt` file is overkill, would it be fine to just use a file like `/boot/dietpi-k3s.yaml`, and go with that (with proper documentation)?

K3s runs in one of two modes: server or agent. Each has a ton of configuration options, controlled via command-line or env-variables. The standard installation creates a systemd service, so the natural way to configure them would probably be by adding env-variables to the service-file?

I see some of the software in dietpi-software does some basic configuration, but in this case it might be better to just start with the defaults and let the user do their own config directly, instead of trying to write a rather complicated configuration routine. What's the stance on this sort of thing?

PS.
@sintaxx If you just want to play around, you can just install a base dietpi, and use [k3sup](https://k3sup.dev/) to install k3s manually.

MichaIng, 2021-06-07:
Do you want to be able to install K3S completely non-interactive? If it's about one or two important settings, `/boot/dietpi.txt` could hold them, but for more I agree that is not the right place. But I'm also not sure whether an additional file is great. It would need to be documented then as well, while `/boot/dietpi.txt` is quite known and documents itself 😉.

I haven't had a deeper look into K3S, but generally:
- We try to set reasonable defaults to cover most use cases.
- We do interactive dialogue prompts for mandatory user inputs, but this means that it cannot be installed non-interactive, or automated at first boot.
- We can add 1-2 mandatory inputs as `/boot/dietpi.txt` settings, but more would blow up that file too much, IMO.

If the official installer creates a systemd service already, it likely contains an `EnvironmentFile=` option to load e.g. `/etc/default/k3s`?

> I see some of the software in dietpi-software does some basic configuration, but in this case it might be better to just start with the defaults and let the user do their own config directly, [...]

If there are settings, different from the defaults, that would be good for the majority of cases, like some hardenings, a default password, network ports or such, it's fine to apply them, but otherwise I agree to not try guessing too much but leaving it to the admins. We have some software options where still much settings are applied or whole config files overwritten, but I actually aim to reduce those cases and instead apply only a few important single settings to the default config files, if such exist.

mortenlj, 2021-06-07:
I'll focus on the installer for now, I think the runtime config is better to leave for the admin to do directly after installation. I also realize I was wrong, the installer only cares about environment variables, so the `/boot/dietpi-k3s.yaml` idea can be ignored.

The installer can run fine without any input from the user. It has reasonable defaults, and doesn't require input from the user to run.

On the other hand, it has ~15 optional settings you may want to change depending on your situation, and they are install-time settings. That is, after installation, you have to uninstall and reinstall to change them properly. I think it would be useful to allow the user to change these, and doing non-interactive installs would be a plus.

closed 2021-06-26 by MichaIng