MichaIng/DietPi #3773: Please add comitup

asahajit, 2020-09-12:
### Creating a software request

#### Formal software information
- Software title | comitup
- Short description | 
- Official URL (if available) |https://davesteele.github.io/comitup/
- GitHub URL (if available) | https://github.com/davesteele/comitup
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
Sorry, I am totally unexperienced in bash coding.
- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-09-12:
Many thanks for your request. But I'm sorry to say that we'll not going to implement this as featured software option. These features are already available on DietPi (dietpi-config + WiFi Hotspot software option) and it is very hard already to get network setup right and bulletproof without having an additional software that is aiming the same and will 100% collide.

I'm working on a nice network setup rework to make it more compatible/flexible and powerful. If comitup has a specific feature that is the reason you're requesting it, I suggest you request this specific feature to be added to our native tools instead.

As well, the installation is more than trivial: `apt install comitup`
Install it, see if it works OOTB with/regardless of the existing DietPi network setup. If so take care to not configure network via dietpi-config anymore as both for sure will break each other 😉. We have bad experience in this regards with `webmin` already, too many cooks spoil the broth.

Joulinar, 2020-09-12:
As well it requires NetworkManager, which is not available on DietPi. Isn't?

MichaIng, 2020-09-12:
Oh, I didn't see that, okay that is again an argument against it 😄. Since [NetworkManager](https://packages.debian.org/buster/network-manager) is as well available through APT, not issue to install it on DietPi if really wanted: `apt install network-manager`
I from my side will not touch such high level tools but concentrate on writing our own lightweight DietPi-style implementations 🙂. But if someone else wants to implement this (including a way to block the dietpi-config network menu to avoid accidental collisions), I'll not say "no".

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not be addressed soon. If anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng