MichaIng/DietPi #3776: No python on raspberry pi 3b 64bit

georgkrause, 2020-09-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=32
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`10.5`
- Kernel version | `uname -a`
`Linux ***** 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 3 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
original raspi powe supply
- SDcard used | (EG: SanDisk ultra)
sandisk ultra 32GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
python
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. flash image
2. run on raspberry pi 3b

#### Expected behaviour

- python should be installed, like it is on raspberry pi 4

#### Actual behaviour

- no python available

#### Extra details

- ...

Joulinar, 2020-09-14:
Hi, 

many thanks for your massage. I did a quick test on my RPi3B+ running 64bit Beta and Python3 is available

```
root@DietPi3:~# uname -a
Linux DietPi3 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
root@DietPi3:~# dietpi-software list|grep python3
id 130 | =0 | python3 pip: python3 package installer |
root@DietPi3:~#
```

georgkrause, 2020-09-14:
I had this behavior on two different Raspberry Pis today. I dont understand the question tag, cant see any question here. You are just saying it works for you.

Joulinar, 2020-09-14:
yes `Python3` is available vie `dietpi-software`. You can check it yourselves running `dietpi-software list|grep python3`.

Or do you looking for Python2? Because Python2 has been removed from `dietpi-software` with latest version https://github.com/MichaIng/DietPi/blob/7bf5e1041f3b2972d7827c48215069d1c90eee07/CHANGELOG.txt#L20

georgkrause, 2020-09-14:
I am looking for python3 and yes, its available. But my point is its not installed by default on raspberry pi 3 while it is already installed on raspberry pi 4.

Joulinar, 2020-09-14:
No that should not be the case. The image used, is the same for all RPI devices. And by default DietPi is not installing any additional software. Means, if you use the same image for RPi3 as well as RPi4, result will be same. Right after initial setup, you would have an empty system.

georgkrause, 2020-09-14:
> No that should not be the case.

That was what I thought in the first place and is the reason for filing an issue.

Anyway, this explanation seems legit and its good to know. Therefore I am closing this :)

closed 2020-09-14 by georgkrause