MichaIng/DietPi #3777: Update problem

jumpmad, 2020-09-17:
#### Details:
- Date           | Thu Sep 17 11:11:01 CEST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -o pre-patch_file
- Exit code      | 6
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (6) Could not resolve host: raw.githubusercontent.com
```

Joulinar, 2020-09-17:
Hi,

many thanks for your message. Do you use any local DNS Server like PiHole?

jumpmad, 2020-09-17:
no, directly between a router/modem.
Sincerely, i cannot found "https://github.com/MichaIng/DietPi/archive/master.tar.gz" even from the browser

Joulinar, 2020-09-17:
I testes it from my RPi3B+ without issues

this is working fine
`curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/pre-patch_file -o pre-patch_file`

No issues to download this file from my Windows desktop
`https://github.com/MichaIng/DietPi/archive/master.tar.gz`

Can you try to change your upstream DNS server to something like Google DNS, Cloudflare or Quad9?

jumpmad, 2020-09-17:
The problem seems to be in the connectivity, the local network works but even if I change the DNS it doesn't go out on the internet. Thank you very much, I close the ticket here, I find a solution to the connection problem and try again. Thanks again.

Joulinar, 2020-09-17:
Ok I will close the issue for now. Feel free to reopen if needed.

closed 2020-09-17 by Joulinar