MichaIng/DietPi #3778: Software-Request: AdGuard Home

Gill-Bates, 2020-09-17:
One of the main reasons for a Raspi is the project "Pihole". Dietpi has made a lot of investments to get Pi-Hole up and running.

But the more I deal with AdGuard Home, the more I have doubts whether Pi-Hole is still the right tool. There have already been requests regarding AdGuard ([Link](https://dietpi.com/phpbb/viewtopic.php?t=5186)). The installation is very easy:

```
wget https://static.adguard.com/adguardhome/edge/AdGuardHome_linux_armv7.tar.gz
tar xfz AdGuardHome_linux_armv7.tar.gz
mv AdGuardHome /etc/
cd /etc/AdGuardHome; sudo ./AdGuardHome -s install
```
The Setup Wizard is listen on Port 3333

Maybe someone can add it to the software catalog.

Joulinar, 2020-09-17:
already on FeatHub https://feathub.com/MichaIng/DietPi/+34

Gill-Bates, 2020-11-20:
Yes, but attached with Installation Script

manilx, 2021-02-10:
> One of the main reasons for a Raspi is the project "Pihole". Dietpi has made a lot of investments to get Pi-Hole up and running.
> 
> But the more I deal with AdGuard Home, the more I have doubts whether Pi-Hole is still the right tool. There have already been requests regarding AdGuard ([Link](https://dietpi.com/phpbb/viewtopic.php?t=5186)). The installation is very easy:
> 
> **Edge Channel**
> 
> ```
> wget https://static.adguard.com/adguardhome/edge/AdGuardHome_linux_armv6.tar.gz
> wget https://static.adguard.com/adguardhome/edge/AdGuardHome_linux_armv7.tar.gz
> ```
> 
> **Stable Channel**
> 
> ```
> wget https://static.adguard.com/adguardhome/release/AdGuardHome_linux_armv6.tar.gz
> wget https://static.adguard.com/adguardhome/release/AdGuardHome_linux_armv7.tar.gz
> ```
> 
> Installation (straight forward)
> 
> ```
> tar xfz AdGuardHome_linux_armvX.tar.gz
> mv AdGuardHome /etc/
> cd /etc/AdGuardHome; sudo ./AdGuardHome -s install
> ```
> 
> The Setup Wizard is listen on Port 3000. If the `AdGuardHome.yaml` file is missing, the installation wizard appears. You could also include a Dietpi-optimized YAML file.
> 
> Adding` [/fritz.box/]192.168.178.1:53` as Upstream-Server will send all local Domains to the Fritzbox (a.k.a Conditional Forwarding)
> 
> Maybe someone can add it to the software catalog.

Hi, 

I want to try this on my pie instead of pihole. Thx for the install steps.

One question: how do I update the installation after it's working?
Via the usual apt update/upgrade?

Thx in advance

Joulinar, 2021-02-10:
If I'm not mistaken, AGH is updated from inside the Gui.

manilx, 2021-02-10:
> If I'm not mistaken, AGH is updated from inside the Gui.

OK. Will test it later today after making a clone of my dietpi pen and then working with that clone-pen, so I can come back if I have issues or don't like it ;)

Gill-Bates, 2021-02-10:
@Joulinar and @manilx You are both correct! When an Update is available, a banner inside the GUI pops up to start the Update Task.

Joulinar, 2021-02-10:
well it is a statistic done by AGH themselves 😉 
Don't trust a statistic you have not manipulated yourself 🤣

manilx, 2021-02-10:
Uninstalled piehole installed AdGuard as described here: https://github.com/AdguardTeam/AdGuardHome
Basically just run this: curl -sSL https://raw.githubusercontent.com/AdguardTeam/AdGuardHome/master/scripts/install.sh | sh

Running OK now. testing.

closed 2021-06-26 by MichaIng

dunxd, 2022-01-08:
Where is the Adguard.yaml file after install through dietpi-software?  It isn't in /etc/ 

Would be great if the configuration tab at https://dietpi.com/docs/software/dns_servers/#adguard-home gave the location for the Adguard.yaml file so that port etc can easily be changed.

Joulinar, 2022-01-08:
There you go

 `/mnt/dietpi_userdata/adguardhome/AdGuardHome.yaml`