MichaIng/DietPi #3779: Pi-hole installation failed

cc13com, 2020-09-18:
### Creating a bug report/issue
bca50705-6c4a-47f3-a8c1-b191f2b50ac3

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=32
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux`
- SBC model | `RPi Zero W (armv6l)`
- Power supply used | (N/A)
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Pi-hole
- The software was installed but the FTL part could not updated. I deinstalled Pi-hole and tried to install it new. During this I get the error message and the installation failed.

- Can this issue be replicated on a fresh installation of DietPi? Not tried.

- Bug report ID | `bca50705-6c4a-47f3-a8c1-b191f2b50ac3`

#### Steps to reproduce

1. ...select pi-hole to install
2. ...installation stops in the middle with running the script ./install-sh

#### Expected behaviour

- ... I want an installed Pi-hole, like before

#### Actual behaviour

- ...Installtion stops

#### Extra details

- ...

MichaIng, 2020-09-18:
Many thanks for your report. Please include the error message, at best the output of the whole Pi-hole installer output. I'll have a look into your bug report.

cc13com, 2020-09-18:
- Command: ./install.sh --disable-install-webserver
                        │  - Exit code: 1
                        │  - DietPi version: v6.32.2 (MichaIng/master) | HW_MODEL: 1 | HW_ARCH: 1 | DISTRO: 5
                        │  - Image creator: DietPi Core Team
                        │  - Pre-image: Raspbian Lite
                        │  - Error log:
                        │
                        │   [^[[1;32m✓^[[0m] Root user check
                        │
                        │         ^[[1;32m.;;,.
                        │         .ccccc:,.
                        │          :cccclll:.      ..,,
                        │           :ccccclll.   ;ooodc
                        │            'ccll:;ll .oooodc
                        │              .;cll.;;looo:.
                        │                  ^[[1;31m.. ','.
                        │                 .',,,,,,'.
                        │               .',,,,,,,,,,.
                        │             .',,,,,,,,,,,,....
                        │           ....''',,,,,,,'.......
                        │         .........  ....  .........
                        │         ..........      ..........
                        │         ..........      ..........
                        │         .........  ....  .........
                        │           ........,,,,,,,'......
                        │             ....',,,,,,,,,,,,.
                        │                .',,,,,,,,,'.
 [i] Update local cache of available packages...^M^[[K  [^[[1;32m✓^[[0m] Update local cache of available packages
                        │   [i] Existing PHP installation detected : PHP version 7.3.19-1~deb10u1
                        │   [^[[1;32m✓^[[0m] Disk space check
                        │
                        │   [i] Checking apt-get for upgraded packages...^M^[[K  [^[[1;32m✓^[[0m] Checking apt-get for upgraded packages... up t
                        │ date!
                        │
                        │   [i] Installer Dependency checks...
                        │   [i] Checking for dhcpcd5...^M^[[K  [^[[1;32m✓^[[0m] Checking for dhcpcd5
                        │   [i] Checking for git...^M^[[K  [^[[1;32m✓^[[0m] Checking for git
                        │   [i] Checking for iproute2...^M^[[K  [^[[1;32m✓^[[0m] Checking for iproute2
                        │   [i] Checking for whiptail...^M^[[K  [^[[1;32m✓^[[0m] Checking for whiptail
                        │   [i] Checking for dnsutils...^M^[[K  [^[[1;32m✓^[[0m] Checking for dnsutils

Here it ends and open the menu to re-run the last comment, etc.

root@DietPi:/home/dietpi# dig +short -t txt versions.pi-hole.net @ns1.pi-hole.net | tr -d '"'
root@DietPi:/home/dietpi#

Nothing happens

MichaIng, 2020-09-18:
Interesting, the dig command should output the following:
```
2020-09-18 13:30:07 root@VM-Buster:/tmp# dig +short -t txt versions.pi-hole.net @ns1.pi-hole.net | tr -d '"'
Raspbian=9,10 Ubuntu=16,18,20 Debian=9,10 Fedora=31,32 CentOS=7,8
```
The supported OS check is done via DNS TXT entry, so a very quick method that doesn't rely on any own webserver or such but requires the name server to be available only, extremely reliable. I was thinking to apply a similar method for DietPi as well to allow update checks and such independently of our own server. Although Pi-hole runs an own name server, which can theoretically be offline, compared to a world-wide reachable public one spread across multiple machines around the globe. However enough praise for their method 😄.

Does `dig` work for other lookups?
```
dig dietpi.com
```

cc13com, 2020-09-18:
root@DietPi:/home/dietpi# dig dietpi.com
Speicherzugriffsfehler

root@DietPi:/home/dietpi# dig
Speicherzugriffsfehler

Means "memory access error"

MichaIng, 2020-09-18:
I just found that there is even an option for "memory usage debugging": https://manpages.debian.org/buster/dnsutils/dig.1.en.html#OPTIONS
```
dig -m
```

Joulinar, 2020-09-18:
did you tried to reboot? Maybe this could solve the memory issue?

cc13com, 2020-09-18:
root@DietPi:/home/dietpi# dig -m
Speicherzugriffsfehler

After the not working installation the Dietpi forced a reboot. today 3-4 times

MichaIng, 2020-09-18:
Can you please show the international error message, that serves for better web search. What I found so far is that it might not even be an error from `dig`, but a OS/kernel level error, so check that as well:
```
export LC_ALL='C.UTF-8'
dig -m
export LC_ALL='de_DE.UTF-8'
dmesg -l emerg,alert,crit,err
```

cc13com, 2020-09-18:
root@DietPi:/home/dietpi# export LC_ALL='C.UTF-8'
root@DietPi:/home/dietpi# dig -m
Segmentation fault
root@DietPi:/home/dietpi# export LC_ALL='de_DE.UTF-8'
root@DietPi:/home/dietpi# dmesg -l emerg,alert,crit,err
[    2.067171] vc_vchi_sm_init: failed to open VCHI service (-1)
[    2.067185] [vc_sm_connected_init]: failed to initialize shared memory service
[   10.366235] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   10.366246] [vc_sm_connected_init]: failed to initialize shared memory service
[   10.946043] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[   12.269177] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   12.531900] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   12.532070] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   12.534913] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   17.956551] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   19.908830] CIFS VFS: Error connecting to socket. Aborting operation.
[   19.908890] CIFS VFS: cifs_mount failed w/return code = -2
[   29.009821] debugfs: File 'le_min_key_size' in directory 'hci0' already present!
[   29.009857] debugfs: File 'le_max_key_size' in directory 'hci0' already present!

MichaIng, 2020-09-18:
A it's a regular segfault, okay.

Hmm, you are on ARMv6. Can you show which APT repositories are installed:
```
cat /etc/apt/sources.list{,.d/*.list}
```

cc13com, 2020-09-18:
root@DietPi:/var/www/html# cat /etc/apt/sources.list{,.d/*.list}
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
deb https://archive.raspberrypi.org/debian/ buster main ui

MichaIng, 2020-09-18:
Okay, definitely the package from Raspbian. Just to be sure, could you try:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/b/bind9/dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
dpkg -i dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
rm dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
dig -m
```

cc13com, 2020-09-18:
root@DietPi:/var/www/html# cd /tmp/
root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/b/bind9/dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
--2020-09-18 15:14:02--  http://raspbian.raspberrypi.org/raspbian/pool/main/b/bind9/dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
Auflösen des Hostnamens raspbian.raspberrypi.org (raspbian.raspberrypi.org)… 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Verbindungsaufbau zu raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: http://debian.bio.lmu.de/raspbian/raspbian/pool/main/b/bind9/dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb [folgend]
--2020-09-18 15:14:02--  http://debian.bio.lmu.de/raspbian/raspbian/pool/main/b/bind9/dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
Auflösen des Hostnamens debian.bio.lmu.de (debian.bio.lmu.de)… 141.84.43.22, 2001:4ca0:4300::1:22
Verbindungsaufbau zu debian.bio.lmu.de (debian.bio.lmu.de)|141.84.43.22|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 360312 (352K) [application/x-debian-package]
Wird in »dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb« gespeichert.

dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf. 100%[=====================================================================================>] 351,87K  1,19MB/s    in 0,3s

2020-09-18 15:14:03 (1,19 MB/s) - »dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb« gespeichert [360312/360312]

root@DietPi:/tmp# dpkg -i dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
(Lese Datenbank ... 93809 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb ...
Entpacken von dnsutils (1:9.11.5.P4+dfsg-5.1+deb10u2) über (1:9.11.5.P4+dfsg-5.1+deb10u2) ...
dnsutils (1:9.11.5.P4+dfsg-5.1+deb10u2) wird eingerichtet ...
root@DietPi:/tmp# rm dnsutils_9.11.5.P4+dfsg-5.1+deb10u2_armhf.deb
root@DietPi:/tmp# dig -m
Speicherzugriffsfehler
root@DietPi:/tmp# export LC_ALL='C.UTF-8'
root@DietPi:/tmp# dig -m
Segmentation fault

Sorry for changing the language to late. Should I do it again?

MichaIng, 2020-09-18:
> Sorry for changing the language to late. Should I do it again?

Nope no issues at all, ich spreche dieselbe Sprache, it was only helpful for the dig error message to now know that it is a "regular" segmentation fault. Never saw the German translation for this 😄.

Okay, probably someone else with RPi 1 or Zero needs to try replicating it. The build is quite fresh from `2020-08-28`, maybe broken for all ARMv6 models.
____________
To get your Pi-hole up, please apply the workaround as mentioned above:
```
sed -i 's/PIHOLE_SKIP_OS_CHECK=1/PIHOLE_SKIP_OS_CHECK=true/' /boot/dietpi/dietpi-software
dietpi-software install 93
```

cc13com, 2020-09-18:
Ok, now I got my Pi-hole back. Thanks a lot. If you need me for more testing, ping me.

Do I have to change the `Skip_OS_Check` back to no?

Danke @MichaIng für deine geduldige und schnelle Hilfe.

MichaIng, 2020-09-18:
> Do I have to change the `Skip_OS_Check` back to no?

Nope, it is wanted there. As said I explicitly wanted to skip the check but either they changed the required value (from `1` to `true`) or I added it wrong in the first place 😅. The other way round, when there is a Pi-hole update available, before running `pihole -up`, you likely need to do `export PIHOLE_SKIP_OS_CHECK=true` first, if the `dig` segfault was not solved somehow.

Joulinar, 2020-09-18:
@MichaIng 
I got a `RPi B (armv6l)` from a friend of mine and was testing `PiHole` installation. For me it was working without issues.

<details>
  <summary>Install log file attached. Click to expand!</summary>

```
 root@DietPi1:~# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[  OK  ] DietPi-Software | Free space check: path=/ | available=26217 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT

[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update
[ INFO ] DietPi-Software | APT upgrade, please wait...
[  OK  ] DietPi-Software | APT upgrade

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

[ INFO ] DietPi-Software | Lighttpd will be installed
[ INFO ] DietPi-Software | PHP will be installed
[ INFO ] DietPi-Software | SQLite will be installed
[ INFO ] DietPi-Software | Git Client will be installed

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Git Client: git clone etc

[ INFO ] DietPi-Software | APT install for: git, please wait...
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package perl-modules-5.28.
(Reading database ... 20969 files and directories currently installed.)
Preparing to unpack .../0-perl-modules-5.28_5.28.1-6_all.deb ...
Unpacking perl-modules-5.28 (5.28.1-6) ...
Selecting previously unselected package libgdbm6:armhf.
Preparing to unpack .../1-libgdbm6_1.18.1-4_armhf.deb ...
Unpacking libgdbm6:armhf (1.18.1-4) ...
Selecting previously unselected package libgdbm-compat4:armhf.
Preparing to unpack .../2-libgdbm-compat4_1.18.1-4_armhf.deb ...
Unpacking libgdbm-compat4:armhf (1.18.1-4) ...
Selecting previously unselected package libperl5.28:armhf.
Preparing to unpack .../3-libperl5.28_5.28.1-6_armhf.deb ...
Unpacking libperl5.28:armhf (5.28.1-6) ...
Selecting previously unselected package perl.
Preparing to unpack .../4-perl_5.28.1-6_armhf.deb ...
Unpacking perl (5.28.1-6) ...
Selecting previously unselected package libcurl3-gnutls:armhf.
Preparing to unpack .../5-libcurl3-gnutls_7.64.0-4+deb10u1_armhf.deb ...
Unpacking libcurl3-gnutls:armhf (7.64.0-4+deb10u1) ...
Selecting previously unselected package libexpat1:armhf.
[... 622 lines cut ...]
Executing: /lib/systemd/systemd-sysv-install disable dhcpcd
Removed /etc/systemd/system/dhcpcd5.service.
Removed /etc/systemd/system/multi-user.target.wants/dhcpcd.service.
  [✓] Restarting DNS server
  [✓] Flushed /var/log/pihole.log
  [✓] Deleted  queries from database
  [i] Disabling logging...
  [✓] Restarting DNS server
  [✓] Logging has been disabled!
  [✓] New password set
[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : php7.3-fpm
[  OK  ] DietPi-Services | dietpi_controlled : lighttpd
[  OK  ] DietPi-Services | dietpi_controlled : cron
[  OK  ] DietPi-Survey | Sending survey data

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Software | System rebooting in 3 seconds to finalise DietPi-Software installations. Please wait...
```
</details>

It was a fresh/clean installation using latest RPi image.

MichaIng, 2020-09-18:
Thanks for testing. So then I am not sure what the issue with cc13com's system is, since we already reinstalled the binary. Of course we could go through every used library.

cc13com, 2020-09-21:
If you don't need anything else from my side I will re-install the whole system. I got a new PI4 last days and will use this for my new dietpi/pi-hole/... system.

MichaIng, 2020-09-21:
If you find time:
```
apt install strace
strace dig dietpi.com
```
But this is just out of interest and likely we are not able to derive the exact issue from that 😉.

Glad you got an RPi4 anyway which solves the issue altogether. Be sure to add some other use to it, as Pi-hole/DNS-only server it is quite some overkill, although the web UI will behave MUCH more fluent compared to RPi Zero from what I read 🙂.

Joulinar, 2020-09-21:
>  Be sure to add some other use to it, as Pi-hole/DNS-only server it is quite some overkill

You can never have enough power 😆

cc13com, 2020-09-21:
`
root@DietPi:/home/dietpi# strace dig dietpi.com
execve("/usr/bin/dig", ["dig", "dietpi.com"], 0xbec14da4 /* 17 vars */) = 0
brk(NULL)                               = 0x12c5000
uname({sysname="Linux", nodename="DietPi", ...}) = 0
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f95000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/bin/dig", 4096) = 12
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v6l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\250\3\0\0004\0\0\0"..., 512) = 512
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
_llseek(3, 8552, [8552], SEEK_SET)      = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
_llseek(3, 8232, [8232], SEEK_SET)      = 0
read(3, "A*\0\0\0aeabi\0\1 \0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 43) = 43
fstat64(3, {st_mode=S_IFREG|0644, st_size=9512, ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f93000
mmap2(NULL, 73772, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f53000
mprotect(0xb6f55000, 61440, PROT_NONE)  = 0
mmap2(0xb6f64000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6f64000
close(3)                                = 0
munmap(0xb6f95000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=30805, ...}) = 0
mmap2(NULL, 30805, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f8b000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libdns.so.1104", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\300)\2\0004\0\0\0"..., 512) = 512
_llseek(3, 2055088, [2055088], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 2054744, [2054744], SEEK_SET) = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 2055088, [2055088], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 2054744, [2054744], SEEK_SET) = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 2055088, [2055088], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 2054744, [2054744], SEEK_SET) = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 2055088, [2055088], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 2054744, [2054744], SEEK_SET) = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
_llseek(3, 2055088, [2055088], SEEK_SET) = 0
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1040) = 1040
_llseek(3, 2054744, [2054744], SEEK_SET) = 0
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 45) = 45
[... 943 lines cut ...]
set_tls(0xb42e1050)                     = 0
mprotect(0xb659c000, 8192, PROT_READ)   = 0
mprotect(0xb430d000, 4096, PROT_READ)   = 0
mprotect(0xb6802000, 4096, PROT_READ)   = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb42de000
mprotect(0xb444d000, 20480, PROT_READ)  = 0
mprotect(0xb682a000, 4096, PROT_READ)   = 0
mprotect(0xb4485000, 4096, PROT_READ)   = 0
mprotect(0xb44b0000, 4096, PROT_READ)   = 0
mprotect(0xb5eb0000, 4096, PROT_READ)   = 0
mprotect(0xb68bf000, 4096, PROT_READ)   = 0
mprotect(0xb602c000, 40960, PROT_READ)  = 0
mprotect(0xb627c000, 36864, PROT_READ)  = 0
mprotect(0xb63ff000, 8192, PROT_READ)   = 0
mprotect(0xb6422000, 4096, PROT_READ)   = 0
mprotect(0xb6438000, 4096, PROT_READ)   = 0
mprotect(0xb6452000, 4096, PROT_READ)   = 0
mprotect(0xb65b8000, 4096, PROT_READ)   = 0
mprotect(0xb6732000, 16384, PROT_READ)  = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_ACCERR, si_addr=0xb674175c} ---
+++ killed by SIGSEGV +++
Segmentation fault
`

I plan to have pi-hole, VPN, Kodi (runs on an old PI3 actual), FHEM on this PI4. And if possible in docker containers.

MichaIng, 2020-09-21:
> I plan to have pi-hole, VPN, Kodi (runs on an old PI3 actual), FHEM on this PI4. And if possible in docker containers.

Sounds great! I see a new software request, German product home automation: https://fhem.de/

The strace indeed does not tell me something. All shared libraries have been loaded successfully, if it would have failed while loading one, that would have been the bread crumbs to follow, but there is the last `close(3)` and it then fails after some [mprotect](https://manpages.debian.org/buster/manpages-dev/mprotect.2.en.html) changes which obviously didn't prevent it from running into an access violation afterwards. Strange thing, no idea how to debug this, forwarding to dig devs or StackExchange is what I would do when it'd be still relevant.

Joulinar, 2020-09-21:
FHEM installation seems to be quite easy 😄 

- Import repository gpg key:
`wget -qO - http://debian.fhem.de/archive.key | apt-key add -`
- Add repository to /etc/apt/sources.list:
`deb http://debian.fhem.de/nightly/ /`
- Update your package administration:
`apt-get update`
- Install fhem:
`apt-get install fhem`

cc13com, 2020-09-21:
I have fhem installed on the same RPi Zero like pi-hole/OpenVPN on DietPi. After playing with some docker containers on my NAS and on a VM from my provider I think it's time to combine this on my new RPi4.

All is running fine on the Zero except the OpenVPN performance and the weekly backup of the system to my NAS because of the slow wifi connection. ;-)

Joulinar, 2020-09-26:
Hi guys, I guess this issue can be closed for now?

closed 2020-09-26 by cc13com

MichaIng, 2020-09-26:
Yes, from my end impossible to say where the segfault comes from and definitely an issue of the particular system, likely some random file corruption or something similar.