MichaIng/DietPi #3782: Error on HA installation

sdomotica, 2020-09-20:
### Creating a bug report/issue

#### Required Information
- DietPi version 
	G_DIETPI_VERSION_CORE=6
	G_DIETPI_VERSION_SUB=32
	G_DIETPI_VERSION_RC=2
	G_GITBRANCH='master'
	G_GITOWNER='MichaIng'

- Kernel version
Linux sdomotica 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

- SBC model 
RPi 3 Model B (armv7l)

#### Additional Information (if applicable)
- Home Assistant (0.115.2)
- freshly
- Can this issue be replicated on a fresh installation of DietPi?
yes

#### Extra details

Systemctl log

pi@sdomotica:~$ sudo journalctl -f
-- Logs begin at Fri 2020-06-12 12:53:56 CEST. --
Sep 20 18:55:00 sdomotica usermod[13471]: add 'homeassistant' to group 'i2c'
Sep 20 18:55:00 sdomotica usermod[13471]: add 'homeassistant' to shadow group 'i2c'
Sep 20 18:55:07 sdomotica sudo[13528]:     root : TTY=pts/0 ; PWD=/tmp/DietPi-Software ; USER=homeassistant ; COMMAND=/bin/dash -c
                                       . /home/homeassistant/pyenv-activate.sh
                                       pyenv install 3.8.0
                                       pyenv local 3.8.0
                                       pip3 install -U pip wheel
                                       [ -z "" ] || pip3 install
                                       pip3 install homeassistant
Sep 20 18:55:07 sdomotica sudo[13528]: pam_unix(sudo:session): session opened for user homeassistant by pi(uid=0)
Sep 20 19:16:59 sdomotica homeassistant[27174]: 2020-09-20 19:16:59 ERROR (SyncWorker_2) [homeassistant.util.package] Unable to install package pillow==7.2.0: ERROR: Command errored out with exit status 1:
Sep 20 19:16:59 sdomotica homeassistant[27174]:    command: /home/homeassistant/.pyenv/versions/3.8.0/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-0yp792vv/pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-0yp792vv/pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-gmji82pf
Sep 20 19:16:59 sdomotica homeassistant[27174]:        cwd: /tmp/pip-install-0yp792vv/pillow/
Sep 20 19:16:59 sdomotica homeassistant[27174]:   Complete output (170 lines):
Sep 20 19:16:59 sdomotica homeassistant[27174]:   running bdist_wheel
Sep 20 19:16:59 sdomotica homeassistant[27174]:   running build
Sep 20 19:16:59 sdomotica homeassistant[27174]:   running build_py
Sep 20 19:16:59 sdomotica homeassistant[27174]:   creating build
Sep 20 19:16:59 sdomotica homeassistant[27174]:   creating build/lib.linux-armv7l-3.8
Sep 20 19:16:59 sdomotica homeassistant[27174]:   creating build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/features.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/_version.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/_util.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/_tkinter_finder.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/_binary.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/__main__.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/__init__.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/XpmImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/XbmImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/WebPImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/WalImageFile.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/TiffTags.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/TiffImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/TgaImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/TarIO.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/SpiderImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/PyAccess.py -> build/lib.linux-armv7l-3.8/PIL
Sep 20 19:16:59 sdomotica homeassistant[27174]:   copying src/PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-3.8/PIL
[... 671 lines cut ...]
Sep 20 19:18:25 sdomotica homeassistant[27174]:     During handling of the above exception, another exception occurred:
Sep 20 19:18:25 sdomotica homeassistant[27174]:
Sep 20 19:18:25 sdomotica homeassistant[27174]:     Traceback (most recent call last):
Sep 20 19:18:25 sdomotica homeassistant[27174]:       File "<string>", line 1, in <module>
Sep 20 19:18:25 sdomotica homeassistant[27174]:       File "/tmp/pip-install-8cxx099z/pillow/setup.py", line 918, in <module>
Sep 20 19:18:25 sdomotica homeassistant[27174]:         raise RequiredDependencyException(msg)
Sep 20 19:18:25 sdomotica homeassistant[27174]:     __main__.RequiredDependencyException:
Sep 20 19:18:25 sdomotica homeassistant[27174]:
Sep 20 19:18:25 sdomotica homeassistant[27174]:     The headers or library files could not be found for jpeg,
Sep 20 19:18:25 sdomotica homeassistant[27174]:     a required dependency when compiling Pillow from source.
Sep 20 19:18:25 sdomotica homeassistant[27174]:
Sep 20 19:18:25 sdomotica homeassistant[27174]:     Please see the install instructions at:
Sep 20 19:18:25 sdomotica homeassistant[27174]:        https://pillow.readthedocs.io/en/latest/installation.html
Sep 20 19:18:25 sdomotica homeassistant[27174]:
Sep 20 19:18:25 sdomotica homeassistant[27174]:     ----------------------------------------
Sep 20 19:18:25 sdomotica homeassistant[27174]: ERROR: Command errored out with exit status 1: /home/homeassistant/.pyenv/versions/3.8.0/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8cxx099z/pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8cxx099z/pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-d4a97x6f/install-record.txt --single-version-externally-managed --prefix '' --user --prefix= --compile --install-headers /mnt/dietpi_userdata/homeassistant/deps/include/python3.8/pillow Check the logs for full command output.
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Setup failed for image: Requirements for image not found: ['pillow==7.2.0'].
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of person. Setup failed for dependencies: image
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Setup failed for person: Could not set up all dependencies.
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of onboarding. Setup failed for dependencies: person
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Setup failed for onboarding: Could not set up all dependencies.
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of frontend. Setup failed for dependencies: onboarding
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Setup failed for frontend: Could not set up all dependencies.
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of safe_mode. Setup failed for dependencies: frontend
Sep 20 19:18:25 sdomotica homeassistant[27174]: 2020-09-20 19:18:25 ERROR (MainThread) [homeassistant.setup] Setup failed for safe_mode: Could not set up all dependencies.

Joulinar, 2020-09-20:
Hi,

many thanks for your report. Basically already known. At least you have following error:

`Sep 20 19:16:59 sdomotica homeassistant[27174]: main.RequiredDependencyException: jpeg`

this might fit to https://github.com/MichaIng/DietPi/pull/3781 and you are missing `libjpeg-dev` package.

pls can you try `apt install libjpeg-dev` and restart installation afterwards.

sdomotica, 2020-09-20:
I do these
sudo apt install libjpeg-dev
sudo dietpi-software reinstall 157
and solved.
Thank you so much
Sandro

closed 2020-09-20 by sdomotica