MichaIng/DietPi #3783: UX: Colorize the Shell

Gill-Bates, 2020-09-21:
The Bash looks pretty boring. I would love to highligt the Path in Bash

`nano ~/.bashrc`

```
# Colorize Shell
PS1='\[\033[1;36m\]\u\[\033[1;31m\]@\[\033[1;32m\]\h:\[\033[1;35m\]\w\[\033[1;31m\]\$\[\033[0m\] '
```
Now, much better 👍 
[image]

Can't you make this default?

Joulinar, 2020-09-21:
Hi,

many thanks for your request. Honestly I'm not sure if everybody like to have such a colourful command line as it is quite specific 😉

MichaIng, 2020-09-21:
On some images, where the default `/etc/skel/.bashrc` is installed for root (which differs from default `/root/.bashrc` installed by debootstrap or Debian installer) the prompt is green. Yeah I guess that is pretty much a question of taste and a the console/terminal one uses. Not all of them support (all) colours and on a serial console you would see the raw colour codes, which is really distracting, especially with a large amount of different colours like in your example. I started to remove colours from our script outputs in some cases, but still need to find a way to reliably identify a serial console or directly colour code support.

I can imagine to add this in a way together with this: #2651
So basically allow to define a set of colour codes which are then read by DietPi-Globals and used to colourise several console outputs in a consistent way. So who uses a serial console only, or one with limited colour code support, can easily disable those at a single place for all outputs that we can control.

I personally leave the command prompt without colour but add those to the `ls` command to quickly identify file types, UNIX permissions, existence of symlink targets etc, so this might be somehow included as well:
```
# Colorize "ls" command
eval "$(dircolors)"
alias ls='ls -A --color=auto'
alias l='ls -lh'
alias ll='l -L'
```
_+ shortcuts for ls list view + showing dot files by default 😉_
[image]

What you can see is that I instead add timestamps to the PS1 so that I can track how long a long-running command took or when I executed what, with certain limitation as the timestamp in from of a command is actually the one from when the previous command ended 😄.