MichaIng/DietPi #3785: Cannot apt update after running Dietpi-Cleaner APT

AlessandroBagnoli, 2020-09-22:
As the title says, after running Dietpi-Cleaner APT, I cannot apt update anymore:

root@DietPi:~# apt update
Trovato:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Trovato:2 https://apt.sonarr.tv develop InRelease
Trovato:3 https://download.mono-project.com/repo/debian raspbianbuster InRelease
Trovato:4 https://archive.raspberrypi.org/debian buster InRelease
Ignorato:5 https://download.webmin.com/download/repository sarge InRelease
Trovato:6 https://download.webmin.com/download/repository sarge Release
Error: No information about packages! (Maybe no deb entries?)
Lettura elenco dei pacchetti... Fatto
E: Problem executing scripts APT::Update::Post-Invoke-Success 'test -x /usr/bin/apt-show-versions || exit 0 ; apt-show-versions -i'
E: Sub-process returned an error code

so APT is now broken. Any solution? Thanks in advance.

Joulinar, 2020-09-22:
Hi,

many tanks for your message. Basically already know. Pls can you try following

```
echo 'Acquire::GzipIndexes "false";' > /etc/apt/apt.conf.d/98dietpi-webmin
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

reference https://github.com/MichaIng/DietPi/commit/5be3fa0258bc154c5d257a1cec5223dfa8a6b360#diff-d92a6ee04e02fd2a2dc23d5bec3e6a98

AlessandroBagnoli, 2020-09-22:
Thank you @Joulinar , worked flawlessly. Issue closed.

closed 2020-09-22 by AlessandroBagnoli