MichaIng/DietPi #3786: Domoticz

Raeymeister, 2020-09-22:
### Creating a software request

#### Formal software information
- Domoticz
- Free and Open Source Home Automation
- https://www.domoticz.com/
- GitHub URL (if available)
- https://www.domoticz.com/downloads/ 

#### Are there similar/alternative software titles available with DietPi-Software?

- Home Assistant

#### What makes your requested software better than the above solutions, if available?

- Open Source Home Automation platform
- Support for other devices than Home Assistant (eg. Bticino / Xiaomi)

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

When making this platform available in the "Optimized Software" of Dietpi, this would centralise and therefore facilitate installation of all software through the dietpi-software interface.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1.  Following the downloads page, it is as simple as:
       'curl -sSL install.domoticz.com | sudo bash'

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- I don't know bash, but am willing to create a youtube video on how to use Dietpi combined with Domoticz

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-09-22:
Didn't you see it? 😄
```
2020-09-22 22:46:29 root@micha:/tmp# dietpi-software list | grep domoticz
id 140 | =0 | domoticz: open-source home automation platform | | https://dietpi.com/phpbb/viewtopic.php?p=23065#p23065
```

Raeymeister, 2020-09-23:
Really!!!  Sorry man, I must be going blind. 
This is fantastic news, I owe you!

Joulinar, 2020-09-23:
Guys, I'm going to close this for now. Feel free to reopen if needed.

closed 2020-09-23 by Joulinar