MichaIng/DietPi #3787: Network went down - had to power cycle

danergo, 2020-09-24:
Hi,

I'm using DietPi (thanks!) on a PI Compute module.
Last night it dropped the eth0 connection, and I had to restart by power recycle. (It is a headless board).

Fortunately DietPi's log was set to "Full", so I still have all the logs, but I could not find any valuable information in them.
Could you please help me to identify what was the reason of the eth0 failure?

MichaIng, 2020-09-24:
Many thanks for your report.

`/var/log/syslog` and `/var/log/daemon.log` should contain the relevant ones, if present. Probably you can find and paste the entries ~10 minutes before reboot. Those are likely mixed with the logs of the next boot until time sync shifted it.

danergo, 2020-09-24:
In the syslog there are DHCPREQUEST and DHCPACK lines followed by a "bound to" line. Then the manual reboot is visible with kernel messages. Nothing special.

daemon.log also doesn't offer too much details, on why eth0 is lost.

MichaIng, 2020-09-24:
Could you provide the logs, in case masking private/identifying parts? It's always good to have more eyes looking on them to not miss some probably important details 😉.

If the last thing was a DHCP lease, then probably something with this went wrong, double check IP, gateway and DNS nameserver, in case applied.

How did you recognise the "dropped eth0" actually, have you tried to ping its IP?

danergo, 2020-09-24:
Yes, I tried to ping it, and also tried to SSH, none worked. Will provide logs a bit later, thank you!

MichaIng, 2020-12-17:
Marking this as closed. Feel free to reopen if issue presists.

danergo, 2021-03-20:
Hi! I've experienced the very same issue again. Which logs could help investigating this further?

MichaIng, 2021-03-20:
Last time I forgot /var/log/kernel.log. Does it contain anything of interest?

Also you might monitor central system stats like RAM usage, CPU temperature and such.

danergo, 2021-03-24:
Hi! 
Tried to check those logs, but don't know. It seems correct, no heavy changes or whatever.

I would happily share my logs with you, but don't want to put it into public domain. If you give me your email, I'll send it over and if you find something we can put the relevant parts here as memento for the future reader.

Thank you!

MichaIng, 2021-03-24:
Easiest is to upload a bug report: `dietpi-bugreport`
Then post the ID here.

danergo, 2021-03-24:
Here it is: 646e6faf-cfc5-4df3-a37c-aa94c758c999

MichaIng, 2021-03-24:
Hmm, I also cannot see any obvious issue or logs related to a crash or something 🤔. Do you remember the exact time when this happend last time?

What I see is that your system is pretty outdated. I recommend that you do package and a DietPi update. Since it has been a while, at best do APT updates separately:
```
apt update
apt upgrade
apt full-upgrade
dietpi-update 1
```

Since it is a headless board, after you updated the DietPi scripts, you could enable headless mode to reduce some overhead kernel and memory-wise:
```
/boot/dietpi/func/dietpi-set_hardware headless enable
```

danergo, 2021-03-24:
Thanks! I have no idea of the exact time when this incident happened. I know however that it was actually power-cycled on 14th March, so anything before that date might be the cause.

I'll do the update later, as it's a remote board, and I don't want to stop it, because manually power-cycling it is very hard. Once I have less distance to it, I'll upgrade it for sure.

MichaIng, 2021-03-24:
Ah okay. At least when using `dietpi-update 1` (updates directly, skips the menu), then no reboot is forced. On current versions, it's as well not when starting it from the menu, but I'm currently not sure when we changed it.
APT upgrades generally do not require a reboot, aside of kernel upgrades. But the last kernel version for RPi on Raspbian Stretch is from 2019, so I think you have the latest available one already.

When you find time and have the RPi by your side. You might as well want to upgrade to Raspbian Buster, as Stretch will be deprecated this summer with the next stable Debian Bullseye release. It ships much never kernel and packages being basically 2 years newer. But that's larger task.

Joulinar, 2021-04-02:
@danergo any update on this? Did you performed the system update as recommended by @MichaIng ?

danergo, 2021-04-04:
@Joulinar: no, not yet. I'm gonna be in safe distance to this board earliest in beginning of May. Until then we can safely close this as there is no more things to do and even after the upgrade it either won't come again or will come again after months of usage.

Joulinar, 2021-04-04:
OK going to close this one, feel free to reopen of needed 😃

closed 2021-04-04 by Joulinar

danergo, 2021-04-05:
@MichaIng, @Joulinar: an obvious idea came into my mind. Can I use RPi's hardware watchdog with DietPi?

It seems accessible with `/dev/watchdog0`, and there are tons of tutorials on how to do it (very easy).
But I need to make this 100% sure. I need to know if there is anything makes using watchdog with DietPi a bad idea.

Please confirm this for me guys.

MichaIng, 2021-04-05:
You can use it like on Raspberry Pi OS. It's the same kernel etc.