MichaIng/DietPi #3788: openjdk version "11.0.8" 2020-07-14 LTS on Raspberry Pi Zero W (armv6)

maartenlangeveld, 2020-09-24:
Managed to get Java OpenJDK 11 working on my Raspberry Pi Zero W (armv6):

Just followed the instruvtions on this [site](https://webtechie.be/post/2020-08-27-azul-zulu-java-11-and-gluon-javafx-11-on-armv6-raspberry-pi/).

MichaIng, 2020-09-24:
Many thanks for providing this information, that is awesome.

I'm wondering about the statements then that Java 11 is not ARMv6-compatible, which are then obviously wrong and only the Raspbian-build isn't. This basically means that it would be possible to ship an ARMv6 compatible build with either Raspbian or RPi repository. Let's see if we can get some attention on this.

maartenlangeveld, 2020-10-10:
Hi @MichaIng,

Found this [site](https://jelastic.com/blog/adoptopenjdk-liberica-zulu-corretto-openj9-graalvm/) listing a variety of OpenJDK distributions. 

Only Zulu OpenJDK11 works on my RPi-Zero-W (armv6):

```
AdoptOpenJDK - can't run java after instalation ("Segmentation fault")
Liberica JDK - can't run java after instalation ("Illegal instruction")
Zulu - works (!)
Corretto - no armv6/32 bit build available
OpenJ9 - Is AdoptOpenJDK with OpenJ9 VM (not available for armv6 / 32 bit)
```

MichaIng, 2021-06-23:
Is the issue actually still present with the latest `openjdk-11-jre-headless` package? I just recognised that the workaround (installing Java 8 on ARMv6) did not work, as we checked `(( G_HW_ARCH == 1 ))` instead of `(( $G_HW_ARCH == 1 ))` 😅. So Java 11 was installed on all ARMv6 Buster systems, but I don't remember any report where this caused an issue. There has been a large amount of minor updates, including a lot of compilation flag changes and such: https://metadata.ftp-master.debian.org/changelogs//main/o/openjdk-11/openjdk-11_11.0.11+9-1~deb10u1_changelog

MichaIng, 2021-06-23:
@Joulinar 
If you find time (fast than Maarten), could you do a quick test of [Komga](https://dietpi.com/docs/software/media/#komga) or [Airsonic](https://dietpi.com/docs/software/media/#airsonic) on RPi1? I'm not keen to fix a workaround that is not required anymore and have ARMv6 models unnecessarily stuck with Java 8 😃.

Joulinar, 2021-06-23:
@MichaIng 
ok `Komga` done. It seems to be running

```
root@DietPi1:~# journalctl -u komga.service
-- Logs begin at Wed 2021-06-23 18:09:24 CEST, end at Wed 2021-06-23 18:36:18 CEST. --
Jun 23 18:35:28 DietPi1 systemd[1]: Started Komga (DietPi).
Jun 23 18:36:18 DietPi1 java[4985]:  ____  __.
Jun 23 18:36:18 DietPi1 java[4985]: |    |/ _|____   _____    _________
Jun 23 18:36:18 DietPi1 java[4985]: |      < /  _ \ /     \  / ___\__  \
Jun 23 18:36:18 DietPi1 java[4985]: |    |  (  <_> )  Y Y  \/ /_/  > __ \_
Jun 23 18:36:18 DietPi1 java[4985]: |____|__ \____/|__|_|  /\___  (____  /
Jun 23 18:36:18 DietPi1 java[4985]:         \/           \//_____/     \/
Jun 23 18:36:18 DietPi1 java[4985]: Version: 0.101.1
```

but there is no LISTEN port 

```
root@DietPi1:~# ss -tulpn | grep LISTEN
tcp     LISTEN   0        1000             0.0.0.0:22            0.0.0.0:*       users:(("dropbear",pid=347,fd=3))                                      
tcp     LISTEN   0        1000                [::]:22               [::]:*       users:(("dropbear",pid=347,fd=4))                                      
root@DietPi1:~#
```

after a couple of minutes it went into failed state

```
Jun 23 18:40:44 DietPi1 java[4985]: 2021-06-23 18:40:44.590 ERROR 4985 --- [           main] com.zaxxer.hikari.pool.HikariPool        : HikariPool-1 - E
xception during pool initialization.
Jun 23 18:40:44 DietPi1 java[4985]: java.sql.SQLException: Error opening connection
Jun 23 18:40:44 DietPi1 java[4985]:         at org.sqlite.SQLiteConnection.open(SQLiteConnection.java:239) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:44 DietPi1 java[4985]:         at org.sqlite.SQLiteConnection.<init>(SQLiteConnection.java:61) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:44 DietPi1 java[4985]:         at org.sqlite.jdbc3.JDBC3Connection.<init>(JDBC3Connection.java:28) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:44 DietPi1 java[4985]:         at org.sqlite.jdbc4.JDBC4Connection.<init>(JDBC4Connection.java:21) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:44 DietPi1 java[4985]:         at org.sqlite.JDBC.createConnection(JDBC.java:115) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:44 DietPi1 java[4985]:         at org.sqlite.JDBC.connect(JDBC.java:90) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
```
```

Jun 23 18:40:47 DietPi1 java[4985]: Caused by: java.lang.Exception: No native library found for os.name=Linux, os.arch=armv6, paths=[/usr/java/packages/
lib/arm:/usr/lib/arm-linux-gnueabihf/jni:/lib/arm-linux-gnueabihf:/usr/lib/arm-linux-gnueabihf:/usr/lib/jni:/lib:/usr/lib]
Jun 23 18:40:47 DietPi1 java[4985]:         at org.sqlite.SQLiteJDBCLoader.loadSQLiteNativeLibrary(SQLiteJDBCLoader.java:367) ~[sqlite-jdbc-3.32.3.6.jar
!/:na]
Jun 23 18:40:47 DietPi1 java[4985]:         at org.sqlite.SQLiteJDBCLoader.initialize(SQLiteJDBCLoader.java:67) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:47 DietPi1 java[4985]:         at org.sqlite.core.NativeDB.load(NativeDB.java:59) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:47 DietPi1 java[4985]:         at org.sqlite.SQLiteConnection.open(SQLiteConnection.java:235) ~[sqlite-jdbc-3.32.3.6.jar!/:na]
Jun 23 18:40:47 DietPi1 java[4985]:         ... 83 common frames omitted
Jun 23 18:40:47 DietPi1 systemd[1]: komga.service: Main process exited, code=exited, status=1/FAILURE
Jun 23 18:40:47 DietPi1 systemd[1]: komga.service: Failed with result 'exit-code'.
```

Joulinar, 2021-06-23:
`Airsonic` seems to be working. Even if it took ages to start up

`org.airsonic.player.Application          : Started Application in 1144.089 seconds (JVM running for 1184.111)`

MichaIng, 2021-06-24:
Many thanks. So the general issue has been fixed. Yeah Airsonic, list many Java applications, start up slowly. Komga requires SQLite, but we never added it as dependency 🤔. Ah it is integrated, but the bug seems to be known: https://github.com/gotson/komga/issues/398

MichaIng, 2021-06-24:
I'll mark this issue as closed and will open a new one to track the Komga ARMv6 issue. Btw, does installing SQLite `apt install sqlite3` solve the issue or does it still tries to use the included library?

closed 2021-06-24 by MichaIng