MichaIng/DietPi #3790: RetroPie on DietPi

jeppenodgaard, 2020-09-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=32
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.6
- Kernel version | `uname -a`
Linux DietPi 4.19.0-11-amd64 #1 SMP Debian 4.19.146-1 (2020-09-17) x86_64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Native PC (x86_64)
- Power supply used | (EG: 5V 1A RAVpower)
Original
- SDcard used | (EG: SanDisk ultra)
Using a Samsung SSD

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
RetroPie
- Was the software title installed freshly or updated/migrated?
Migrated I guess
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Follow this guide: https://dietpi.com/docs/user-guide_install_uefi/
2. Follow this guide: https://github.com/MichaIng/DietPi/issues/1653
3. Reboot or run 'emulationstation'

#### Expected behaviour

- EmulationStation starts

#### Actual behaviour

`error: XDG_RUNTIME_DIR not set in the environment.
Sep 26 18:48:31 lvl0:   Error initializing SDL!
        No available video device
Sep 26 18:48:31 lvl0:   Renderer failed to initialize!
Sep 26 18:48:31 lvl0:   Window failed to initialize!`

#### Extra details

- I can launch emulationstation from LXDE.

MichaIng, 2020-09-26:
Did you check out this one: #3090
If it starts from within a desktop session but not from console, then I guess it needs to be started via `xinit`.

> I can launch emulationstation from LXDE.

Do you mean the desktop icon (I found one being created with the install script) or how did you launch it? If some executable launches from within X session but not from console, then generally `xinit /path/to/executable` might work instead. However, I remember that it used SDL, even your error message states that, so X session should not be required.

Which user did you try it with? Try to add it to video and render group:
```
usermod -aG video <username>
usermod -aG render <username>
```

jeppenodgaard, 2020-09-26:
> Did you check out this one: #3090

I think so. It seems like the issue ended where mine started..?

xinit seems to run fine, but it exits immediately `xinit emulationstation`:

```X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.19.0-10-amd64 x86_64 Debian
Current Operating System: Linux DietPi 4.19.0-11-amd64 #1 SMP Debian 4.19.146-1 (2020-09-17) x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.19.0-11-amd64 root=UUID=2e9d8509-a765-48ba-ba71-4110f21ef219 ro net.ifnames=0 consoleblank=0 quiet
Build Date: 27 August 2020  08:51:48AM
xorg-server 2:1.20.4-1+deb10u1 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sat Sep 26 23:42:49 2020
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
```
The log file: [Xorg.0.log](https://github.com/MichaIng/DietPi/files/5287432/Xorg.0.log)

> Do you mean the desktop icon 

I launched it by enabling LXDE autostart, open a terminal and run emulationstation.

> Which user did you try it with?

root - I ran the commands you suggested for root before running xinit.

I tried the steps suggested here, but it does not make a difference:
https://wiki.libsdl.org/FAQLinux#I_get_the_error:_.22no_video_devices_available.22

jeppenodgaard, 2020-09-27:
xinit /opt/retropie/supplementary/emulationstation/emulationstation works!
emulationstation points to /usr/bin/emulationstation which seems to be a bash script that wraps the executable.

I guess I need to update the init script?

MichaIng, 2020-09-27:
It is `/opt/retropie/supplementary/emulationstation/emulationstation.sh` with `sh` file ending, right?
Good to know as we then would need to adjust our autostart option as well. But needs some verification as this was obviously not required before: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L60

jeppenodgaard, 2020-09-27:
I tried with `emulationstation.sh` as well and it works too.
However emulators does not seem to work. There's just a completely black screen with the mouse (as a cross) in the middle. I can move the mouse, but nothing else. I have to do ctrl +alt + backspace to exit.

If I run xinit and then run `emulationstation`, `/opt/retropie/supplementary/emulationstation/emulationstation` or `/opt/retropie/supplementary/emulationstation/emulationstation.sh` they work fine.

jeppenodgaard, 2020-09-28:
This seems to do the trick `startx < /bin/sh -c "echo /opt/retropie/supplementary/emulationstation/emulationstation.sh"`. I'm not sure why. My guess is that it runs a shell in the x session and calls emulationstation in that shell. So it is similar to running `xinit` then `/opt/retropie/supplementary/emulationstation/emulationstation.sh` ?

If I run it to soon or add it to dietpi-login the keyboard input isn't captured by emulationstation, but I can exit the session with ctrl + alt + backspace.

@MichaIng Do you think it is time to open an issue at [EmulationStation](https://github.com/RetroPie/EmulationStation)?

MichaIng, 2020-09-28:
`startx < /bin/sh -c "echo /opt/retropie/supplementary/emulationstation/emulationstation.sh"` definitely does not what one might expect and might only work due to some strange circumstances. It should:
- Append the `/bin/sh` executable to startx STDIN. That should not have any effect, actually.
- Pass `-c` and `"echo /opt/retropie/supplementary/emulationstation/emulationstation.sh"` as arguments to the `startx` command, NOT to the `sh` command. Probably due to some unexpected argument spliting it somehow leads to emulationstation.sh being executed after all.

One of the following two options matches much more the regular way that startx is used:
```
startx /bin/sh /opt/retropie/supplementary/emulationstation/emulationstation.sh
```
or
```
startx /opt/retropie/supplementary/emulationstation/emulationstation.sh
```
so passing the command that should be executed within the X session as argument.

But actually `startx` is usually used as standalone command which pre-defined client in one of these files:
```
cat ~/.startxrc
cat ~/.xinitrc
cat /etc/X11/xinit/xinitrc
```

To call a custom client, `xinit` is what one would use normally, so:
```
xinit /bin/sh /opt/retropie/supplementary/emulationstation/emulationstation.sh
```
or
```
xinit /opt/retropie/supplementary/emulationstation/emulationstation.sh
```

> Do you think it is time to open an issue at EmulationStation?

If the official/documented way to start RetroPie/EmulationStation fails after having it installed (successfully) with the official installer, then yes. But you said starting it from desktop via desktop icon works, right? I am not sure how it is aimed to be started from console (outside X) nowadays. We call `/opt/retropie/supplementary/emulationstation/emulationstation.sh` from console directly which worked in the past, but probably that is simply wrong/outdated and does not match the intended way anymore?

jake-g, 2020-10-04:
im trying the same thing on a intel nuc pc (with dietpi), and this thread was really helpful. Im at the point now where i can get the emulationstation gui to show up with:

`xinit /opt/retropie/supplementary/emulationstation/emulationstation.sh`

It detects my usb gamepad, but it seems the button presses (required to advance the menu or set up the controller) aren't registering (i know these gamepads work with an older Pi version of retroarch). Also pressing f4 on my usb keyboard should exit emulationstation, but that doesnt work either. I did some basic troubleshooting...

1. Followed steps to test the usb gamepad mentioned (here)[https://www.tinkerboy.xyz/retropie-testing-the-usb-controller-joystick-via-the-command-line-or-terminal/] and everything checked out (i needed to install the joystick library with `sudo apt install joystick`). This was helpful in verifying my dietpi installation sees gamepad button presses.

2. I found some similar issue threads and it looks like emulationstation depends on button events being routed through X. This gave me the idea to start up my xfce desktop. Sure enough opening emulationstation there works fine and my controller and keyboard are detected fine.

Any ideas around this? Ideally i boot right into emulationstation WITH usb controller/keyboard support. I think the only desktop i have installed is xfce4 and a rarely use it. Happy to try out alternatives.

MichaIng, 2020-10-04:
`startx` btw is a wrapper for `xinit` only, which either loads a defined default X session/desktop (one of the three files mentioned above) or, if an input argument is given, behaves exactly like `xinit`. So `startx emulationstation.sh` and `xinit emulationstation.sh` should behave exactly the same, the latter then preferred since it is a bid cleaner and clearer about what is wanted. `startxfce4` on the other hand starts `xinit` explicitly with Xfce desktop session (based on `~/.config/xfce4/xinitrc` or `/etc/xdg/xfce4/xinitrc`), and ignores input arguments completely. This is the reason `startxfce4 emulationstation.sh` starts the desktop only, ignoring emulationstation.

There are a lot of X server options that can be passed via `xinit`:
- https://manpages.debian.org/xinit
- https://manpages.debian.org/Xorg

Probably one of those can be used to pass keyboard input. Probably also the `startxrc` or `xinitrc` file contents give a hint. I'll have a closer look after DietPi v6.33 release.

MichaIng, 2020-12-17:
I gonna mark this as closed. Testing and in case re-implementation of RetroPie is a dedicated topic.

closed 2020-12-17 by MichaIng