MichaIng/DietPi #3792: Zerotier

cocoflan, 2020-09-27:
### Creating a software request

#### Formal software information
- Software title | 
- Short description | 
- Official URL (if available) | 
- GitHub URL (if available) | 
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

- ...

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

MichaIng, 2020-09-27:
Many thanks for your request.

Remember that you already opened it a while ago 😄: #2843
I close this in favour of the old issue which contains some more info and discussion around it.

closed 2020-09-27 by MichaIng

cocoflan, 2020-09-28:
Ok, still hope it will get in the software list!