MichaIng/DietPi #3793: Problem with ffmpeg

maury77, 2020-09-27:
### Creating a bug report/issue

#### Required Information
DietPi v6.32.2 : 18:01 - Sun 27/09/20
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B+ (armv7l)
 - CPU temp : 47'C : 116'F (Optimal temperature)
 - LAN IP : 192.168.0.93 (eth0)
 - MOTD : RPi users, please do not lower your idle CPU frequency yet:
          https://github.com/MichaIng/DietPi/issues/3690

#### Steps to reproduce
I have a problem but I don't know this i the right forum

I need to use ffmpeg on HA but the ffpeg have this issue

ldd /usr/bin/ffmpeg
/usr/bin/ffmpeg: error while loading shared libraries: /usr/lib/arm-linux-gnueabihf/libflite_cmu_us_kal.so.1: unsupported version 29300 of Verneed record

MichaIng, 2020-09-27:
Many thanks for your report.

This library is pulled as dependency with FFmpeg:
`ffmpeg` > `libavfilter7` > `libflite1`

Can you please verify that all are pulled from the same repository:
```
apt policy ffmpeg libavfilter7 libflite1
```

maury77, 2020-09-27:
apt policy ffmpeg libavfilter7 libflite1
ffmpeg:
  Installed: 7:4.1.6-1~deb10u1+rpt1
  Candidate: 7:4.1.6-1~deb10u1+rpt1
  Version table:
 *** 7:4.1.6-1~deb10u1+rpt1 500
        500 https://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
     7:4.1.6-1~deb10u1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
libavfilter7:
  Installed: 7:4.1.6-1~deb10u1+rpt1
  Candidate: 7:4.1.6-1~deb10u1+rpt1
  Version table:
 *** 7:4.1.6-1~deb10u1+rpt1 500
        500 https://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
     7:4.1.6-1~deb10u1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
libflite1:
  Installed: 2.1-release-3
  Candidate: 2.1-release-3
  Version table:
 *** 2.1-release-3 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
        100 /var/lib/dpkg/status

MichaIng, 2020-09-27:
All as it should be, strange then. I'll try to replicate on my RPi, but it's currently compiling new Amiberry builds, so will have to wait until tomorrow. The `ffmpeg` fails completely because of this, right? In case reinstalling everything is always worth a shot:
```
apt update
apt install --reinstall ffmpeg libavfilter7 libflite1
```

maury77, 2020-09-28:
thanks NOW WORK

closed 2020-09-28 by maury77