MichaIng/DietPi #3797: Network file transfer speed drops and returns to full 1 Gbit/s after nmbd restart

Pillendreher, 2020-09-29:
- DietPi version: G_DIETPI_VERSION_CORE=6; G_DIETPI_VERSION_SUB=31; G_DIETPI_VERSION_RC=2; G_GITBRANCH='dev'; G_GITOWNER='MichaIng'
- Distro version: buster | 10.5 | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version: Linux DietPi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
- SBC model: RPi 4 Model B (aarch64)
- Power supply used: Standard Raspberry Pi 4 Power Supply
- SDcard used: I'm booting from a USB SSD

Over the last couple of weeks I've encountered an issue with the file transfer via ethernet over Samba. It starts out fine at ~100 MB/s and then suddenly drops all the way to ~10 MB/s. Unfortunately it stays there. The only way to get it back up to ~100 MB/s is to restart nmbd in dietpi-services. You can see the drop here:

[image]

The speed drop is not occuring every time, but still fairly regularly. 

I can not speak to this issue impacting FTP file transfer, but the next time it'll happen, I'll stop the transfer and try it via FTP.

Is there any way for me to figure out why this is happening?

MichaIng, 2020-09-29:
Many thanks for your report.

Did you watch RAM usage, probably swap file usage? And in case voltage drop, please paste the result of: `vcgencmd get_throttled`
Do you run other services/programs on that RPi that might include some timer or cron tasks?

Also check the following in case it is anything kernel related: `dmesg | tail -10`

Pillendreher, 2020-09-30:
At this point I can confirm that FTP is affected as well and that restarting vsftpd does not help; only a restart of all services brought the network speed back to normal.

I'll get back to you on the other stuff later today.

Joulinar, 2020-09-30:
If you have an additional spare SD card, would it be possible to test with the regular 32bit image? Just to exclude side effects from the 64bit Raspberry OS BETA.

Pillendreher, 2020-09-30:
> Did you watch RAM usage, probably swap file usage? And in case voltage drop, please paste the result of: `vcgencmd get_throttled`

So right now I'm trying to send two files to the Pi via FTP. Speed doesn't get above 1.2 MB/s. This is what htop looks like:

[image]

There is not a single process that hits double digits in CPU usage. 

Regarding the Throttling:

```
root@DietPi:~# vcgencmd get_throttled
throttled=0x60000
```

> Do you run other services/programs on that RPi that might include some timer or cron tasks?

Probably. I'm running Nextcloud, Sonarr, Radarr, SABNzbd, Telegraf, Grafphana, Influxdb and JDownloader permanently. 

> Also check the following in case it is anything kernel related: `dmesg | tail -10`

Nothing. The last kernel message is more than four days old.

Oh: And once again restarting nmbd fixes the network speed. Even though I was using FTP. 

Just as a precaution: Here's my smb.conf:

```
        passwd program = /usr/bin/passwd %u
        passwd chat = *Enter\snew\s*\spassword:* %n\n *Retype\snew\s*\spassword$
        pam password change = yes
        #map to guest = bad user

        load printers = no
        printcap name = /dev/null
        disable spoolss = yes

[dietpi]
        comment = DietPi Share
        path = /mnt
        browseable = yes
        create mask = 0664
        directory mask = 0775
        valid users = dietpi
        writeable = yes
max connections = 8
```

MichaIng, 2020-10-01:
> `throttled=0x60000`

Hmm, that means actually: https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md

Bit | Hex value | Meaning
-- | -- | --
17 | 20000 | Arm frequency capping has occurred
18 | 40000 | Throttling has occurred

ARM frequency was capped and throttling has occurred but a reason is missing (temperature or voltage). Strange 🤔.

> Oh: And once again restarting nmbd fixes the network speed. Even though I was using FTP.

Probably you can switch to NFS or SSHFS instead. I'll see if I find time for some tests with Samba.

Pillendreher, 2020-10-13:
> 
> > Oh: And once again restarting nmbd fixes the network speed. Even though I was using FTP.
> 
> Probably you can switch to NFS or SSHFS instead. I'll see if I find time for some tests with Samba.

The weird thing is this: Why should nmbd affect FTP speed? 

I'll delete Samba and report back. Maybe it fixes this issue.

MichaIng, 2020-12-17:
I'll mark this as closed. Feel free to reopen if issue persists.

closed 2020-12-17 by MichaIng