MichaIng/DietPi #3798: help me Nextcloud

ludji49, 2020-09-29:
DietPi Version | ```cat /DietPi/dietpi/.version 6.32.2
SBC Device (EG: RPi 3)
Power supply used (EG: 5v 1A RAVpower) yes
SD card used EG: Sandisk ultra 32 Go
Distro (EG: Jessie) | François

hello for a few days I can no longer open nextcloud could - you help me
 thank you

[image]

MichaIng, 2020-09-29:
Many thanks for your report.

Please paste the output of the following command:
```
journalctl -u apache2
```

ludji49, 2020-09-29:
here is the newspaper

[image]

Joulinar, 2020-09-29:
Looks like you have SSL/HTTPS enabled but there is no certificate found.

ludji49, 2020-09-29:
> Looks like you have SSL/HTTPS enabled but there is no certificate found. If needed try to recreate the certificate.

how? can you explain me

Joulinar, 2020-09-29:
did you used something like `Let’s Encrypt` to create the certificate before?

MichaIng, 2020-09-29:
Also it is the `default-ssl.conf` that is enabled, not the one that Certbot would place (`000-default-le-ssl.conf`), so probably that config got accidentally enabled? Solution would be:
```
a2dissite default-ssl
systemctl restart apache2
```

ludji49, 2020-09-29:
I did what you told me.

[image]

ludji49, 2020-09-29:
[image]

Joulinar, 2020-09-29:
pls can you post again

`journalctl -u apache2`

ludji49, 2020-09-29:
[image]

ludji49, 2020-09-29:
[image]

Joulinar, 2020-09-29:
ok looks like Apache is running now, as well you created the certificate? Are you able to access NextCloud from local network?

And pls remove your domain name from the screen shots

ludji49, 2020-09-30:
how to find my local network

MichaIng, 2020-09-30:
Try to access your Nextcloud now from a computer or mobile phone that is in the same LAN or WLAN as your Nextcloud server (=local network), respectively from you home.

Joulinar, 2020-09-30:
and you should use the same IP address (192.168.x.x) to connect, like you use to access your system via SSH (putty).

ludji49, 2020-09-30:
I can't access the site all the time

MichaIng, 2020-09-30:
Okay lets check the binary error log:
```
tail -10 /var/log/apache2/error.log
```
If this is empty, please restart Apache (`systemctl restart apache2`), try to access Nextcloud to trigger the 50X error and run the above log tail again.

And lets check the Nextcloud logs:
```
tail -10 /mnt/dietpi_userdata/nextcloud_data/nextcloug.log
```

@Joulinar 
The webroot btw is Nextcloud itself in this case: https://github.com/MichaIng/DietPi/issues/1554
Probably this was not the best idea considering the confusion/additional effort on support requests.

Another thing that confuses me is the missing `ServerName` directive, as we explicitly set it that time: https://github.com/MichaIng/DietPi/issues/1554#issuecomment-368225992
@ludji49 
Did you change something in the meantime, flashed new DietPi, upgraded the distribution or such? Probably we should get an overview about which Apache2 configs and sites are actually enabled:
```
ls -al /etc/apache2/{conf,sites}-enabled/*.conf
```

ludji49, 2020-09-30:
[image]

ludji49, 2020-09-30:
[image]

ludji49, 2020-09-30:
I only changed my internet box and since then I have no more access to the site.

MichaIng, 2020-09-30:
> I only changed my internet box and since then I have no more access to the site.

I guess then you simply need to recreate the port forwarding rules? Did you try to access your Nextcloud via plain IP address instead of domain name? Most browsers will at least throw a warning since you enabled HSTS and certificates do not support IP addresses reasonably, but you should be able to ignore this warning to access Nextcloud.

Btw, since you use PuTTY to access your console:
- Do not create screenshots.
- Instead select the relevant lines within the PuTTY window, so that they are highlighted. This will automatically copy the text into your clipboard.
- Then go to GitHub and hit `ctrl` + `v` keys to paste the content.
- Then, in GitHub, with the lines there still selected, hit the `< >`/`Insert code` button to make the logs nicely readable, like:
```
2020-09-30 13:42:21 root@micha:/var/log# ls -al /etc/apache2/{conf,sites}-enabled/*.conf
lrwxrwxrwx 1 root root 42 Sep 17 03:16 /etc/apache2/conf-enabled/dietpi-dav_redirect.conf -> ../conf-available/dietpi-dav_redirect.conf
lrwxrwxrwx 1 root root 28 Sep 17 03:35 /etc/apache2/conf-enabled/micha.conf -> ../conf-available/micha.conf
lrwxrwxrwx 1 root root 42 Sep 17 03:44 /etc/apache2/sites-enabled/000-default-le-ssl.conf -> ../sites-available/000-default-le-ssl.conf
lrwxrwxrwx 1 root root 40 Sep 17 03:16 /etc/apache2/sites-enabled/dietpi-nextcloud.conf -> ../sites-available/dietpi-nextcloud.conf
```

ludji49, 2020-09-30:
even with the local ip address this site is inaccessible

Joulinar, 2020-09-30:
Port 80 is forwarded as UDP. Pls can you forward it as TCP. Same like port 443.

ludji49, 2020-09-30:
this website is inaccessible

MichaIng, 2020-09-30:
What is the exact error message the browser shows when you connect via local IP? As said, due to HSTS, it is expected that it at least throws a warning, probably blocks access completely.

And yes, TCP port is required, UDP protocol forwarding can actually be removed.

About Nextcloud log, I see in the old threads that yours is stored at a different location, please paste:
```
cat /datastore/nextcloud.log
```

And to rule out the ServerName directive as issue:
```
G_CONFIG_INJECT 'ServerName' 'ServerName nextcloud-myweb.ddns.net' /etc/apache2/apache2.conf 'DocumentRoot'
sed -i '/^[[:blank:]]*ServerName/c\ServerName nextcloud-myweb.ddns.net/' /etc/apache2/sites-available/000-default.conf
sed -i '/^[[:blank:]]*ServerName/c\ServerName nextcloud-myweb.ddns.net/' /etc/apache2/sites-available/000-default-le-ssl.conf
```

And actually, since something has changed inside, lets review the Apache2 config:
```
cat /etc/apache2/apache2.conf
```

ludji49, 2020-09-30:
```
:~# cat /etc/apache2/apache2.conf
Mutex file:${APACHE_LOCK_DIR} default

PidFile ${APACHE_PID_FILE}

Timeout 60

KeepAlive On

MaxKeepAliveRequests 20

KeepAliveTimeout 5

User ${APACHE_RUN_USER}
Group ${APACHE_RUN_GROUP}

HostnameLookups Off

ErrorLog ${APACHE_LOG_DIR}/error.log

LogLevel error

# Include module configuration:
IncludeOptional mods-enabled/*.load
IncludeOptional mods-enabled/*.conf

# Include list of ports to listen on
Include ports.conf

<Directory />
        Options FollowSymLinks
        AllowOverride None
        Require all denied
</Directory>

<Directory /usr/share>
        AllowOverride None
        Require all granted
</Directory>

<Directory /var/www/>
        DirectoryIndex index.php index.html
        Options Indexes FollowSymLinks
        AllowOverride None
        Require all granted
</Directory>

AccessFileName .htaccess

<FilesMatch "^\.ht">
        Require all denied
</FilesMatch>

LogFormat "%v:%p %h %l %u %t \"%r\" %>s %O \"%{Referer}i\" \"%{User-Agent}i\"" v                                               host_combined
LogFormat "%h %l %u %t \"%r\" %>s %O \"%{Referer}i\" \"%{User-Agent}i\"" combine                                               d
LogFormat "%h %l %u %t \"%r\" %>s %O" common
LogFormat "%{Referer}i -> %U" referer
LogFormat "%{User-agent}i" agent

# Include generic snippets of statements
IncludeOptional conf-enabled/*.conf

# Include the virtual host configurations:
IncludeOptional sites-enabled/*.conf
root@DietPi:~#
```

MichaIng, 2020-09-30:
Ah I forgot that we set DocumentRoot via vhost only, hence the following should work:
```
G_CONFIG_INJECT 'ServerName' 'ServerName nextcloud-myweb.ddns.net' /etc/apache2/apache2.conf 'HostnameLookups'
```
Can you also paste the content of the vhosts:
```
cat /etc/apache2/sites-available/000-default{,-le-ssl}.conf
```
And the `nextcloud.log` tail, wherever it's stored, in your Nextcloud data directory.

ludji49, 2020-09-30:
```
# Include the virtual host configurations:
IncludeOptional sites-enabled/*.conf
root@DietPi:~# G_CONFIG_INJECT 'ServerName' 'ServerName nextcloud-myweb.ddns.net' /etc/apache2/apache2.conf 'HostnameLookups'
[  OK  ] G_CONFIG_INJECT | Added setting ServerName nextcloud-myweb.ddns.net to /etc/apache2/apache2.conf after line HostnameLookups Off
root@DietPi:~# cat /etc/apache2/sites-available/000-default{,-le-ssl}.conf
<VirtualHost *:80>
ServerAdmin webmaster@localhost
        ServerName nextcloud-myweb.ddns.net
        Redirect permanent / https://nextcloud-myweb.ddns.net/
DocumentRoot /var/www/nextcloud

        #CustomLog ${APACHE_LOG_DIR}/access.log combined
</VirtualHost>
<IfModule mod_ssl.c>
<VirtualHost *:443>
ServerAdmin webmaster@localhost
        ServerName nextcloud-myweb.ddns.net
        ServerName nextcloud-myweb.ddns.net
DocumentRoot /var/www/nextcloud

        ErrorLog ${APACHE_LOG_DIR}/error.log
        #CustomLog ${APACHE_LOG_DIR}/access.log combined
SSLCertificateFile /etc/letsencrypt/live/nextcloud-myweb.ddns.net/fullchain.pem
SSLCertificateKeyFile /etc/letsencrypt/live/nextcloud-myweb.ddns.net/privkey.pem
Include /etc/letsencrypt/options-ssl-apache.conf
Header always set Strict-Transport-Security "max-age=31536000"
</VirtualHost>
</IfModule>
root@DietPi:~#
```

MichaIng, 2020-09-30:
Okay, that looks all correct. There is a doubled `ServerName` directive in `/etc/apache2/sites-available/000-default-le-ssl.conf` but that does not cause issues.

If you don't know where your Nextcloud data dir is located, search for the log file:
```
find / -name nextcloud.log
```

And to exclude PHP/Nextcloud internal issues, you could do the following:
```
echo success > /var/www/nextcloud/index.htm
```
Then try to access this file via local IP address, like: `https://192.168.1.100/index.htm`
And also try it from console:
```
curl -ILk https://192.168.1.100/index.htm
```

ludji49, 2020-09-30:
Access from an unapproved domain
Please contact your administrator. If you are an administrator, edit the variable "trusted_domains" in the config/config.php file as the example in the config/config.sample.php file.

Further information on configuration can be found in the documentation .

[image]

ludji49, 2020-09-30:
```
root@DietPi:~# curl -ILk https://192.168.0.00/index.htm
HTTP/1.1 400 Bad Request
Date: Wed, 30 Sep 2020 16:12:24 GMT
Server: Apache/2.4.25 (Raspbian)
Strict-Transport-Security: max-age=31536000
Referrer-Policy: no-referrer
X-Content-Type-Options: nosniff
X-Download-Options: noopen
X-Frame-Options: SAMEORIGIN
X-Permitted-Cross-Domain-Policies: none
X-Robots-Tag: none
X-XSS-Protection: 1; mode=block
Set-Cookie: oc5cbhxlsgns=jcuf35dc1n173dc9e5i3e8p2mj; path=/; secure; HttpOnly
Expires: Thu, 19 Nov 1981 08:52:00 GMT
Cache-Control: no-store, no-cache, must-revalidate
Pragma: no-cache
Set-Cookie: oc_sessionPassphrase=HXqI4TvBwJPA74nmPYPnZIAtVUt7rFm%2FvgdjBYhOeECplhc0awTakY6Lxc8dwDFFULI8EZQ1BTiTSnSACwEqQMvfhBzM                          Xj7uENDMbQytZKkHg3yvW82wTL2PJaub5jy1; path=/; secure; HttpOnly
Content-Security-Policy: default-src 'self'; script-src 'self' 'nonce-cFR4QjNCbFE4dlBydmQ5cXhMZkNJdzRaU1NvaDljRXF0YVBwNkdHT0tyb                          z06NUZFT25WQTJ3TWU1K0xBRHNmU1JkenBnSjBRVmdvcDZqY21kaFFPNlRPaz0='; style-src 'self' 'unsafe-inline'; frame-src *; img-src * data                          : blob:; font-src 'self' data:; media-src *; connect-src *; object-src 'none'; base-uri 'self';
Set-Cookie: __Host-nc_sameSiteCookielax=true; path=/; httponly;secure; expires=Fri, 31-Dec-2100 23:59:59 GMT; SameSite=lax
Set-Cookie: __Host-nc_sameSiteCookiestrict=true; path=/; httponly;secure; expires=Fri, 31-Dec-2100 23:59:59 GMT; SameSite=stric                          t
Connection: close
Content-Type: text/html; charset=UTF-8

root@DietPi:~#

```

MichaIng, 2020-09-30:
Okay so far so good, the webserver reacts as expected. Let's now check the same with domain:
```
curl -ILk https://nextcloud-myweb.ddns.net/index.htm
```

Joulinar, 2020-09-30:
@MichaIng 
If I'm not mistaken, webserver is reachable again according this post https://github.com/MichaIng/DietPi/issues/3798#issuecomment-701491019

ludji49, 2020-09-30:
does not work

```
 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Bryce
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# curl -ILk https://nextcloud-myweb.ddns.net/index.htm
```

MichaIng, 2020-09-30:
Then this is an issue with your router connection or port forwarding. Please try to follow the recommendation of Joulinar and remove the port 80 UDP forwarding and replace it with port 80 TCP.

Also check your No-IP account. I remember one needs to login at their web UI every 30 days manually? And check the client state: `systemctl status noip2`

ludji49, 2020-10-01:
```
 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# systemctl status noip2
● noip2.service - noip2
   Loaded: loaded (/etc/systemd/system/noip2.service; disabled; vendor preset: e
nabled)
   Active: active (running) since Wed 2020-09-30 13:42:56 CEST; 16h a
go
  Process: 860 ExecStart=/usr/local/bin/noip2 (code=exited, status=0/SUCCESS)
 Main PID: 861 (noip2)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/noip2.service
           └─861 /usr/local/bin/noip2

sept. 30 13:42:56 DietPi systemd[1]: Starting noip2...
sept. 30 13:42:56 DietPi noip2[861]: v2.1.9 daemon started with NAT enabled
sept. 30 13:42:56 DietPi systemd[1]: Started noip2.
sept. 30 13:42:59 DietPi noip2[861]: No user '' found at dynupdate.no-ip
.com.
sept. 30 14:13:19 DietPi noip2[861]: Can't gethostbyname for dynupdate.n
o-ip.com
sept. 30 14:13:19 DietPi noip2[861]: Can't get our visible IP address fr
om ip1.dynupdate.no-ip.com
root@DietPi:~#
```

ludji49, 2020-10-01:
can the error come from ip?
can the error come from ip?
because I'm not the same ip since I've changed my place of residence

Joulinar, 2020-10-01:
can you login to NoIP website and check status of your account

MichaIng, 2020-10-01:
The No-IP client was definitely not able to report your IP, hence your domain is pointing to a wrong IP, which explains all the issues. Please redo the No-IP client config:
```
mkdir -p /usr/local/etc
noip2 -C
# ... follow the instructions
systemctl restart noip2
```

ludji49, 2020-10-01:
```
root@DietPi:~# systemctl restart noip2
root@DietPi:~# mkdir -p /usr/local/etc
root@DietPi:~# noip2 -C

Configuration file '/usr/local/etc/no-ip2.conf' is in use by process 7474.
Ending!

root@DietPi:~# # ... follow the instructions
root@DietPi:~# systemctl restart noip2
root@DietPi:~#
```

MichaIng, 2020-10-01:
```
systemctl stop noip2
noip2 -C
systemctl start noip2
```

ludji49, 2020-10-01:
```

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# systemctl stop noip2
root@DietPi:~# noip2 -C

Auto configuration for Linux client of no-ip.com.

Please enter the login/email string for no-ip.com

```

MichaIng, 2020-10-01:
Yes go ahead to reconfigure the client with your No-IP account credentials. The errors before indicated that this information was somehow lost.

ludji49, 2020-10-01:
I have modified my ip

[image]

Joulinar, 2020-10-01:
Ok for me it's working now and I'm able to connect to your Nextcloud installation.

ludji49, 2020-10-01:
The server is unable to execute your request.

If this happens again, please send the technical details below to the server administrator.

The server log file may provide more information.

Technical details
Remote address: 192.168.0.254
Application ID: GTB23136MkqsAUytOFxd
[image]

Joulinar, 2020-10-01:
Can you check all services running

`dietpi-service status`

ludji49, 2020-10-01:
```
 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# dietpi-service status
-bash: dietpi-service : commande introuvable
root@DietPi:~#
```

Joulinar, 2020-10-01:
sorry my fault.

`dietpi-services status`

ludji49, 2020-10-01:
```
root@DietPi:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | redis-server         active (running) since Thu 2020-                                                                   10-01 14:48:24 CEST; 3h 35min ago
[  OK  ] DietPi-Services | mariadb              active (running) since Thu 2020-                                                                   10-01 14:48:28 CEST; 3h 35min ago
[  OK  ] DietPi-Services | apache2              active (running) since Thu 2020-                                                                   10-01 14:48:29 CEST; 3h 35min ago
[  OK  ] DietPi-Services | urbackupsrv          active (running) since Thu 2020-                                                                   10-01 14:48:30 CEST; 3h 35min ago
[ INFO ] DietPi-Services | noip2                inactive (dead)
[  OK  ] DietPi-Services | cron                 active (running) since Thu 2020-                                                                   10-01 14:48:30 CEST; 3h 35min ago
[  OK  ] DietPi-Services | dropbear             active (running) since Thu 2020-                                                                   10-01 14:48:17 CEST; 3h 35min ago
[  OK  ] DietPi-Services | dietpi-ramlog        active (exited) since Thu 2020-1                                                                   0-01 14:17:08 CEST; 4h 6min ago
[  OK  ] DietPi-Services | dietpi-preboot       active (exited) since Thu 2020-1                                                                   0-01 14:17:08 CEST; 4h 6min ago
[  OK  ] DietPi-Services | dietpi-boot          active (exited) since Thu 2020-1                                                                   0-01 14:48:17 CEST; 3h 35min ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Thu 2020-1                                                                   0-01 14:48:17 CEST; 3h 35min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor  inactive (dead)
root@DietPi:~# ncc log:tail
 --------- ------------------- --------------------------------------------------------------------- --------------------------                    -
  Level     App                 Message                                                               Time                                         
 --------- ------------------- --------------------------------------------------------------------- --------------------------                    -
  Error     index               OCP\Files\NotPermittedException: Could not create path at             2020-10-01T14:37:20+00:00                    
                                lib/private/Files/Node/Folder.php line 185                                                                         
                                                                                                                                                   
                                 0. .../SimpleFolder.php line 83                                                                                   
                                    OC\Files\Node\Folder->newFile("4039-7dda-merged.css.gzip")                                                     
                                 1. .../SCSSCacher.php line 342                                                                                    
                                    OC\Files\SimpleFS\SimpleFolder->newFile("4039-7dda-merg ... p")                                                
                                 2. .../SCSSCacher.php line 181                                                                                    
                                    OC\Template\SCSSCacher->cache(                                                                                 
                                      "\/var\/www\/nextcloud\/a ... s",                                                                            
                                      "4039-7dda-merged.css",                                                                                      
                                      "merged.scss",                                                                                               
                                      OC\Files\SimpleFS\SimpleF ... {},                                                                            
                                      "\/apps\/files\/css"                                                                                         
                                    )                                                                                                              
                                 3. .../CSSResourceLocator.php line 109                                                                            
                                    OC\Template\SCSSCacher->process(                                                                               
                                      "\/var\/www\/nextcloud\ ... s",                                                                              
                                      "css\/merged.scss",                                                                                          
                                      "files"                                                                                                      
[... cut ...]
                                                                                                                                                   
  Warning   core                Could not send password reset email: Could not find user              2020-10-01T15:56:26+00:00                    
                                                                                                                                                   
 --------- ------------------- --------------------------------------------------------------------- --------------------------                    -

root@DietPi:~# dietpi-services status

```

Joulinar, 2020-10-01:
as far as I can see it has 2 issues. While I'm not sure if the secound is a result out  of the first one

`Could not create path at`
`Could not find user`

@MichaIng 
Could it be a persmission issue or a full file system?

MichaIng, 2020-10-01:
Probably but not 100% sure.

@ludji49 
Please try:
```
ncc files:scan # This might take a long time
ncc files:cleanup
ncc files:scan-app-data
chown -R www-data:www-data /datastore # In case, replace with the real location of your Nextcloud data
```

ludji49, 2020-10-01:
```
root@DietPi:~# ncc files:scan # This might take a long time
Please specify the user id to scan, --all to scan for all users or --path=...
root@DietPi:~# ncc files:cleanup
0 orphaned file cache entries deleted
root@DietPi:~# ncc files:scan-app-data

Scanning AppData for files

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 5289    | 13282 | 00:02:06     |
+---------+-------+--------------+
root@DietPi:~# chown -R www-data:www-data /datastore # In case, replace with the real location of your Nextcloud data                                                                                                                        chown: modification du propriétaire de '/datastore/appdata_oc5cbhxlsgns/css/files/4039-7dda-merged.css.gzip': La structure a besoin d'un nettoyage
chown: modification du propriétaire de '/datastore/appdata_oc5cbhxlsgns/preview/25110/1024-768.jpg': Message invalide
chown: impossible de lire le répertoire '/datastore/appdata_oc5cbhxlsgns/preview/40419': Message invalide
chown: modification du propriétaire de '/datastore/appdata_oc5cbhxlsgns/preview/25111/1024-768.jpg': Message invalide
chown: impossible de lire le répertoire '/datastore/appdata_oc5cbhxlsgns/preview/40421': Message invalide
chown: impossible de lire le répertoire '/datastore/appdata_oc5cbhxlsgns/preview/40423': Message invalide
chown: impossible de lire le répertoire '/datastore/appdata_oc5cbhxlsgns/preview/40420': Message invalide
chown: impossible de lire le répertoire '/datastore/appdata_oc5cbhxlsgns/preview/40417': Message invalide
chown: modification du propriétaire de '/datastore/appdata_oc5cbhxlsgns/preview/25112/1024-768.jpg': Message invalide
chown: modification du propriétaire de '/datastore/appdata_oc5cbhxlsgns/preview/25105/1024-768.jpg': Message invalide
chown: impossible de lire le répertoire '/datastore/appdata_oc5cbhxlsgns/preview/40422': Message invalide
chown: impossible de lire le répertoire '/datastore/appdata_oc5cbhxlsgns/preview/40418': Message invalide
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp5042858973': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp0546689271': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp8444459195': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp3391550035': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp9727968722': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp2588663178': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp1679363818': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp2051887457': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp5483998778': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp4480971498': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp9884850153': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp3652694089': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp4254113145': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp8218121518': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp3750474779': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp5379239276': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp6405237999': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp7560660646': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp7219430598': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp5845950740': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp4225973237': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp1919056511': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp8786101473': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp4491999191': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp4441764539': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp7304172140': Aucun fichier ou dossier de ce type
[... cut ...]
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp4335787758': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp1411686938': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp6798141235': Aucun fichier ou dossier de ce type
chown: modification du propriétaire de '/datastore/files_external/rootcerts.crt.tmp9093353962': Aucun fichier ou dossier de ce type
root@DietPi:~#
```

MichaIng, 2020-10-01:
Okay the only errors are with external storage files, in particular this temporary certificate, and a few previews or their dirs, but that should not matter.

I've forgotten the `--all` argument for the full file scan:
```
ncc files:scan --all
```

ludji49, 2020-10-01:
dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# ncc files:scan --all
Starting scan for user 1 out of 5 (admin)
Home storage for user admin not writable
Make sure you're running the scan command only as the user the web server runs as
Starting scan for user 2 out of 5 (Cloud_Alicia)
Starting scan for user 3 out of 5 (Cloud_Chloe)
Starting scan for user 4 out of 5 (My_cloud_dias)
Starting scan for user 5 out of 5 (Neau_Michel)
+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 2807    | 22519 | 00:02:41     |
+---------+-------+--------------+
root@DietPi:~#

MichaIng, 2020-10-01:
Okay, as previews was something before, we could do some failsafe steps:
```
ncc preview:repair
ncc preview:delete_old # This errors out as long as you don't have the PreviewGenerator app enabled
```
If web UI access fill does not work, try it with a different browser or after clearing browser cache, i.e. reload the website via `ctrl`+`c` keys. And if it still does not work, try a Nextcloud repair step (paste the last logs again before):
```
ncc log:tail
ncc maintenance:repair
```

ludji49, 2020-10-01:
15. .../base.php line 997                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    [... cut ...]
     - 0 changes without a calendar subscription have been cleaned up
     - 0 contacts without an addressbook have been cleaned up
     - 0 properties without a contact have been cleaned up
     - 0 changes without an addressbook have been cleaned up
 - Remove activity entries of private events
     - Removed 0 activity entries
 - Fix the share type of guest shares when migrating from ownCloud
 - Copy the share password into the dedicated column
 - Update OAuth token expiration times
 - Add background job to check for backup codes
root@DietPi:~#

MichaIng, 2020-10-01:
And? Still no access?

ludji49, 2020-10-02:
I have access by ipad and iphone but with windows the page doesn't open

Joulinar, 2020-10-02:
If you have access from iPad and iPhone but not from Windows, it's a client issue and not a server one. Pls clear browser date like cache or use a different browser.

[31 more replies]

closed 2020-12-17 by MichaIng