MichaIng/DietPi #3799: dietpi-config Performance Options exits with syntax error on Sparky

dsnyder0pc, 2020-09-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.32.2
- Distro version | buster
- Kernel version | Linux loft-sparky 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- SBC model | Sparky SBC (armv7l)
- Power supply used | Allo 5v, 3A
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | dietpi-config
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce

1. Flash microSD card with DietPi_SparkySBC-ARMv7-Buster_AlloGUI.img from 2020-09-21
2. Boot the device and login via ssh
3. Run dietpi-config
4. Choose option 3 - Performance Options

#### Expected behaviour

- I should be able to adjust performance options, including CPU Governor, etc.

#### Actual behaviour

-  dietpi-config exists with the following message:
```
/boot/dietpi/dietpi-config: line 1593: 989 -  : syntax error: operand expected (error token is "-  ")
```

#### Extra details

- ... Remember that Sparky is still on the 3.10.x kernel, so `/proc/meminfo` looks a bit different. Instead of `MemAvailable`, we have `CmaFree`. Perhaps do something like this (reports contiguous free memory, but close enough?)
```
   1592                 local memory_free=$(mawk '/MemAvailable:/ || /CmaFree:/{printf "%.0f",$2/1024;exit}' /proc/meminfo)
```

MichaIng, 2020-09-30:
Many thanks for your report and the included solution.

Seems like testing with "old" kernel is not enough anymore for Sparky SBC.

`MemFree` is available as well, right? AFAIK this is what we used before and, even again not the same, it is at least a reliable fallback where `CmaFree` reports `0 kB` or other very different values in cases, depending on hardware.

Another method, which we actually use everywhere else, is `free -m` which should have the "total" and "used" values as well. I guess on Sparky, the "available" column is missing as well?
```
              total        used        free      shared  buff/cache   available
Mem:            975         251         271          57         451         650
```
total - used != available
even
total - used - shred != available
so a bid difficult to get close 😄. However, it is for visual information only, nothing else is based on those values, so we'll just take the closest we can get and that's fine then. Could you print both for an overview, please:
```
free -m
cat /proc/meminfo
```

MichaIng, 2020-10-01:
Solved with: https://github.com/MichaIng/DietPi/commit/d87af24acc191796fb35372ccc38b831510e1c9e
Changelog: https://github.com/MichaIng/DietPi/commit/1e8521c64a8c0f953af3f5ad65d1ea08a1d674eb

@dsnyder0pc 
Since MemFree is located above MemAvailable, your attempt with a single mawk call sadly does not work. But in attempt to go with a simpler and more consistent solution, could you please paste the output of `free -m` on Sparky SBC?

closed 2020-10-17 by MichaIng