MichaIng/DietPi #3801: cannot install on Odroid U3+

jcw, 2020-10-01:
#### Required Information
- DietPi / Distro / Kernel version | latest (as of 2-Oct-2020)
- SBC model | Odroid U3+
- Power supply used | 5V 2A, Odroid official
- SDcard used | (EG: eMMC 64G

#### Steps to reproduce
1. download Odroid U3 image (reports as U4)
2. flash to eMMC w/ adapter, using BalenaEtcher
3. connect LAN cable & power up

#### Expected behaviour
- blinking blue LED (I think)

#### Actual behaviour
- steady blue LED
- nothing else, no boot activity

#### Extra details
- this board + eMMC used to run a very old version of DietPi
- but I lost the ssh key (and p/w was disabled), so I decided to re-flash and start over
- I waited many minutes, but the steady blue LED tells me it's stuck
- could it be that the current d/l image on dietpi.com is not for the U3+ ???

MichaIng, 2020-10-02:
Many thanks for your report.

I think you mixed up "U3" wtih "XU3", "XU4" etc, right? From what I see we supported Odroid U3 once (before my time) but do not offer an image anymore.

Although it's EOL for some years, there is still a Debian image by Meveric available: https://forum.odroid.com/viewtopic.php?f=79&t=17770
You could flash it and dist-upgrade it to Debian Buster with our script:
```
bash -c "$(curl -sSf https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh)"
```
If does a change that needs to be reverted for this device, probably leading to an error prompt. If so, open a subshell (else after DietPi-PREP finished) and do:
```
sed -Ei 's|https://dietpi\.com|https://oph.mdrjr.net|' /etc/apt/sources.list.d/meveric*.list
```
as we don't have the U3 components in our mirror. If else everything works fine, I might add them.
As well change all Meveric repo suites to Buster:
```
sed -Ei 's|jessie|buster|' /etc/apt/sources.list.d/*.list
apt update
apt upgrade
apt full-upgrade
```

jcw, 2020-10-02:
Wow ... _fantastic_ help, Micha! Some minor issues, and one I can't seem to solve at the end:

* had to `apt-get install curl`, as it wasn't on Meveric's image (oops, I didn't try `wget`)
* URL fixed: <https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh> (i.e. raw)

Not sure what to do about these:

```
183 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: There is no public key available for the following key IDs:
648ACFD622F3D138
 NO_PUBKEY DCC9EFBF77E11517
W: There is no public key available for the following key IDs:
648ACFD622F3D138
W: There is no public key available for the following key IDs:
112695A0E562B32A
W: There is no public key available for the following key IDs:
648ACFD622F3D138
```

Then, with `apt upgrade` it hangs after reporting what it wants to do, with `apt` pegging one CPU at 100%:

```
The following packages have been kept back:
  curl
The following packages will be upgraded:
  acl adduser apt apt-transport-https apt-utils base-files base-passwd bash
[...]
  vim-common vim-runtime vim-tiny wget whiptail wireless-tools wpasupplicant zlib1g
182 upgraded, 63 newly installed, 0 to remove and 1 not upgraded.
```

A reboot doesn't solve it. The upgrade is stuck, but I don't know why.

(one thing I did was apt update+upgrade before running the prep-script)

jcw, 2020-10-02:
Oh, hold the presses - it was just taking _ages_ ...

jcw, 2020-10-02:
Upgrade completed. This does mean I did not finish the prep-script, and seem to be locked out - but I can't get past the pw prompt via ssh, neither in root nor dietpi logins.

Shall I try to repeat the process and retry the failed step in the prep-script once the subshell upgrade completes?

Joulinar, 2020-10-02:
probably best to try again and complete the failed steps. Regarding the 5 missing keys, you could do following

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys <ADD KEY>`

jcw, 2020-10-02:
**Brilliant!**

I redid the install, everything fine now. For reference, here's a summary of what I ended up doing:

* used wget iso curl
* fixed url, as mentioned before
* no apt update/upgrade before running prep-script, following your instructions w/ meveric's image as is
* I did not see the "subshell" in the scrollable pane at first, so I logged in via 2nd ssh session
* did all the steps you mentioned, let it run to completion (upgrade _appears_ to hang for 3 mins or so)
* when done, switched back to other ssh session and selected `retry`
* after that, smooth sailing, all the way to first-run install via `root@dietpi` (no longer `root@odroid-jessie`)

Success:

```
$ ssh dietpi@dietpi
dietpi@dietpi's password:
 ─────────────────────────────────────────────────────
 DietPi v6.32.2 : 22:14 - Fri 10/02/20
 ─────────────────────────────────────────────────────
 - Device model : Odroid U3 (armv7l)
 - CPU temp : 45'C : 113'F (Optimal temperature)
 - LAN IP : 192.168.188.99 (eth0)
 - MOTD : Our website is now on GitHub: https://github.com/MichaIng/DietPi-Website
 ─────────────────────────────────────────────────────
[...]
$ uname -a
Linux DietPi 3.8.13.30 #1 SMP PREEMPT Wed Oct 26 17:53:58 CEST 2016 armv7l GNU/Linux
```

And ...

```
# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temperature  |     48'C : 118'F (Optimal temperature)
 Governor     |     performance

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1704 MHz      200 MHz    1704 MHz
 CPU1         |      1704 MHz      200 MHz    1704 MHz
 CPU2         |      1704 MHz      200 MHz    1704 MHz
 CPU3         |      1704 MHz      200 MHz    1704 MHz
```

Going from my confused Q about an "unsupported" SBC to total success in less than 24h - wow, thank you!

-jcw

PS. I also have an even older U2, and now DietPi makes these boards as usable as any, especially with eMMC.

MichaIng, 2020-10-02:
Alternative to add the missing keys for meverics repo and assure those stay updated:
```
cd /tmp
wget https://dietpi.com/meveric/pool/main/m/meveric-keyring/meveric-keyring_2019.7.18_all.deb
dpkg -i meveric-keyring_2019.7.18_all.deb
rm meveric-keyring_2019.7.18_all.deb
```

Ah lol sorry for non-raw URL 😄.

Yes indeed on slower devices and/or slower SD cards APT takes long on DietPi for two reasons:
- We disable APT cache by default, to avoid writing 60 - 100 MiB to SD card on each apt update.
- We leave downloaded APT list files xz-compressed to further reduce SD card writes by about the same or more, depending on how many lists need to be updated.
- This means APT needs to extract the lists and parse them in text (instead of cached binary) form, which takes a while.
- Since SD card deaths are still number one reason for system failures and data loss, we prioritise reduced SD card writes over performance here 😉.
__________
Actually I am wondering why do don't have Odroid U3 images since it's hardware ID is present and kernel etc is available via repository. I guess Kodi/GPU acceleration is not working on Buster, but at least as headless server image, with the option to compile graphics software manually, if required, it should work fine. I'll convert this issue into an image request. @jcw tell us if anything is not working as expected, so we can evaluate whether providing this image is worth it or not 🙂.

jcw, 2020-10-02:
Thx, keyring mods applied.

I found one glitch so far, in `dietpi-config` performance options (wanted to set governor to `ondemand`):

```
root@druide:/tmp# dietpi-config
/boot/dietpi/dietpi-config: line 1593: 2023 -  : syntax error: operand expected (error token is "-  ")
```

The other mild surprise was that "advanced options" takes about 10s to show its screen, it stays blank until then.

MichaIng, 2020-10-03:
> The other mild surprise was that "advanced options" takes about 10s to show its screen, it stays blank until then.

Ah yes a known issue with Linux 3.X, which has been fixed for DietPi v6.33 already: #3799

mickeyj4j, 2022-01-27:
Hi all so what happened to this image for the Odroud U3. I would love to have a working os with GuUI for my old board. Currently I have a debign bullseye os on an sd card running but its so slow. This will not install and boot from the emmc module so frustrating. 
I would love to use this for fun as a small os, its just sitting not doing much.

MichaIng, 2022-01-27:
Did you try to convert the Debian image to DietPi via [DietPi-PREP](https://dietpi.com/docs/hardware/#make-your-own-distribution)? At best create a full clone of the SD card first 😉.

But it won't add eMMC support, when it is really not supported by the Debian bootloader/kernel/dtb stack. Is it the Debian installer which does not offer to install to eMMC in the first place? Probably it simply doesn't ship the driver necessary to detect it while the system itself would be able to boot from it.
- Is the eMMC module detected by the Debian system, can be formatted and mounted?
- Do you have an eMMC USB adapter so that you can try to clone the SD card onto the eMMC? If not, I can give you some instructions how to clone the running system onto the eMMC.

jon-uk, 2022-04-28:
What are the possibilities of getting **Odroid-U3 kernel-headers** built and put into the repository?

**sudo apt-get install linux-headers-$(uname -r)** gives:

E: Unable to locate package linux-headers-5.15.12-stb-exy
E: Couldn't find any package by glob 'linux-headers-5.15.12-stb-exy'
E: Couldn't find any package by regex 'linux-headers-5.15.12-stb-exy'

Joulinar, 2022-04-28:
Basically we don't do any kernel development and we don't manage the apt package repository.

MichaIng, 2022-04-28:
As we do not provide an image for Odroid U3 since a long time, where did you get the image from? As the image shipped with an  kernel, probably a related APT repository, this repository is where you'd need to look for matching headers as well.

jon-uk, 2022-05-01:
Thanks @Joulinar  & @MichaIng ... Will be looking into building kernel & headers option, (a bit of a daunting task for me at the moment). Will share odroid-u3 header packages if possible. For odroid-u3 users switching to DietPi, I'd strongly recommend using the following images as it saves a lot of hassle from upgrading Meveric's Jessie images.

hexdump0815 images (based on hexdump0815's) at https://github.com/hexdump0815/linux-mainline-and-mali-on-odroid-u3

or antonyloytorrens bullseye image from https://githubhot.com/repo/hexdump0815/imagebuilder/issues/31

Personally using the latter which allows using DietPi Debian 11 bullseye nicely (**** except for packages requiring kernel headers ****).

Keep the justice for SBC going ;)

MichaIng, 2022-05-01:
Did you test with Debian's kernel and U-Boot? It has a U3 device tree and U-Boot for "odroid-xu3" and "odroid", so sounds like it should work:
- Kernel: https://packages.debian.org/bullseye/linux-image-armmp (backports can be used for Linux 5.16)
- U-Boot: https://packages.debian.org/bullseye/u-boot-exynos

But U-Boot needs to be flashed manually, AFAIK, but then I guess `boot.ini`, `boot.src` and `extlinux` support is included, the latter likely being the easiest choice as there are no device tree overlays that would need to be added with a fully fledged U-Boot script.

gociii, 2022-08-30:
For those of you looking for an up to date debian image, you can use the one provide by @hexdump0815 

https://github.com/hexdump0815/imagebuilder/releases/tag/220824-01

It uses the latest 5.19.1 kernel and works fine including kernel headers.

Maillot-david, 2022-09-02:
hello @gociii. Do you create a new image dietpi thanks to new @hexdump0815?

gociii, 2022-09-02:
@Maillot-david - Can do... All it requires is to

1. Download @hexdump0815 image above.
2. Burn it to an SD/MMC card and boot Odroid.
3. Run the dietpi-install script: 

`bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-installer')"`

Will try to upload it for the ODROID-U3 (debian-bullseye-dietpi-8.8.1) somewhere in the coming days.