MichaIng/DietPi #3804: dietpi-update:

Raketenforscher, 2020-10-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.31.2
- Distro version | 10.6
- Kernel version | Linux radio 5.4.51-v7+ #1333
- SBC model | RPi (Unknown) (armv7l)

- Bug report ID | f0bf20f2-0428-4cbe-bd24-4c2382b1a1c6

#### Steps to reproduce
execute command 'dietpi-update'

#### Expected behaviour
updating

#### Actual behaviour

DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[ INFO ] DietPi-Update | Invalid server version string: NULL.NULL.NULL
[ INFO ] DietPi-Update | Checking mirror: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
[ INFO ] DietPi-Update | Invalid server version string: NULL.NULL.NULL
[FAILED] DietPi-Update | Unable to access any update server. Please check your network connection, then rerun dietpi-update.
If this error persists, please report at: https://github.com/MichaIng/DietPi/issues

#### Extra details

network connection ok, https request ok, no fw/proxy

MichaIng, 2020-10-03:
@Raketenforscher 
Many thanks for your report. Strange issue since curl obviously does not report any error. Can you can it manually to see what content you get?
```
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
```

Raketenforscher, 2020-10-03:
6
32
2

Maybe it's worth mentioning that I quit a previous dietpi-update process with Ctrl-C because mpd service could not be stopped.

Joulinar, 2020-10-03:
did you tried to reboot and restart `dietpi-update` afterwards ?

Raketenforscher, 2020-10-03:
...sigh...

Always, ALWAYS the first step: https://www.youtube.com/watch?v=p85xwZ_OLX0

Yeah, worked. OF COURSE, STUPID ME! Thank you. 8)

closed 2020-10-03 by Raketenforscher

MichaIng, 2020-10-03:
A reboot should not even be required. Since the file download and content is as expected now, I'd simply try to redo the `dietpi-update`.

Hmm, now that you say hanging service stop and I see your kernel and DietPi version, you might be affected by: #3690
Please paste the output of the following:
```
vcgencmd get_config arm_freq_min
```
If that reports something different than 600 or 700, or you know that you lowered your idle frequency, please comment that settings and reboot your RPi:
```
sed -i 's/^[[:blank:]]*arm_freq_min/#arm_freq_min/' /boot/config.txt
reboot
```

Also that an "Unknown" RPi model is detected makes me wonder that there is some deeper issue since actually since v6.31 (or before?) DietPi should detect all RPi models correctly.

MichaIng, 2020-10-03:
Ah lol, too slow 😄.

@Raketenforscher could you still have a look through my post above? The update should have commented the setting already, but another reboot is required to have it becoming effective. Ah or probably the last reboot did when you had MOTD enabled in banner.
Is the RPi model now detected correctly?

Raketenforscher, 2020-10-03:
after reboot:
RPi 2 Model B armv7l
min freq 600

MichaIng, 2020-10-03:
Okay that is great, looks all fine then 🙂.