MichaIng/DietPi #3806: build shairport-sync with support for d-bus and mpris

seniorgod, 2020-10-03:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
i try to control shairport-sync with d-bus. it seems that shairport-sync is not responding. There is no such target at the bus. 
It seems that shairport-sync is compiled without d-bus or mpris support

#### Describe the solution you'd like:
please include support for d-bus or mpris like described here: 
[https://github.com/mikebrady/shairport-sync/issues/1027](url)

./configure --help

[...]
  --with-dbus-interface = include support for the native Shairport Sync D-Bus interface
  --with-dbus-test-client = compile dbus test client
  --with-mpris-interface = include support for a D-Bus interface conforming to the MPRIS standard
  --with-mpris-test-client = compile mpris test client
  --with-mqtt-client = include a client for MQTT -- the Message Queuing Telemetry Transport protocol
[...]
  

#### Describe alternatives you've considered:
the alternative is to build my own version of shairport

#### Additional context
because of some glitches while changing between mpd, spotify an shairport-sync, i would like to add some scripts.
What should these scripts do?
When start playing on mpd i will stop spotify and shairport. 
When start playing Spotify i will stop playing shairportsync and mpd 

Therefore i need to remote control the players.
mpd can be controlled by mpc 
shairport could be controlled by d-bus or mpris if d-bus or mpris is enabled 
for Spotify i use spotifyd which is also controllable by d-bus / MPRIS

MichaIng, 2020-10-03:
Many thanks for your request.

Just to be sure, dbus itself is installed and running right? Since it is no by default on DietPi:
```
apt install dbus
```

But I see now this needs to be enabled explicitly, which was not mentioned in their docs yet when I compiled those binaries. Same for MQTT I remember now: https://github.com/MichaIng/DietPi/issues/3216#issuecomment-602114941

Okay will do that the next days, I'll apply v6.33 milestone although this can be done completely independently from DietPi release as it's provided as an APT package.

seniorgod, 2020-10-03:
> But I see now this needs to be enabled explicitly, which was not mentioned in their docs yet when I compiled those binaries. Same for MQTT I remember now: #3216 (comment) <https://github.com/MichaIng/DietPi/issues/3216#issuecomment-602114941>

i found that result in other results running ps ax 

  844 ?        Ss     0:00 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation

while running dbus example commands from shairport-sync website i get a result which make me hoping dbus is running but miss the address from shairport-sync on the bus 

 dbus-send --print-reply --system --dest=org.gnome.ShairportSync /org/gnome/ShairportSync org.freedesktop.DBus.Properties.Get string:org.gnome.ShairportSync.Diagnostics string:Verbosity

 Error org.freedesktop.DBus.Error.ServiceUnknown: The name org.gnome.ShairportSync was not provided by any .service files

Currently i doesn’t use MQTT, but perhaps other people do it….

MichaIng, 2020-10-03:
Yes all makes sense, I was not sure if D-Bus support needs to be enabled explicitly, but it needs, which is even noted in the config files 😉. I hope you have another day to wait for the new package. Which device and Debian version are you running, so I can start with that build?

seniorgod, 2020-10-03:
i run dietpi 6.32.2 on 
Allo Sparky with Usbridge	

uname -a give: Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux

thanks a lot

MichaIng, 2020-10-03:
Debian Stretch or Buster? `echo $G_DISTRO_NAME`

seniorgod, 2020-10-03:
Stretch

MichaIng, 2020-10-07:
Build ready:
```
cd /tmp
wget https://dietpi.com/downloads/binaries/stretch/shairport-sync_armv7l.deb
apt install ./shairport-sync_armv7l.deb
rm shairport-sync_armv7l.deb
```
Based on v3.3.7rc3, not a stable release, but the third release candidate, up for a while already with no recent changes and including a bunch of bug fixes from v3.3.6 on, so should be worth it.

Config flags:
```
./configure --with-alsa --with-avahi --with-ssl=openssl --with-soxr --with-metadata --with-systemd --with-dbus-interface --with-mpris-interface --with-mqtt-client
```

seniorgod, 2020-10-08:
shairport runs and included now support for d-bus and mpris
It does not connect yet, perhaps D-Bus it not really running.
I will examine 
Thanks so far for including D-Bus and MPRIS in the binaries 

Okt 08 21:22:26 DietPi shairport-sync[25982]:          0.000006000 "rtsp.c:158" Creating metadata queue "mqtt".
Okt 08 21:22:26 DietPi shairport-sync[25982]:          0.022902250 "mdns_avahi.c:209" avahi: service '81D8D2407091@airplay project' group is not yet committed.
Okt 08 21:22:26 DietPi shairport-sync[25982]:          0.007426000 "mdns_avahi.c:244" avahi: request to add "_raop._tcp" service with metadata
Okt 08 21:22:26 DietPi shairport-sync[25982]:          0.003412750 "mdns_avahi.c:375" avahi_dacp_monitor_start Avahi DACP monitor successfully started
Okt 08 21:22:26 DietPi shairport-sync[25982]:          0.001226750 "dbus-service.c:1054" *warning: could not acquire a Shairport Sync native D-Bus interface "org.gnome.ShairportSync.i2
5982" on the system bus.
Okt 08 21:22:26 DietPi shairport-sync[25982]:          0.000403750 "mpris-service.c:349" *warning: could not acquire an MPRIS interface named "org.mpris.MediaPlayer2.ShairportSync.i259
82" on the system bus.
Okt 08 21:22:26 DietPi shairport-sync[25982]:          0.002977250 "mdns_avahi.c:213" avahi: service '81D8D2407091@airplay project' group is registering.
Okt 08 21:22:27 DietPi shairport-sync[25982]:          0.836905251 "mdns_avahi.c:180" avahi: service '81D8D2407091@airplay project' successfully added.
Okt 08 21:22:28 DietPi shairport-sync[25982]:          0.625808500 "shairport.c:194" soxr_delay_index: 7.
Okt 08 21:22:28 DietPi shairport-sync[25982]:          0.000114750 "shairport.c:201" "soxr" interpolation has been chosen.

MichaIng, 2020-10-08:
Hmm, at least it installs `dbus` as dependency which should be running by default. To check:
```
systemctl status dbus
journalctl -u dbus
```

seniorgod, 2020-10-09:
systemctl status dbus
● dbus.service - D-Bus System Message Bus
   Loaded: loaded (/lib/systemd/system/dbus.service; static; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-03 16:57:22 CEST; 5 days ago
     Docs: man:dbus-daemon(1)
 Main PID: 844 (dbus-daemon)
   CGroup: /system.slice/dbus.service
           └─844 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation

Okt 03 16:57:22 DietPi systemd[1]: Started D-Bus System Message Bus.
Okt 03 16:57:22 DietPi dbus[844]: [system] Activating via systemd: service name='org.freedesktop.PolicyKit1' unit='polkit.service'
Okt 03 16:57:22 DietPi dbus[844]: [system] Successfully activated service ‚org.freedesktop.PolicyKit1'

journalctl -u dbus
-- Logs begin at Sat 2020-10-03 16:57:20 CEST, end at Fri 2020-10-09 08:41:16 CEST. --
Okt 03 16:57:22 DietPi systemd[1]: Started D-Bus System Message Bus.
Okt 03 16:57:22 DietPi dbus[844]: [system] Activating via systemd: service name='org.freedesktop.PolicyKit1' unit='polkit.service'
Okt 03 16:57:22 DietPi dbus[844]: [system] Successfully activated service 'org.freedesktop.PolicyKit1'

MichaIng, 2020-10-09:
Found it I think. Please try:
```
wget https://raw.githubusercontent.com/mikebrady/shairport-sync/master/scripts/shairport-sync-dbus-policy.conf -O /etc/dbus-1/system.d/shairport-sync-dbus-policy.conf
```
This needs to be added to our packages. I'll do that the next days.

seniorgod, 2020-10-10:
looks better for D-Bus

Okt 10 11:26:45 DietPi shairport-sync[1253]:          0.000293750 "dbus-service.c:1046" Shairport Sync native D-Bus service started at "org.gnome.ShairportSync" on the system 
bus.
Okt 10 11:26:45 DietPi shairport-sync[1253]:          0.002348250 "mdns_avahi.c:375" avahi_dacp_monitor_start Avahi DACP monitor successfully started
Okt 10 11:26:45 DietPi shairport-sync[1253]:          0.003315500 "mdns_avahi.c:213" avahi: service '81D8D2407091@airplay project' group is registering.
Okt 10 11:26:45 DietPi shairport-sync[1253]:          0.002016250 "mpris-service.c:349" *warning: could not acquire an MPRIS interface named "org.mpris.MediaPlayer2.Sh
airportSync.i1253" on the system bus.
Okt 10 11:26:45 DietPi shairport-sync[1253]:          0.849316000 "mdns_avahi.c:180" avahi: service '81D8D2407091@airplay project' successfully added.
Okt 10 11:26:46 DietPi shairport-sync[1253]:          0.601002250 "shairport.c:194" soxr_delay_index: 7.
Okt 10 11:26:46 DietPi shairport-sync[1253]:          0.000119000 "shairport.c:201" "soxr" interpolation has been chosen.

d-bus command runs successfully 

dbus-send --print-reply --system --dest=org.gnome.ShairportSync /org/gnome/ShairportSync org.freedesktop.DBus.Properties.Get string:org.gnome.ShairportSync.Diagnostics string:Verbosity
method return time=1602322170.880432 sender=:1.69 -> destination=:1.70 serial=9 reply_serial=2
   variant       int32 3

mpris was not working  —> loading mpris config for d-bus from shairport-sync repo

wget https://raw.githubusercontent.com/mikebrady/shairport-sync/master/scripts/shairport-sync-mpris-policy.conf -O /etc/dbus-1/system.d/shairport-sync-mpris-policy.conf

restart shairport-sync 

—> 
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.000099000 "dbus-service.c:400" >> start including delta time in logs
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.000118750 "dbus-service.c:413" >> start including file and line in logs
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.000264750 "dbus-service.c:1046" Shairport Sync native D-Bus service started at "org.gnome.ShairportSync" on the system bus.
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.005115500 "mdns_avahi.c:244" avahi: request to add "_raop._tcp" service with metadata
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.002147500 "mpris-service.c:342" MPRIS service started at "org.mpris.MediaPlayer2.ShairportSync" on the system bus.
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.001632250 "mdns_avahi.c:375" avahi_dacp_monitor_start Avahi DACP monitor successfully started
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.004108500 "mdns_avahi.c:213" avahi: service '81D8D2407091@airplay project' group is registering.
Okt 10 11:38:22 DietPi shairport-sync[1305]:          0.836823250 "mdns_avahi.c:180" avahi: service '81D8D2407091@airplay project' successfully added.
Okt 10 11:38:23 DietPi shairport-sync[1305]:          0.616750000 "shairport.c:194" soxr_delay_index: 7.
Okt 10 11:38:23 DietPi shairport-sync[1305]:          0.000128000 "shairport.c:201" "soxr" interpolation has been chosen.

runs!

MichaIng, 2020-10-10:
Perfect, okay I'll add this two files to our packages.

MichaIng, 2020-10-28:
New build done with dbus/impris policies included:
```
cd /tmp
wget https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/shairport-sync_$G_HW_ARCH_NAME.deb
apt install ./shairport-sync_$G_HW_ARCH_NAME.deb
rm shairport-sync_$G_HW_ARCH_NAME.deb
```

seniorgod, 2020-10-31:
thanks,

it works

closed 2020-12-20 by MichaIng