MichaIng/DietPi #3808: Upgrade lighttpd

eikaramba, 2020-10-04:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
I need to write a proxy to a different server AND rewrite the url. The current version of lighttpd(1.4.45) does not support this.

#### Describe the solution you'd like:
Since lighttpd 1.4.46 one can use proxy.header to achieve this task: e.g.
```
$HTTP["host"] == "xxx.host.org" {
	$HTTP["url"] =~ "^/api/" {
		proxy.server = ( "" => (("host" => "192.168.178.11", "port"=>1337)))
		proxy.header = ( "map-urlpath" => ("/api" => "/") )
	}
}
```

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

Joulinar, 2020-10-04:
Hi,

many thanks for your request. But `lighttpd` is installed vie APT package directly from Debian repository. I guess you are on Debian Stretch, because there latest version is indeed 1.4.45-1. Therefore you would need to upgrade to Buster. Latest version on Debian Buster is 1.4.53-4 😉 

https://packages.debian.org/search?suite=all&searchon=names&keywords=lighttpd

eikaramba, 2020-10-05:
thank you very much for that info! i wasn't aware of this. :)
i read that it can be tricky to upgrade the distro, but will try it after a sd card backup.

closed 2020-10-05 by eikaramba

Joulinar, 2020-10-05:
Even if DietPi did not support Destro upgrades, you might have a look to following link https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988