MichaIng/DietPi #3810: Failed Webmin installation

JackSabbath, 2020-10-05:
#### Details:
- Date           | Mon Oct  5 20:10:02 CEST 2020
- Bug report     | 5b6c21ad-3535-48f1-8528-408c7d970187
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux RaspiNAS 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq --allow-change-held-packages install webmin
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
I tried to install Webmin, RPi Monitor and Netdata via dietpi-software.
#### Expected behaviour:
All of those programs are installed.
#### Actual behaviour:
Webmin installation fails due to dependency problems.
#### Extra details:
I changed my rootfs to btrfs previously. I can't garantuee that that didn't break something (symlinks or so.. although I think I took care of that).

Setting up apt-show-versions (0.22.11) ...
** initializing cache. This may take a while **
Error: No information about packages! (Maybe no deb entries?)
dpkg: error processing package apt-show-versions (--configure):
 installed apt-show-versions package post-installation script subprocess returned error exit status 255
dpkg: dependency problems prevent configuration of webmin:
 webmin depends on apt-show-versions; however:
  Package apt-show-versions is not configured yet.

dpkg: error processing package webmin (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 apt-show-versions
 webmin
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2020-10-05:
Hi,
Many  thanks for your report. Basically already known. It's an issue on Webmin side and a workaround will be implemented next release 6.33. 

```
echo 'Acquire::GzipIndexes "false";' > /etc/apt/apt.conf.d/98dietpi-webmin
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

MichaIng, 2020-10-05:
For reference: https://github.com/MichaIng/DietPi/commit/5be3fa0258bc154c5d257a1cec5223dfa8a6b360
And the fix/workaround from Webmin: https://github.com/webmin/webmin/pull/1302
But the change has not yet been released: https://github.com/webmin/webmin/releases

MichaIng, 2020-12-05:
Marking as closed, change has been released. I'll remove our workaround as well.

closed 2020-12-05 by MichaIng