MichaIng/DietPi #3811: Node-RED Compatibility Issues

Cjkeenan, 2020-10-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
root@BH-3B:~# cat /boot/dietpi/.version                                                                                                                                                         
G_DIETPI_VERSION_CORE=6                                                                                                                                                                         
G_DIETPI_VERSION_SUB=32                                                                                                                                                                         
G_DIETPI_VERSION_RC=2                                                                                                                                                                           
G_GITBRANCH='master'                                                                                                                                                                            
G_GITOWNER='MichaIng' 
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
```
root@BH-3B:~# echo $G_DISTRO_NAME                                                                                                                                                               
buster                                                                                                                                                                                          
root@BH-3B:~# cat /etc/debian_version                                                                                                                                                           
10.6   
```
- Kernel version | `uname -a`
```
root@BH-3B:~# uname -a                                                                                                                                                                          
Linux BH-3B 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux 
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
root@BH-3B:~# echo $G_HW_MODEL_NAME                                                                                                                                                             
RPi 3 Model B (armv7l)    
```
- Power supply used | (EG: 5V 1A RAVpower)
5V 2.4A APC Power Strip
- SDcard used | (EG: SanDisk ultra)
WD BLUE SATA SSD 250GB with ASM1153e controller

#### Additional Information (if applicable)
I contacted some other users using this set of nodes on NR, but I figured I would try here as well to see if the error looks familiar in some way, potentially a permissions issue?
```
TypeError: Cannot read property 'arlo' of undefined
    at /mnt/dietpi_userdata/node-red/node_modules/node-red-contrib-arlo/arlo.js:10:10
    at Layer.handle [as handle_request] (/usr/local/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
    at next (/usr/local/lib/node_modules/node-red/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/usr/local/lib/node_modules/node-red/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/usr/local/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
    at /usr/local/lib/node_modules/node-red/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/usr/local/lib/node_modules/node-red/node_modules/express/lib/router/index.js:335:12)
    at next (/usr/local/lib/node_modules/node-red/node_modules/express/lib/router/index.js:275:10)
    at expressInit (/usr/local/lib/node_modules/node-red/node_modules/express/lib/middleware/init.js:40:5)
    at Layer.handle [as handle_request] (/usr/local/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
```

#### Steps to reproduce

1. Install Node-RED and install `node-red-contrib-arlo` nodes
2. Setup account and search for devices associated with account

#### Expected behaviour

- Devices associated with account should populate the list

#### Actual behaviour

- Error above

StephanStS, 2020-10-06:
@Cjkeenan: Generally, this should not be the forum to get help for Node-RED nodes. I understand that the "online help" or whatever someone will call the node's docu is a bit poor. :-)

But to look at your problem: Could you send me an export of your flow to me via email? Then I could try to look into it.

MichaIng, 2020-10-06:
One question in the front: How did you install `node-red-contrib-arlo` node? When using the CLI `node-red`, take care to call it as `nodered` user. I just see that this is missing in our docs. I'll add some shell alias so that calling `node-red` will automatically call `sudo -u nodered node-red`, regardless of actual login user (as long as sudo permissions are granted). Did that for other software titles with CLI as well, so the correct configs are loaded and correct permissions are granted.

Cjkeenan, 2020-10-06:
Like I said I asked in a more appropriate place, but since I had issues with node-red previously specific to DietPi I wanted to create a post here in case it looks familiar.

> How did you install `node-red-contrib-arlo` node?

I installed via the GUI's palette manager, which I am guessing is already using the nodered user since it is my understanding that nodered user starts the instance. The weird thing is this did work for a while, and I am uncertain what changed that broke it. According to other users their systems are still up and running so I doubt it is a change Netgear made for Arlo. Is there way to check what user node-red was started with?

Cjkeenan, 2020-10-07:
Solved, it is an issue with the `node-red-contrib-arlo` nodes and their inability to handle 2FA.

closed 2020-10-07 by Cjkeenan