MichaIng/DietPi #3812: dietpi-upgrade 6.25.3 -> 6.32.2: dietpi-ramdisk: No such file or directory

badgertastic, 2020-10-06:
#### Details:
- Date           | Tue  6 Oct 15:13:24 BST 2020
- Bug report     | N/A
- DietPi version | v6.25.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1253 SMP Thu Aug 15 11:49:46 BST 2019
- Distro         | stretch (index=4)
- Command        | /DietPi/dietpi/func/dietpi-ramdisk 1
- Exit code      | 127
- Software title | DietPi-Update

#### Steps to reproduce:

1. dietpi-upgrade
(6.25.3 -> 6.32.2)

#### Expected behaviour:

- no error/failure

#### Actual behaviour:

- error: /DietPi/dietpi/func/dietpi-globals: line 1316: /DietPi/dietpi/func/dietpi-ramdisk: No such file or directory

#### Extra details:

- same on 2 separate Pi updates.

#### Additional logs:
```
Log file contents:
/DietPi/dietpi/func/dietpi-globals: line 1316: /DietPi/dietpi/func/dietpi-ramdisk: No such file or directory
```

badgertastic, 2020-10-06:
System is running ok, reboots ok etc. just concern that Ram is no-longer used and SD card being used instead (eventually leading to failure/corruption/performance degrade of SD). Please comment if dietpi-ramdisk is now obsolete and what replaced it if so. thanks for all your support :)

Joulinar, 2020-10-06:
Hi,

many thanks for your report. If I'm not mistaken this can be ignored. Isn't it @MichaIng 

maybe similar to this?? https://github.com/MichaIng/DietPi/issues/3532#issuecomment-628528041

MichaIng, 2020-10-06:
Many thanks for your report.

First of all: Yes `DietPi-RAMdisk` has been removed. This included only the `/boot/dietpi.txt`, `/boot/dietpi` and `/boot/config.txt` on RPi. The SD writes are likely lower now since in the past, the whole RAMdisk content was written to disk on every reboot and after every update (as it was attempted on your case), now only when one actually changes a setting, the single config file is written. But it depends a bid und usage of course. The main purpose of the RAMdisk was not to reduce SD writes (those are minimal for config files) but to speed up SD reads, i.e. DietPi scripts and settings, and this is now assured via `vmtouch` service, which keeps those files locked in native file system (read-)cache. The reason for this was to simplify our boot and shutdown process (no need to copy back and forth the RAMdisk content) and to eliminate the risk that a system crash or unclean shutdown leads to lost config changes and in case broken/reverted updates. So it should be all more solid now.

And yes, we had this update issue one or two times before, which occurred because the updater somehow did not kill itself (more precisely, its parent script) as intended and I am still now sure how this can happen. But as Joulinar said, it can be ignored since the updater already finished. It is an obsolete parent process which should have been killed but wasn't and now finished with outdated post-update steps, e.g. writing the RAMdisk content to disk which is not required and obviously not possible anymore.

But just to be very sure, do you still have the updater log in terminal, or the related log file:
```
cat /var/tmp/dietpi/logs/dietpi-update.log
```

badgertastic, 2020-10-06:
Thanks for the timely and informative post!  The situation is exactly as you describe and the installer did complete successfully. I'll paste the log FYI in any case. thanks again.

`[  OK  ] DietPi-Patch | Everything done! Terminating the obsolete DietPi-Update parent instance...

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Completed

[ INFO ] DietPi-Update | Current version : v6.25.3 (branch: Fourdee/master)
[ INFO ] DietPi-Update | Latest version  : v6.32.2 (branch: MichaIng/master)
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
chdir: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
[  OK  ] DietPi-Survey | Purging survey data
[FAILED] DietPi-Update | /DietPi/dietpi/func/dietpi-ramdisk 1
[FAILED] DietPi-Update | /DietPi/dietpi/func/dietpi-ramdisk 1
[FAILED] DietPi-Update | /DietPi/dietpi/func/dietpi-ramdisk 1

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------`

Details as previously shared.

The log:

`# cat /var/tmp/dietpi/logs/dietpi-update.log
[ SUB2 ] DietPi-Update > Applying pre-patches
[  OK  ] DietPi-Update | Downloading pre-patch file
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] Pre-patch 13 | Disabling dpkg status file translations by default, which have hardly any use, but raise APT update times + disk I/O 
[ INFO ] Pre-patch 20 | Mark ifupdown as manually installed to prevent its autoremoval
ifupdown was already set to manually installed.
[  OK  ] DietPi-Update | Successfully applied pre-patches
[ SUB2 ] DietPi-Update > Upgrading APT packages
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
[  OK  ] DietPi-Update | APT upgrade
[ SUB2 ] DietPi-Update > Installing new DietPi code
[  OK  ] DietPi-Update | Downloading update archive
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts
[  OK  ] DietPi-Update | Installing new DietPi system files
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts
[ SUB3 ] DietPi-Set_software > verify_dietpi.txt ()
[  OK  ] DietPi-Set_software | Checking URL: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[ INFO ] DietPi-Set_software | Updating dietpi.txt with new entry: AUTO_SETUP_ACCEPT_LICENSE=0
[ INFO ] DietPi-Set_software | Updating dietpi.txt with new entry: AUTO_SETUP_NET_WIFI_COUNTRY_CODE=GB
[ INFO ] DietPi-Set_software | Updating dietpi.txt with new entry: CONFIG_CHECK_CONNECTION_IP=9.9.9.9
[ INFO ] DietPi-Set_software | Updating dietpi.txt with new entry: CONFIG_CHECK_DNS_DOMAIN=dns9.quad9.net
[ INFO ] DietPi-Set_software | Updating dietpi.txt with new entry: SOFTWARE_HOMEASSISTANT_APT_DEPS=
[ INFO ] DietPi-Set_software | Updating dietpi.txt with new entry: SOFTWARE_HOMEASSISTANT_PIP_DEPS=
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed
[ SUB2 ] DietPi-Update > Applying incremental patches
[ INFO ] DietPi-Update | Current version : v6.25.3
[ INFO ] DietPi-Update | Latest version  : v6.32.2
[ INFO ] DietPi-Patch | Patching 6.25 to 6.26
[ SUB4 ] DietPi-Set_hardware > rpi-camera (disable)
[  OK  ] rpi-camera disable | Completed
[  OK  ] DietPi-Patch | Added setting dtoverlay=disable-bt to end of file /boot/config.txt
[  OK  ] DietPi-Patch | Added setting dtoverlay=disable-wifi to end of file /boot/config.txt
[  OK  ] DietPi-Patch | Comment in /boot/config.txt converted to setting: initial_turbo=20
[ INFO ] DietPi-Patch | APT purge for: haveged, please wait...
[... 107 lines cut ...]
[  OK  ] DietPi-Software | No changes applied for: Grafana
[ INFO ] DietPi-Software | 90: phpMyAdmin is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 90" to install phpMyAdmin.
[  OK  ] DietPi-Software | No changes applied for: phpMyAdmin
[ INFO ] DietPi-Software | 147: Jackett is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 147" to install Jackett.
[  OK  ] DietPi-Software | No changes applied for: Jackett
[ INFO ] DietPi-Software | 163: GMediaRender is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 163" to install GMediaRender.
[  OK  ] DietPi-Software | No changes applied for: GMediaRender
[ INFO ] DietPi-Software | 98: HAProxy is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 98" to install HAProxy.
[  OK  ] DietPi-Software | No changes applied for: HAProxy
[ INFO ] DietPi-Software | 108: Amiberry is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 108" to install Amiberry.
[  OK  ] DietPi-Software | No changes applied for: Amiberry
[ INFO ] DietPi-Software | 160: Allo_update is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 160" to install Allo_update.
[  OK  ] DietPi-Software | No changes applied for: Allo_update
[ INFO ] DietPi-Patch | APT autopurge, please wait...
(Reading database ... 28136 files and directories currently installed.)
Removing libhavege1:armhf (1.9.1-5+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
[  OK  ] DietPi-Patch | APT autopurge
[  OK  ] DietPi-Update | Incremental patching to v6.32.2 completed`

MichaIng, 2020-10-06:
Okay, great, everything is fine then. Luckily there are not much systems on v6.25 anymore, since a reliable fix on a 6.25 system prior to the update is not reliably possible. We could hack something inside via MOTD but not everyone has it enabled.

closed 2020-10-06 by badgertastic