MichaIng/DietPi #3815: DietPi-Software | dietpi_userdata directory creation on software installation run??

Joulinar, 2020-10-08:
@MichaIng 
I'm not sure how this is related but in the last days I have seen the the following lines during dietpi-software installations. There are executed right at the beginning after stopping all services.

```
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
```

Until now I did not find a pattern and I'm not sure what the trigger is.

MichaIng, 2020-10-08:
These steps were always done, but there was no console output for it until now. I wrapped those commands into our error handler G_EXEC, hence you now see what is happening. Generally I wrapped a lot of trivial commands into the error handler during the last releases, so install verbosity generally increased, also other scripts did. I remember there being some request for this (showing more verbosely what is happening during installs) in the forum a while ago and it is generally good to error handle also trivial things, so when e.g. a directory creation fails (broken binary, R/O mount, something else serious), install is halted immediately instead of running into a failure somewhere in the middle, where its harder to debug.

For these in particular, many software titles use those directories and (cross) R/W access is assured by adding or running services with `dietpi` group. But we could also switch to a less intrusive method and only create and chown/chmod the directories with the individual software installs, e.g. on qBittorrent install, downloads is created, on image gallery install, Pictures is created etc.

Joulinar, 2020-10-08:
ok that explain's it 😄 
I was not aware and just wondering. 

Regarding the method  how to have it implemented. Basically it depends what you prefer. Current one seems easy as it is done globally. While putting it to each individual software tile could cause quite some effort 🤔

MichaIng, 2020-10-08:
> While putting it to each individual software tile could cause quite some effort 🤔

Yes, would increase the code lines at least and bears the risk that the one or the other case might be overseen. I personally am a bid annoyed by additional directories and things I don't need in my particular case. I always remove those directories, as well `/mnt/samba` and the other file transfer client directories with readme inside, as I don't like to have directories around that I never use. I even remove the `dietpi` group and user 😄. But of course, if I start to use this or that additional feature or software title I need the user and/or locations for that to work. And in case of userdata folders it is a very common thing, the same on Windows and MacOS etc, so some users might be wondering or complaining if there is no such structure available when opening the file manager. So yeah, I think it is fine for now, and if we remove any of such pre-creations, we should do it step-by-step or even raise some poll to allow for feedback 🙂.

MichaIng, 2020-12-19:
I mark this as closed.

closed 2020-12-19 by MichaIng