MichaIng/DietPi #3816: Issue installing octoprint

vcliment89, 2020-10-08:
#### Details:
- Date           | Thu Oct  8 11:16:49 CDT 2020
- Bug report     | 52b91532-4b83-4ca0-b448-e35bbaf977cc
- DietPi version | v6.32.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | Linux DietPi 5.4.28-sunxi64 #20.02.7 SMP Sat Mar 28 17:25:10 CET 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | pip3 install -U .
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 930: /usr/bin/pip3: No such file or directory
```

MichaIng, 2020-10-08:
Not sure currently why on v6.32 the install would even call pip3, but with v6.33 we migrated the OctoPrint install to Python 3.
You can test it with the beta:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /boot/dietpi.txt
dietpi-update 1
```

Joulinar, 2020-10-19:
Hi,

DietPi v6.33.3 is out now. I guess installation should be working. 

@vcliment89 
pls could you try again and report if there are issues still. thx

vcliment89, 2020-10-20:
Yes, I'm now able to install octoprint. Thanks!

Joulinar, 2020-10-20:
@MichaIng 
Do you like to keep this open for further investigation? Or can it be closed?

closed 2020-10-20 by vcliment89

MichaIng, 2020-10-20:
Just ran another test and all works as expected. Okay to have it closed and we'll keep it in mind in case a similar report is done.