MichaIng/DietPi #3817: Provide an explicit option for "no web server"

davidje13, 2020-10-08:
Currently it is possible to chose between Apache, NGINX and Lighttpd using the `AUTO_SETUP_WEB_SERVER_INDEX` option, but there is no (documented) way to avoid installing a webserver entirely, unlike the other similar options (e.g. `AUTO_SETUP_SSH_SERVER_INDEX` has `0`).

From a quick experiment, it appears that setting any unrecognised value (e.g. -99) will prevent the installation of a web server, but this is fragile.

Can you add an explicit (documented) value for "install no web server"?

Joulinar, 2020-10-08:
Hi,

many thanks for your request. Basically there will no web server installed during first initial setup. This is just the default value that will be chosen as soon as you install a software title who require a web server.

davidje13, 2020-10-08:
Ah; that's not clear from the dietpi.txt info then!

It would still be nice to have a "nothing" option for explicitness-sake. My use-case is a small device which (after installation) will have no internet access, so being able to explicitly avoid anything which might try to open a port would be nice. (presumably devices which _do_ have internet access may also want that for security reasons, to ensure it's only running the exact software they expect)

Joulinar, 2020-10-08:
As said, there will be no web server installed by default. In general if you run an installation on default values, nothing will be installed except core services to run your device and an SSH server to be able to connect in headless mode.

davidje13, 2020-10-08:
Sure, I understand. Adding more info to the `AUTO_SETUP_WEB_SERVER_INDEX` line in `dietpi.txt` would probably help explain that to future users.

I understand why a "nothing" option isn't really needed and fair enough if you don't want to add one, but it's a bit inconsistent with the other similar-looking flags which led to my initial confusion (`AUTO_SETUP_SSH_SERVER_INDEX` / `AUTO_SETUP_FILE_SERVER_INDEX` / `AUTO_SETUP_LOGGING_INDEX`)

Joulinar, 2020-10-08:
well it's called `Webserver Preference Selection` 
Not meaning that something will be installed 😉

MichaIng, 2020-10-08:
Yeah it's the wording "preference" vs "choice" where a choice means that you select something to be installed and preference means that you select a preference for which webserver should be installed, when one is required. So e.g. when you install Nextcloud, which requires a webserver, the selected preference will be followed.

But I agree we could make that a bid clearer, especially since a desktop preference and a browser preference is planned as well and even a database preference, which although is much more work and would work a bid different.

MichaIng, 2021-05-18:
Menus as well as dietpi.txt entries have been clarified: #4314
It should now be clear that selecting a preference won't install it directly but only respected when another selected software title requires a webserver (respectively desktop or browser, which have a preference menu now as well).

closed 2021-05-18 by MichaIng