MichaIng/DietPi #3818: hcitool lescan very slow and cannot find bluetooth devices

cc13com, 2020-10-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=32
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `buster` or `10.6`
- Kernel version | `Linux DietPi4 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)` or (EG: RPi3)
- Power supply used | (5V 2.3 A)
- SDcard used | (USB-SSD-Adapter with 250GB SSD)

#### Additional Information (if applicable)
Working image transferred from SD-Card to SSD connected via USB-SSD-Adapter.

- Bug report ID | `5cb24ec4-70a5-4be8-a3d3-db156237953a`

#### Steps to reproduce
I activated Bluetooth in dietpi-config and will scan for other bluetooth devices. First I check if bluetooth is active:

`root@DietPi4:/home/dietpi# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-10 16:45:43 CEST; 12min ago
     Docs: man:bluetoothd(8)
 Main PID: 578 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/bluetooth.service
           └─578 /usr/lib/bluetooth/bluetoothd

Okt 10 16:45:43 DietPi4 systemd[1]: Starting Bluetooth service...
Okt 10 16:45:43 DietPi4 bluetoothd[578]: Bluetooth daemon 5.50
Okt 10 16:45:43 DietPi4 systemd[1]: Started Bluetooth service.
Okt 10 16:45:43 DietPi4 bluetoothd[578]: Starting SDP server
Okt 10 16:45:43 DietPi4 bluetoothd[578]: Bluetooth management interface 1.14 initialized
Okt 10 16:45:43 DietPi4 bluetoothd[578]: Sap driver initialization failed.
Okt 10 16:45:43 DietPi4 bluetoothd[578]: sap-server: Operation not permitted (1)
Okt 10 16:45:43 DietPi4 bluetoothd[578]: Failed to set privacy: Rejected (0x0b)`

I know I can ignore the SAP error message. Now I start the scan with:

`root@DietPi4:/home/dietpi# hcitool lescan
LE Scan ...`

Thats all, after a couple of minutes I can stop this. A scan in parallel with my RPi Zero and DietPi it will find more then 10 devices in around 10 sec. Both RPis are on the same desk.

#### Expected behaviour
Show existing bluetooth devices next to the RPi.

#### Actual behaviour
Will find nothing.

#### Extra details
It was working on the same hardware RPi4 with SD-Card. Could it be that the USB-SSD-Adapter is the reason? When I stop the scanning with control+c after a couple of minutes I see the following:

`root@DietPi4:/home/dietpi# hcitool lescan
LE Scan ...
^CDisable scan failed: Input/output error`

MichaIng, 2020-10-10:
Many thanks for your report.

First of all, is the SSD (adapter) powered via dedicated power supply? As powering it through RPi USB will not work without regular voltage drops.

Please paste the output of the following two commands:
```
vcgencmd get_throttled
dmesg | tail -10
```

cc13com, 2020-10-10:
no, the ssd is not powered via dedicated power.

here the output:

`
dietpi@DietPi4:~$ vcgencmd get_throttled
throttled=0x0
dietpi@DietPi4:~$ dmesg | tail -10
[ 3888.214316] br-4fa2b31adeb0: port 1(vethe4cfbfd) entered blocking state
[ 3888.214324] br-4fa2b31adeb0: port 1(vethe4cfbfd) entered disabled state
[ 3888.214536] device vethe4cfbfd entered promiscuous mode
[ 3888.214741] br-4fa2b31adeb0: port 1(vethe4cfbfd) entered blocking state
[ 3888.214754] br-4fa2b31adeb0: port 1(vethe4cfbfd) entered forwarding state
[ 3889.030672] br-4fa2b31adeb0: port 1(vethe4cfbfd) entered disabled state
[ 3889.381308] eth0: renamed from veth0e43496
[ 3889.422075] IPv6: ADDRCONF(NETDEV_CHANGE): vethe4cfbfd: link becomes ready
[ 3889.422182] br-4fa2b31adeb0: port 1(vethe4cfbfd) entered blocking state
[ 3889.422190] br-4fa2b31adeb0: port 1(vethe4cfbfd) entered forwarding state
`

MichaIng, 2020-10-10:
Okay, no low voltage at least, but I'd still try go with a USB 2,5" docking station with dedicated power supply. Not sure about side effects which are probably not seen by this command.

Strange is the quick state change of the Ethernet adapter. Did you play around with that, `tcpdump` (promiscuous mode) or something?

This can be used to check for all kernel errors (leaving out non-error info messages): `dmesg -l emerg,alert,crit,err`

And what is the status of the hciuart service? `systemctl status hciuart`
Serial device available but serial console disabled?
```
systemctl status serial-getty@ttyAMA0.service
ls -l /dev/ttyAMA0
```

cc13com, 2020-10-11:
Hi, no I didn't played with `tcpdump`. At the moment I don't have the possibility to use a docking station. I will try if I have the hardware.

`
dietpi@DietPi4:~$ sudo dmesg -l emerg,alert,crit,err
dietpi@DietPi4:~$
`

`
dietpi@DietPi4:~$ sudo systemctl status hciuart
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-10 16:45:42 CEST; 18h ago
  Process: 268 ExecStart=/usr/bin/btuart (code=exited, status=0/SUCCESS)
 Main PID: 575 (hciattach)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/hciuart.service
           └─575 /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow -

Okt 10 16:45:23 DietPi4 systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Okt 10 16:45:42 DietPi4 btuart[268]: bcm43xx_init
Okt 10 16:45:42 DietPi4 btuart[268]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Okt 10 16:45:42 DietPi4 btuart[268]: Set Controller UART speed to 3000000 bit/s
Okt 10 16:45:42 DietPi4 btuart[268]: Device setup complete
Okt 10 16:45:42 DietPi4 systemd[1]: Started Configure Bluetooth Modems connected by UART.
`

`
dietpi@DietPi4:~$ sudo systemctl status serial-getty@ttyAMA0.service
● serial-getty@ttyAMA0.service - Serial Getty on ttyAMA0
   Loaded: loaded (/lib/systemd/system/serial-getty@.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:agetty(8)
           man:systemd-getty-generator(8)
           http://0pointer.de/blog/projects/serial-console.html
`

`
dietpi@DietPi4:~$ ls -l /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 Okt 10 16:45 /dev/ttyAMA0
`

I try to explain how I installed all: Installation of DietPi on my RPi4 and SD-Card. Some additional packages (OpenSSH, Docker, PIVPN-Wireguard, Kodi). Backup (rsync) of the whole system via [raspibackup](https://www.linux-tips-and-tricks.de/de/raspibackup/) to my NAS. After connecting the SSD to the RPi4 I did a restore with raspiBackup. Shutdown the system, remove the SD-Card and starting the RPi4 with the SSD. The boot needed more then 5 min. I searched in the net and found the following:

`
dietpi@DietPi4:~$ lsusb
Bus 002 Device 002: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

So I added `usb-storage.quirks=152d:0578:u` to my cmdline.txt and the boot is fast as expected.

MichaIng, 2020-10-11:
Okay, another drive/firmware that does not support UAS. Another solution for this would be:
```
echo 'blacklist uas' > /etc/modprobe.d/blacklist-uas.conf
```
This disables the UAS module completely and by this safes some RAM as well 😉. But only reasonable if the non-supporting SSD is the only attached USB drive.

hciuart connects to the serial1 symlink. Let's assure this is pointing to ttyAMA0 and there is no serial console active on the symlink as well. The file permissions already show it, however lets assure:
```
ls -l /dev/serial1
systemctl status serial-getty@serial1
```

But since services and what we checked all are fine, I still think it's possible hat the SSD, drawing too much power from the USB bus, has an influence. And of course since this is the only thing you changed 😉. It is a general fact that SBC USB ports are not reliable to power 2,5" drives. 1.2A is the maximum current that is provided shared across all ports on RPi: https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md
That means a shared overall limit of 6W. That may work for modern low power SSDs, but considering peaks and fluctuations in voltage, one cannot rely on it.

cc13com, 2020-10-11:
I did `echo blacklist uas` thing and will reboot after this post.

`
root@DietPi4:/etc/modprobe.d# ls -l /dev/serial1
lrwxrwxrwx 1 root root 7 Okt 10 16:45 /dev/serial1 -> ttyAMA0
`

`
root@DietPi4:/etc/modprobe.d# systemctl status serial-getty@serial1
● serial-getty@serial1.service - Serial Getty on serial1
   Loaded: loaded (/lib/systemd/system/serial-getty@.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:agetty(8)
           man:systemd-getty-generator(8)
           http://0pointer.de/blog/projects/serial-console.html
root@DietPi4:/etc/modprobe.d#
`

MichaIng, 2020-10-11:
Okay so from the basics all looks fine.

Little hint: When pasting multi line logs or code, wrap them into tripple ``` instead of single ` as the latter omits all line breaks and multi white space, hence is only useful for single line commands, log/code snippets 😉. I edited you last post accordingly to show the difference.

cc13com, 2020-10-12:
Hello @MichaIng,

thanks for the hint and the correction of my last code-snippet. I found a page with a similar problem "RPI4+SSD+no Bluetooth". There was written that this problems are only if the SSD is connected via USB3. With USB2 connected all is fine. I will test that with my setup later this week and will give feedback.

MichaIng, 2020-10-12:
Many thanks to doing some research about this. Could you post the link to the page as well? In such cases, when not yet done, I'd like to report the issue to the RPi devs 🙂.

If it remains not possible, at least we know about it and can think of a section in our docs to collect such information and limitations.

cc13com, 2020-10-12:
Here are one example thread I [found](https://www.raspberrypi.org/forums/viewtopic.php?t=247085). But I can check that in my setup latest Wednesday.

cc13com, 2020-10-13:
So, I changed the SSD from USB3 to USB2 and the bluetooth scan with `hcitool lescan` is fast as expected and found some devices. And the bluetooth device is visible in FHEM. All is fine here.

Unfortunately the RootFS benchmark decreased from around 104 to 28 MiB/s. But as long as I don't use the RPi4 as a fileserver I'll go with this.

MichaIng, 2020-10-13:
Great to know, I'll check on RPi repositories if this has been reported already or is an expected limitation, else report.

MichaIng, 2021-05-07:
I close this as known issue. Outstanding is to test a dedicated power supply to rule that out. Since just is known for other hardware as well, there is otherwise nothing we can do our side.

closed 2021-05-07 by MichaIng