MichaIng/DietPi #3819: DietPi-Software | Switching Ralmlog to #2, leads Sonarr, Lidarr, Radarr to fail

Joulinar, 2020-10-10:
@MichaIng 

There is a report on the forum about this issue. https://dietpi.com/phpbb/viewtopic.php?f=11&t=8163

Looks like switching the Log mode will leads Sonarr, Lidarr, Radarr to fail. It seems files in `/var/log` are missing after switching to the new log mode. I was able to replicate the issue and indeed creating the files is fixing it. I guess it's happen somewhere on migrating data

 ```
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing DietPi-RAMlog: minimal, optimised logging

[  OK  ] DietPi-Software | Storing /var/log metadata to disk
[  OK  ] DietPi-Software | Cleaning /var/log mountpoint
[  OK  ] DietPi-Software | Mounting tmpfs to /var/log
[  OK  ] DietPi-Software | Restoring metadata to /var/log tmpfs
```

On my test, I noticed that `/var/log` was totally empty right after log mode switch and before reboot.

```
root@DietPiVM1:/var/log# ls -la
total 4
drwxrwxrwt  2 root root   40 Oct 11 01:29 .
drwxr-xr-x 12 root root 4096 Apr 16 12:07 ..
root@DietPiVM1:/var/log#
```
Probably the restore is not working correctly? 🤔

MichaIng, 2020-10-11:
I think the only thing that is missing is a `/boot/dietpi/func/dietpi-ramlog 0` indeed because the service is already running, hence `systemctl start dietpi-ramlog` is simply without any effect. The installer seems to expect that DietPi-RAMlog was disabled before. Also there is no unmount done.

I think we should assure that the service is stopped. So if it is `systemctl is-active dietpi-ramlog`, run `systemctl stop dietpi-ramlog`, else `dietpi-ramlog 1`. Similarly, if `findmnt /var/log`, `umount -R /var/log` to assure we really clean the dir below the mount point and no the tmpfs.

MichaIng, 2020-10-11:
This should have fixed it: https://github.com/MichaIng/DietPi/commit/84e2efb175f38b52cf1e62645adc1094d31bacbf
I'll run some tests.

Joulinar, 2020-10-11:
yep seems to be working now

closed 2020-10-17 by MichaIng