MichaIng/DietPi #3820: Installing XFCE/LXDE with RealVNC fails to fetch remote file

dracnoc, 2020-10-11:
#### Details:                                                                                                                                                                                
- Date           | Sun Oct 11 10:39:02 BST 2020                                                                                                                                              
- DietPi version | v6.32.2 (MichaIng/master)                                                                                                                                                 
- Image creator  | DietPi Core Team                                                                                                                                                          
- Pre-image      | Raspbian Lite                                                                                                                                                             
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)                                                                                                                                            
- Kernel version | Linux DietPi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux                                                                                           
- Distro         | buster (ID=5,RASPBIAN=1)                                                                                                                                                  
- Command        | wget --spider -t 1 -T 5 https://raw.githubusercontent.com/MichaIng/DietPi/master/.conf/dps_6/99-dietpi-dpms_off.conf                                                      
- Exit code      | 4
- Software title | DietPi-Software
#### Steps to reproduce:

1. Select LXDE or XFCE with RealVNC from dietpi-software optimised list.
2. Install as normal
#### Expected behaviour:

- Install LXDE or XFCE desktop environments with RealVNC server software
#### Actual behaviour:

- A remote file is not fetched and the installation stops to retry or exit
#### Extra details:

- No extra steps, automatic installation
#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2020-10-11 10:39:02--  https://raw.githubusercontent.com/MichaIng/DietPi/master/.conf/dps_6/99-dietpi-dpms_off.conf
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... failed: Name or service not known.
wget: unable to resolve host address ‘raw.githubusercontent.com’
```

Joulinar, 2020-10-11:
Hi,

many thanks for your message. It seems you have a local DNS issue as you are not able to resolve hoste name `raw.githubusercontent.com`. Are you using a local DNS server? Something like PiHole?

dracnoc, 2020-10-11:
Thank you! That nailed it.

PiHole is on a second RPi and adding raw.githubusercontent.com to the whitelist allowed the installation to continue as normal.

Joulinar, 2020-10-11:
ok good that it is fixed. I'm going to close the issue. Feel free to reopen if needed.

closed 2020-10-11 by Joulinar

dracnoc, 2020-11-20:
Hi! If you've received this then I'm currently not available to
      respond at the moment.
    I'll get back to you soon, or later ... or whenever. Not to
      worry, just hum a tune like you're on a call hold and time will
      just fly by.
    Cheers!