MichaIng/DietPi #3821: fail2ban failed to start

vinhtq115, 2020-10-11:
### Creating a bug report/issue

#### Required Information
- DietPi version | `v6.32.2 MichaIng/master`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | CanaKit 3.5A Raspberry Pi 4 Power Supply (USB-C)
- SDcard used | Samsung Evo 16GB

#### Additional Information (if applicable)
- Software title | fail2ban
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi?
Not sure
- Bug report ID | `5889cee9-1de9-435b-9c7f-258cd988a79a`

#### Steps to reproduce
1. Install DietPi with OpenSSH, fail2ban
2. Cannot start fail2ban service

#### Expected behaviour
- fail2ban service should start

#### Actual behaviour
- fail2ban service failed to load

#### Extra details
This is the output from `sudo systemctl status fail2ban`. Not sure why it mentioned dropbear since I never used it.
```
dietpi@DietPi:~$ sudo systemctl status fail2ban
● fail2ban.service - Fail2Ban Service
   Loaded: loaded (/lib/systemd/system/fail2ban.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-10-11 17:05:49 +07; 8min ago
     Docs: man:fail2ban(1)
  Process: 809 ExecStartPre=/bin/mkdir -p /var/run/fail2ban (code=exited, status=0/SUCCESS)
  Process: 810 ExecStart=/usr/bin/fail2ban-server -xf start (code=exited, status=255/EXCEPTI
ON)
 Main PID: 810 (code=exited, status=255/EXCEPTION)

Oct 11 17:05:48 DietPi systemd[1]: Starting Fail2Ban Service...
Oct 11 17:05:48 DietPi systemd[1]: Started Fail2Ban Service.
Oct 11 17:05:49 DietPi fail2ban-server[810]:  Failed during configuration: Bad value substitution: o
ption 'filter' in section 'dropbear' contains an interpolation key 'mode' which is not a valid optio
n name. Raw value: '%(__name__)s[mode=%(mode)s]'
Oct 11 17:05:49 DietPi fail2ban-server[810]:  Async configuration of server failed
Oct 11 17:05:49 DietPi systemd[1]: fail2ban.service: Main process exited, co
de=exited, status=255/EXCEPTION
Oct 11 17:05:49 DietPi systemd[1]: fail2ban.service: Failed with result 'exi
t-code'.
```

Joulinar, 2020-10-11:
Hi,

many thanks for your message. Yes this is an already know issue and will be fixed with next release v6.33 #3813 

As workaround, you would need to add `mode = normal` before `filter` within `/etc/fail2ban/jail.conf` to get it working. Should looks like this

```
[DEFAULT]
enabled = true
ignoreip = 127.0.0.1/8
ignorecommand =
backend = systemd
mode = normal
filter = %(__name__)s[mode=%(mode)s]
findtime  = 600
```

Once done, restart the service `systemctl restart fail2ban.service`

vinhtq115, 2020-10-14:
The workaround worked.

closed 2020-10-14 by vinhtq115